betaflight/betaflight #7585: RC Commanded throttle and roll jumps ,but not by user ?

Pablo18011, 2019-02-16:
Flashed BF version 3.5.5 to a Matek F405-CTR  and noticed jumps in motor outputs not commanded
by me?
Apologies in advance if I have posted this in the wrong place and this has nothing to do with BF release 3.5.5.
Any help to point me in the right direction will be greatly appreciated

**To Reproduce**
https://www.youtube.com/watch?v=-lQ9jaHbpbU&feature=youtu.be

**Expected behavior**
The quad still performs and is able to fly well ,but a bit unnerving when this happens.

**Flight controller configuration**
///# diff

# version
# Betaflight / MATEKF405 (MKF4) 3.5.5 Jan 15 2019 / 10:05:36 (88921670b) MSP API: 1.40

board_name MATEKF405
manufacturer_id 

# name
name MARTIAN 2.1

# resources
resource MOTOR 5 NONE
resource PWM 3 NONE
resource LED_STRIP 1 A15
resource SERIAL_TX 2 NONE
resource SERIAL_TX 11 A02

# mixer

# servo

# servo mix

# feature
feature GPS
feature LED_STRIP
feature AIRMODE

# beeper
beeper -RX_LOST_LANDING
beeper -BAT_CRIT_LOW
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# beacon

# map

# serial
serial 0 2 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 3 16384 115200 57600 0 115200
serial 30 32 115200 57600 9600 115200

# led
led 0 8,12::AO:6
led 1 9,12::AO:6
led 2 10,12::AO:6
led 3 11,12::AO:6
led 4 8,13::AO:0
led 5 9,13::AO:0
led 6 10,13::AO:0
led 7 11,13::AO:0

# color

# mode_color
mode_color 6 1 2

# aux
aux 0 0 1 1800 2100 0 0
aux 1 2 5 1350 1650 0 0
aux 2 46 6 1800 2100 0 0
aux 3 13 5 1800 2100 0 0
aux 4 35 2 1800 2100 0 0
aux 5 36 0 1800 2100 0 0
aux 6 37 3 1800 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_calibration = -11,80,57
set mag_hardware = NONE
set rc_smoothing_type = FILTER
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = OFF
set fpv_mix_degrees = 29
set dshot_idle_value = 1500
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 20
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1500
set beeper_dshot_beacon_tone = 5
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_angle = 30
set gps_rescue_initial_alt = 20
set gps_rescue_descent_dist = 100
set gps_rescue_ground_speed = 1000
set gps_rescue_min_sats = 5
set yaw_deadband = 5
set pid_process_denom = 1
set tlm_inverted = ON
set osd_rssi_alarm = 30
set osd_cap_alarm = 1500
set osd_tim1 = 769
set osd_vbat_pos = 2124
set osd_rssi_pos = 2114
set osd_tim_1_pos = 2134
set osd_current_pos = 417
set osd_mah_drawn_pos = 2497
set osd_craft_name_pos = 2505
set osd_gps_speed_pos = 2544
set osd_gps_lon_pos = 2097
set osd_gps_lat_pos = 2081
set osd_gps_sats_pos = 2537
set osd_home_dir_pos = 2170
set osd_home_dist_pos = 2167
set osd_altitude_pos = 2517
set osd_warnings_pos = 329
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set vtx_band = 5
set vtx_power = 4
set vtx_freq = 5658

# profile
profile 0

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 7000
set iterm_relax = RP
set throttle_boost = 0
set p_pitch = 54
set p_roll = 50
set p_yaw = 66
set f_yaw = 0
set i_level = 0
set d_level = 0

# rateprofile
rateprofile 0

set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 70
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 85
///

**Setup / Versions**
BF 3.5.5 with GPS rescue enabled.
Matek F405-CTR 8k 8k
 Build 
Martian II 250
Racerstar RS30A V2 Blheli s Dshot 600 
Rubycon low esr capacitors all ESC's and VBat
BrotherHobby Returner R5 2306 2450KV 4-5S 
Dalprop Cyclone T5045C 3-Blade Props
Matek Systems BetaFlight F405-CTR (The good gyro)
RunCam Swift Rotor Riot Special Edition IR Block
Matek 5.8G 40CH 25/200/500mW  VTX-HV 
Realacc 5.8G 5dBi 50W Omnidirectional Pagoda Antenna 
FrySky Taranis X9D plus
FrySky R9M
FrySky R9MM Receiver
SBUS 1 to 16 channels
8M Ublox M8N GPS Module

**Additional context**
I have spent a bit of time looking into this and when it first happened I suspected maybe failsafe stage 1? but I could not see a flag for this in a previous log.I'm also using R9M module and GPS rescue and
thought may have something to do with this .I have had twitches before(mostly yaw) on other models
but this seems different.
Log file here
 https://www.dropbox.com/home?preview=LOG00028+EV200D_0023+GOPR1945+Stage+1+poss+at+110..BFL

etracer65, 2019-02-16:
I have no desire to create an account to download the log.

If you're seeing uncommanded rcCommand inputs (actually showing in the rcCommand traces) then it's some problem with your rx, tx or gimbals. Obvious things to check would be that you're running the latest receiver firmware version.

If you think you're seeing rcCommand inputs because the activity shows up in the motors (but not rcCommand) then it could be glitches from a failing gyro.

Pablo18011, 2019-02-16:
Thanks for your reply ,I only included the link to the log to provide as much information as possible.
I agree with you this could be tx /rx  or failing gyro issue and completely unrelated to BF 3.5.5.
What is strange for me is how fast the change is ,that's why I know this is not physical input on the sticks
from me.

Pablo18011, 2019-02-16:
The gimbals are new M9 and I have the latest version of firmware running on Taranis X9D plus ,R9M and R9MM.

jflyper, 2019-03-01:
I see this hard roll right was commanded ...

[image]

Pablo18011, 2019-03-01:
Thanks for the reply ,I see that too ,the problem is it was not commanded by me.
I have HD video and fpv video.The first small jump put me on alert as I was flying a lot higher than I
normally do, so when the second one occurred I know 100% this was not due to me.
Also if you watch the 3rd jump ,its way to quick for a normal slip on the sticks.
It may not  be Betaflight at all ,however I would like to find out what caused it .I have another build that I will soon test on the exact same set up to either rule out the quad or the TX?
The flying I was doing that day was scenic video so very smooth and flowing inputs from me ,not my 
normal acro.
Whats odd is I have experienced similar before on different builds and different TX's.
But it's first time I have clearly recorded it happen and been 100% certain I did not command the inputs?

etracer65, 2019-03-03:
Looks like a failing gimbal to me. If it was a problem with the receiver I'd expect to see the glitch on all axes at the same time. Having it only be on one axis points towards a gimbal. Definitely not a problem with Betaflight as it's just reacting (correctly) to the input.

Pablo18011, 2019-03-03:
That's the head scratcher, gimbals are brand new M9 and I have not had this issue on my other quad?
That's not to say that it's not the gimbals, maybe I have just not experienced it yet on my other quad?

JFly02, 2019-03-17:
I had the same problem (hello from the YouTube comments) but only on 1W(auto) mode with the r9m. I think it happened when the power mode was switching. That day I fixed it by flying with a stable Outputpower and now I'm back on 1W and it don't appeared again. 
And the twitching on my quad were that the right motors sped up for a sec and then go back to the the level of my input.

JFly02, 2019-03-17:
https://photos.app.goo.gl/fHzeGP9FC3dGRttC9
here is a link to some gopro video, one is after retightening the props and the other one begins with the twitching on 1W and after I set it on 500mW there is no more twitching.

Pablo18011, 2019-03-24:
Thanks for the reply, I'm going to close this now as I do believe my issue was caused by the R9MM or R9
module firmware and not Betaflight .What you describe and others from my Youtube comments are
very similar.This can also be caused by poor connection pins failing on the Qx7 as a user had this issue.
I have made firmware updates and will monitor, as I'm in a good position to gather as much data as I can if it happens again, it seems this issue can be caused by lots of reasons and difficult to pin right down.
If it happens again and I can prove exactly what causes my issue I will post on YouTube for others.
Thanks for all your input.

Pablo18011, 2019-03-24:
Now closed as I now believe this issue was software related R9MM/R9 modules.

closed 2019-03-24 by Pablo18011