betaflight/betaflight #7587: Erroneous ARMSWITCH/ANGLE arming disabled reason

fujin, 2019-02-16:
**Describe the bug**
While testing for #7582, I noticed that while the bug condition was active (zero filtered gyro outputs), and attempting to arm, I would receive arming disabled flags 'ARMSWITCH' and 'ANGLE' (cycling between the two).

I'd have expected 'ANGLE' code only in the context of `small_angle` exceeded. The data being 0, perhaps we have a `small_angle < 0` - however, the accelerometer data was properly coming through. It's like it's somehow coupled to the gyro readings instead of acc.

The ARMSWITCH.. not sure.

**To Reproduce**
1) Have a bug e.g. #7582 which can cause gyro output to 0,0,0
2) Observe 'ANGLE' arming disabled condition e.g. in `status`.
3) Attempt to arm configuration
4) Observe 'ARMSWITCH'/'ANGLE' arming disabled cycling.

**Expected behavior**
ARMSWITCH/ANGLE codes should only be presented when that is the reason for disabling arming.

**Flight controller configuration**
```
defaults nosave
set gyro_sync_denom = 2
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 1
set scheduler_optimize_rate = ON
save
```

**Setup / Versions**
 - Flight controller: AIKONF4 / N.A.
 - Other components: AIKON AK32PINF0 35A 4in1
 - Wiring: JST-SH-8P.

**Additional context**
Perhaps we should expect BAD GYRO or similar arming disabled status.

mikeller, 2019-02-16:
Taking a stab what is happening:
1. the invalid gyro output caused by the bug has the follow-on effect that the attitude calculation used for `small_angle` detection is within triggering range _intermittently_;
2. if `small_angle` was triggering while the arm switch was flipped, this will result in arming being disabled with an `ANGLE` warning;
3. when the trigger condition goes away, the arming disabled condition is deferred to `ARMSWITCH`. This is by design, to avoid the situation where arming is initially disabled, but then the condition is rectified, and the craft suddenly and without warning arms.

I think 2. and 3. are good the way they are - they just lead to a weird looking read out in this case. Maybe we could change `ARMSWITCH` to `ARM DISABLED` or suchlike in this case, but I think indicating to the user that the initial arming disable reason has been rectified is the right thing to do.

As to 1., I suspect the only way to mitigate invalid input data caused by bugs would be to implement far reaching (and probably stateful) plausibility validation for all flight relevant input data - not sure if the required effort is justified?

etracer65, 2019-02-16:
Fixed in #7589 

The displayed reasons were actually correct however the logic to suppress `ARMSWITCH` from displaying unless it's the only reason wasn't working because the arming disabled count wasn't updated when reasons were removed in an earlier PR.

etracer65, 2019-02-16:
Also in this case it wouldn't be appropriate to display the `NOGYRO` reason as that's based on the gyro being detected. That was not a problem for the situation described.

fujin, 2019-02-17:
Sounds good to me, thanks @etracer65 @mikeller.

PR #7589 (2019-02-19): Fix arming disabled flags count
Fixes #7587
closed 2019-02-19 by mikeller