betaflight/betaflight #7597: 3D Throttle Input and Curve Interpretation

drumdude0, 2019-02-17:
I have found that 3D center throttle, along with high/low deadbands are moved away from stick center when a throttle expo curve is applied under the pid/rates tab.

Radio, BF, and ESC's are defined and calibrated to 1010/1500/1990, with high deadband at 1510, low deadband at 1490, and bf's radio page throttle deadband of 10.

When configuring for 3D my previous 2D throttle expo of 0.01/0.22 remained, and was being applied to the throttle for 1000-2000 effectively moving 'center throttle' for 3D to around 1517/1518.

Is there an option to apply a throttle curve to 3D such that it curves from 1500 to both high and low throttle directions, and not from 1000-2000?

etracer65, 2019-02-17:
No, that's not currently implemented. The throttle curve is not 3D mode aware at this point.

TiagoTiago, 2019-12-31:
Isn't it a bug for the curve to be applied to the whole range in 3d mode instead of being applied to each half mirrored?

mikeller, 2019-12-31:
@TiagoTiago:

> Isn't it a bug for the curve to be applied to the whole range in 3d mode instead of being applied to each half mirrored?

Yes, correct. It's just a matter of somebody implementing and then testing it - unfortunately the number of 3D pilots testing for Betaflight is quite small.

TiagoTiago, 2019-12-31:
> @TiagoTiago:
> 
> > Isn't it a bug for the curve to be applied to the whole range in 3d mode instead of being applied to each half mirrored?
> 
> Yes, correct. It's just a matter of somebody implementing and then testing it - unfortunately the number of 3D pilots testing for Betaflight is quite small.

Should a separate issue be filed for the bug, or does this just need to be changed from Feature Request?

TiagoTiago, 2020-01-04:
Btw, if the concern is this code change could introduce bugs for existing users; how about adding a setting to enable the new code, something like "3d Throttle Curve Fix (experimental)", and if that setting is disabled, the old code is used instead?

etracer65, 2020-01-04:
Someone just needs to implement and test it. Unfortunately none of the active devs are 3D pilots so there’s not much interest.

zxol, 2022-04-12:
Due to the ambiguity of the interaction between the throttle curve and 3D mode, I think this should be tagged as a bug.  There is a safety concern for unintended take off on arm if the user has a throttle curve that will produce upward thrust when the throttle is within the deadband.  Any curve setting where midpoint is > 0.5 and expo > 0.1 can potentially cause this.

closed 2026-09-26 by blckmn