betaflight/betaflight #7598: Boat: Differential Thrust

webbhm, 2019-02-17:
There are several of us trying to create air boats with multiple motors and propellers (non-reversing).  Arduplane has some great differential thrust code, but it is hard to get a boat to ignore all the flight oriented fail-safe features (air-speed, altitude, etc) to work readily on a boat.
It would be nice if there was a boat option for differential thrust, similar to skid-steer (right and left motors tied to  both throttle and rudder), but without the reversing.
The other option would be a full blown servo mixing; rather than hard coding the differential thrust (thrust and rudder), allow a user to define the mixing (and mixing ration) for any input and output.
I don't see a work around to accomplish this at this time for a Rover/Boat.

etracer65, 2019-02-17:
First of all I just want to make sure you realize that this is the Betaflight project and not related to "ardu-anything"...

Definitely not something Betaflight is designed for but if you're creative you might be able to get something to work. See here: 

https://github.com/betaflight/betaflight/blob/master/docs/Mixer.md#example-4-custom-airplane-with-differential-thrust

webbhm, 2019-02-17:
Oops.  Sorry, I was writing on the wrong tab.  I have two projects going,
betaflight is doing fine.

closed 2019-02-17 by etracer65

webbhm, 2019-02-18:
Wrong git