betaflight/betaflight #7602: 4.0 #1441 RTH_rescue home set needs to be on first ARM

laurentopia, 2019-02-18:
**Is your feature request related to a problem? Please describe.**
Currently, home is set on each arm.
This is a problem when for example I land the quad on some faraway place, disarm to take some snapshot or use the gimbal to look around and then re-arm the quad.
If I need to RTH then it will be that spot, not my home.

**Describe the solution you'd like**
Add an option which only set home on first ARM

McGiverGim, 2019-02-18:
This was done some time ago: https://github.com/betaflight/betaflight/pull/7369

laurentopia, 2019-02-20:
Lovely, was it added to 4.0 default?

McGiverGim, 2019-02-20:
No, you need to go to CLI in 4.0 to enable it.

mikeller, 2019-02-20:
@laurentopia: Like @McGiverGim mentioned earlier, see #7369 for details:
- the pull request has '4.0' as the milestone, so yes, it will be in 4.0;
- the description mentions the CLI setting `gps_set_home_point_once` that is used to enable it.

closed 2019-03-31 by stale[bot]