betaflight/betaflight #7607: ESC RPM HZ OSD Element

spatzengr, 2019-02-19:
To create an ESC RPM HZ OSD element to report the real-time ESC RPM HZ data that you get from Debug Mode = RPM_FILTER, **for each motor**, in the OSD.  This would enable one to record DVR, and sync that with a log; to record raw noise and get a sense where the new ESC RPM telemetry is telling the notches to reside at certain points of the flight.  

To adjust the ESC RPM OSD element to report all 4 motor RPM data vs. averaging all 4 motors.

McGiverGim, 2019-02-19:
I don't know if this will be useful, the OSD data has a very low refresh rate. You will not see the real time RPM value. 
For example, for the max current, I can see in my blackbox logs a maximum of 80A at the end of rolls but the OSD maximum records only 60A, because the max is only reached a very few miliseconds.

spatzengr, 2019-03-22:
Added through https://github.com/betaflight/betaflight/pull/7671

closed 2019-03-22 by spatzengr