betaflight/betaflight #7608: Frsky SPI RX binds fine, but no channel signals detected

salavatov, 2019-02-19:
1. "EMAX E6" 6ch TX (from Tinyhawk RTF) is claimed to be Frsky D8. Works fine with Tinyhawk.
2. Eachine Trashcan with Crazybee F4FR PRO fc with integrated Frsky RX on spi

TX binds fine with Crazybee F4 FC RX, but no channels detected. RF signal is detected (checked with 
debug_mode = RX_SIGNAL_LOSS and debug_mode = RX_FRSKY_SPI by turning bound TX on/off).

Same Trashcan works fine with older Taranis and Horus, so spi RX is fully functional.

Exactly same behavior experienced with factory firmware:
CRAZYBEEF4FR (C4FR) 4.0.0 Dec 11 2018 / 01:44:17 (3b479f92d) MSP API: 1.41
and latest 4.0.0
and current stable 3.5.6

```
# diff

# version
# Betaflight / CRAZYBEEF4FR (C4FR) 4.0.0 Dec 11 2018 / 01:44:17 (3b479f92d) MSP API: 1.41

board_name CRAZYBEEF4FR
manufacturer_id 

# name
name Trashcan

# resources

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature -ANTI_GRAVITY
feature -DYNAMIC_FILTER
feature MOTOR_STOP
feature LED_STRIP

# beeper

# beacon

# map

# serial
serial 0 2048 115200 57600 0 115200

# led
led 0 4,7::CO:2
led 1 5,7::CO:2
led 2 6,7::CO:2
led 3 7,7::CO:2
led 4 8,7::CO:2

# color

# mode_color

# aux
aux 0 0 1 1525 2100 0 0
aux 1 1 0 900 1300 0 0
aux 2 2 0 1675 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_calibration = 57,-106,-135
set rx_spi_protocol = FRSKY_D
set blackbox_device = NONE
set dshot_idle_value = 250
set motor_pwm_protocol = DSHOT600
set vbat_min_cell_voltage = 29
set vbat_warning_cell_voltage = 30
set ibata_scale = 1175
set yaw_motors_reversed = ON
set osd_warn_esc_fail = OFF
set osd_vbat_pos = 2102
set osd_rssi_pos = 2083
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2134
set osd_flymode_pos = 2456
set osd_throttle_pos = 2423
set osd_crosshairs_pos = 2253
set osd_current_pos = 2400
set osd_mah_drawn_pos = 2432
set osd_craft_name_pos = 2443
set osd_warnings_pos = 329
set osd_disarmed_pos = 2346
set vtx_band = 5
set vtx_channel = 8
set vtx_freq = 5917
set frsky_spi_tx_id = 139,9
set frsky_spi_offset = -93
set frsky_spi_bind_hop_data = 116,171,226,25,80,135,190,245,44,99,154,209,8,63,118,173,228,27,82,137,192,247,46,101,156,211,10,65,120,175,230,29,84,139,194,249,48,103,158,213,12,67,122,177,232,31,86,0,0,0

# profile
profile 0

# rateprofile
rateprofile 0

set thr_expo = 40
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 130
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 80

```

jflyper, 2019-02-19:
Are these correct?
Taranis x Trashcan : OK
Horus x Trashcan : OK
EMAX E6 x Trashcan : NG
EMAX E6 x Unknown Receiver (TinyHawk) : OK

salavatov, 2019-02-19:
Yes. 
+ Tinyhawk's built in RX is Frsky D8 compatible on SBUS
+ EMAX E6 x CrazyBeeF4FR - RSSI detected correctly

mikeller, 2019-02-19:
@salavatov: Looks like EMAX E6 isn't FrSky compatible - maybe open a support request with them?

salavatov, 2019-02-19:
That means their RX is kind of dual functioning? Proprietary with E6 and regular D8 with others?

mikeller, 2019-02-19:
@salavatov: That, or the implementation of FrSky D that they use is broken to an extent that it 'just works' FrSky TX => their RX, but not their TX => FrSky RX.

salavatov, 2019-02-20:
I will try to bind this TX to other Frsky RXs this Saturday

mikeller, 2019-02-20:
@salavatov: Good idea, please report back in here, there might be other users interested in finding out what works and what doesn't.

salavatov, 2019-10-15:
@mikeller any updates from EMAX?

mikeller, 2019-10-15:
@salavatov: Sadly no reaction whatsoever. :disappointed:

mikeller, 2019-10-15:
@salavatov: Yes, sadly Eachine is another manufacturer that is well known to throw their users under the bus by selling hardware that comes with pre-installed binaries for custom firmware, with no source available.
Also, note that this isn't a case of 'policies' - these manufacturers are in breach of the licensing conditions for Betaflight, which are stated in the [GPLv3](https://github.com/betaflight/betaflight/blob/master/LICENSE). In most jurisdictions, customers who bought a product from Eachine that contains unlicensed firmware could probably use this to get a refund for their product.
Efforts to try and work with Eachine to get them to do the right thing by their customers have been fruitless so far - at some point we'll probably have to look into adding detection of targets that are in breach of the GPL into configurator, and then pop up a warning to customers alerting them to the bad deal they are getting, and asking them to stop supporting these manufacturers by buying their products.

mikeller, 2019-10-30:
@salavatov: Some complaining helped, looks like they've come around. They've published the source that they claim can build a working firmware for their hardware on https://github.com/dk3chao/betaflight-4.0.4. I built this file to put their claim to the test:

[betaflight_4.0.4_MATEKF411RX.zip](https://github.com/betaflight/betaflight/files/3788056/betaflight_4.0.4_MATEKF411RX.zip)

I don't have the hardware, so can you please test if this is working for you?

Unfortunately their repository is a full upload and does not contain a commit history, so at the moment it is only possible to make educated guesses at what they changed. One candidate is:

```
mikeller@pc:~/git/betaflight$ git diff 4.0.4 -- src/main/rx/cc2500_frsky_d.c
diff --git a/src/main/rx/cc2500_frsky_d.c b/src/main/rx/cc2500_frsky_d.c
index 32a9a169b..38d6b3655 100644
--- a/src/main/rx/cc2500_frsky_d.c
+++ b/src/main/rx/cc2500_frsky_d.c
@@ -165,7 +165,8 @@ void frSkyDSetRcData(uint16_t *rcData, const uint8_t *packet)
     decodeChannelPair(channels + 4, packet + 12, 4);
     decodeChannelPair(channels + 6, packet + 14, 3);
 
-    for (int i = 0; i < RC_CHANNEL_COUNT_FRSKY_D; i++) {
+    //for (int i = 0; i < RC_CHANNEL_COUNT_FRSKY_D; i++) {
+    for (int i = 0; i < RC_CHANNEL_COUNT_FRSKY_D-2; i++) {     //modify by dk3
         if ((channels[i] < 800) || (channels[i] > 2200)) {
             dataError = true;
```

No idea why they'd change this, or why not having this change results in this not working (one possible explanation is that for whatever reason their hardware would be sending invalid channel values for the last two channels).

descampsa, 2021-01-30:
Hey, any news on this? I have a similar problem with the emax e6 from tinyhawk rtf. 
The binding works, but the reception does not work properly. I see some response to my inputs in betaflight, but it is intermitent, and most of the time, all controls are stuck to their default value. 
The receivers i have tried are the Quanum FR8 Nano and RadioMaster R81.
The led ont the receivers, that should be permanently on, is also intermitently on.
I have also tried to connect the RadioMaster R81 to an arduino to see what happens on the sbus port (with [this](https://github.com/bolderflight/SBUS) library, and i see a lot of 'frame lost' and 'failsafe activated' flags (most of the frames actually).
Finally, i have tried to apply the path above on betaflight (version 4.2.6, FURYF4), with no change.
I stand to be corrected, but all of this makes me think that it is not something that can be fixed in betaflight, but is simply due to a broken implementation of the protocol in the Emax E6, that is only compatible with their receiver.

mikeller, 2021-02-01:
@descampsa: Yes, these TX will never work with hardware RX, only with flight controllers with integrated SPI RX.

MAAZ396, 2022-10-23:
Hi guys, is there any solution found for this? 
I bought a new mobula 7 hd to use with a emax e6 controller with d8 but it doesn’t seem to bind , it binds but nothing moves in th e reciever tab even when I cycle the power the red light is not always solid sometimes it’s there sometimes it’s not or sometimes it’s blinking in 3 flashes pattern please help guys it’s driving me crazy , I can’t bind mobula 7hd to emax e6 , help!

MAAZ396, 2022-10-27:
> @descampsa: Yes, these TX will never work with hardware RX, only with flight controllers with integrated SPI RX.

Hi ! 

 bought a new mobula 7 hd to use with a emax e6 controller with d8 but it doesn’t seem to bind , it binds but nothing moves in th e reciever tab even when I cycle the power the red light is not always solid sometimes it’s there sometimes it’s not or sometimes it’s blinking in 3 flashes pattern please help guys it’s driving me crazy , I can’t bind mobula 7hd to emax e6 , help

MAAZ396, 2022-10-27:
@mikeller hi brother I  bought a new mobula 7 hd to use with a emax e6 controller with d8 but it doesn’t seem to bind , it binds but nothing moves in th e reciever tab even when I cycle the power the red light is not always solid sometimes it’s there sometimes it’s not or sometimes it’s blinking in 3 flashes pattern please help guys it’s driving me crazy , I can’t bind mobula 7hd to emax e6 , help, sorry for spamming this I'm super worried if I wasted my money on this please help

MAAZ396, 2022-10-27:
@salavatov  bought a new mobula 7 hd to use with a emax e6 controller with d8 but it doesn’t seem to bind , it binds but nothing moves in th e reciever tab even when I cycle the power the red light is not always solid sometimes it’s there sometimes it’s not or sometimes it’s blinking in 3 flashes pattern please help guys it’s driving me crazy , I can’t bind mobula 7hd to emax e6 , help

Lolenze25, 2023-04-15:
YES guys i dit it, my emax E6 Transmitter and my Mobula7 HD 2-3S are Bound and the Transmitter movement is showing in Betaflight, here is how i dit it, i just flashed the .hex data, wich was shared by mikeller, this one 

https://github.com/betaflight/betaflight/files/3788056/betaflight_4.0.4_MATEKF411RX.zip

im not done with the whole config and settings for the drone, but i wanted to share that it works.
Thanks

MAAZ396, 2023-04-15:
> YES guys i dit it, my emax E6 Transmitter and my Mobula7 HD 2-3S are Bound and the Transmitter movement is showing in Betaflight, here is how i dit it, i just flashed the .hex data, wich was shared [...]

Thanks for the reply , well I asked happymodel directly they sent a hex file , it bound and all good until I tried Flying it , I've tried everything it shows "runaway" even tho everything was correct , IDek what to do now it's been 6-7 months and I've broken one of the motor mistakenly pulled a wire ....

Lolenze25, 2023-04-16:
can you provide the hex file please

closed 2026-09-26 by blckmn