betaflight/betaflight #7612: Dhot_bidir not working with higher bf builds or blheli testcodes > 32.6.2

Hydroculture, 2019-02-19:
**Describe the bug**
Betaflight / OMNIBUSF4 (OBF4) 4.0.0 Jan 30 2019 / 00:57:37 (a4d2a987a) MSP API: 1.41 works very well
with dshot_bidir enabeled, when running BLheli_32 testcode 32.6.2

- updating BlHeli_32 to a higher build eg. 32.6.(3/4/5) breaks the running system
   - startup tones not synced
   - quad flies but when throtteling up, motor stops occur, reproduceable
   - turning off dshot_bdir solves the problems
- updating BF to a higher build than # Betaflight / OMNIBUSF4 (OBF4) 4.0.0 Jan 30 2019 / 00:57:37 
  (a4d2a987a) MSP API: 1.41 (@mikeller build it for me to get SA2.1) breaks the running system
   - startup tones not synced
   - quad flies but when throtteling up, motor stops occur, reproduceable
   - turning off dshot_bdir solves the problems
 
**To Reproduce**
- Flash the mentioned FW versions (and take care to save a copy of BLHELI32.6.2 testcode, as it cant be 
  found @github anymore)
- throttle up

**Expected behavior**
- dshot startup tones are synced
- quad flies even when throtteling up, motor stops do not occur
- turning on dshot_bdir works
- latest BLHeli_32 test codes work
- latest BF build work
- I dont want to sound unfriendly, thankful for your time, dedication and knowledge

**Flight controller configuration**
```
# diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 4.0.0 Jan 30 2019 / 00:57:37 (a4d2a987a) MSP API: 1.41

board_name OMNIBUSF4
manufacturer_id AIRB

# name

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 B14
resource SERIAL_TX 12 A01
resource SERIAL_RX 12 A01
resource LED 1 NONE
resource ESCSERIAL 1 NONE
resource ADC_CURR 1 NONE

# dmaopt

# mixer

# servo

# servo mix

# feature
feature -RX_SPI
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY

# beeper
beeper -ON_USB

# beacon
beacon RX_SET

# map
map EATR1234

# serial
serial 2 64 115200 57600 0 115200
serial 5 16384 115200 57600 0 115200
serial 31 2048 115200 57600 0 115200

# led

# color

# mode_color
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 13
mode_color 5 4 10
mode_color 5 5 3

# aux
aux 0 0 0 1300 2100 0 0
aux 1 2 0 1300 1700 0 0
aux 2 13 1 1300 1700 0 0
aux 3 26 2 1450 1700 0 0
aux 4 26 2 900 1300 0 0
aux 5 28 0 1300 2100 0 0
aux 6 30 2 1450 2100 0 0
aux 7 33 1 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 s 1000
rxfail 5 s 1500
rxfail 6 s 1350
rxfail 7 s 1000

# master
set gyro_sync_denom = 2
set dyn_lpf_gyro_max_hz = 700
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 8
set fpv_mix_degrees = 30
set max_aux_channels = 4
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set blackbox_record_acc = OFF
set min_throttle = 1064
set dshot_idle_value = 600
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_poles = 9
set failsafe_delay = 10
set current_meter = NONE
set vbat_detect_cell_voltage = 30
set vbat_scale = 112
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 10
set yaw_deadband = 10
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_rssi_alarm = 40
set osd_cap_alarm = 850
set osd_tim2 = 2817
set osd_vbat_pos = 2496
set osd_rssi_pos = 2273
set osd_tim_1_pos = 54
set osd_tim_2_pos = 23
set osd_flymode_pos = 333
set osd_anti_gravity_pos = 474
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs_pos = 234
set osd_ah_sbar_pos = 234
set osd_ah_pos = 200
set osd_current_pos = 470
set osd_mah_drawn_pos = 472
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2505
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2283
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 473
set osd_esc_rpm_pos = 333
set osd_core_temp_pos = 324
set osd_stat_rtc_date_time = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_used_mah = OFF
set osd_stat_max_fft = ON
set debug_mode = FFT_FREQ
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_freq = 5917

# profile
profile 0

set dterm_lowpass2_hz = 0
set anti_gravity_gain = 6600
set iterm_relax_type = GYRO
set throttle_boost = 9
set f_pitch = 100
set f_roll = 100

# rateprofile
rateprofile 0
```
**Setup / Versions**
 - three identical flight controllers: HGLRC XJB F440 (OBf4 target, DinoShot 40A ESC)
 - Other components [RX=CRSF nano, VTX=TBS pro 32 nano, Motors 3br 1105 6000KV
 
**Additional context**
@sskaug was friendly enough to upload back the dinoshot 32.6.2 hex file, saved my day

best
Holger

Ipoc78, 2019-02-20:
set motor_poles = 9? what kind of motor is this?

Hydroculture, 2019-02-20:
@Ipoc78 but i opened the motor to be sure, and counted nine motor poles?

Ipoc78, 2019-02-20:
You need to be counting magnets.
But that aside I've never seen a motor that is odd number it's either 10 12 14 Etc.
I am curious to know what motor this is please? 

Sooo..
quad flies but when throtteling up, motor stops occur, reproduceable.
Double check and recount magnets I can see that being an issue.
( I'm not smart just trying to help)

As far as start-up Tones not being synced
I have had that problem but it still flew fine usually if I do a power cycle the next time I plug in there in sync again

joelucid, 2019-02-20:
@sskaug, do you know what this could be? It's not true for all blheli32 escs for sure. Just installed 32.6.5 on my ori32 today and it works.

Hydroculture, 2019-02-20:
Can someone please confirm whether set motor_poles results from counting statorpoles or magnets.
Motors are 3br 1105 6000kv
best 
Holger

Counting magnets now :)

Ipoc78, 2019-02-20:
@hydroculture .
It is mentioned in Joe lucid's guide to RPM filter
https://github.com/betaflight/betaflight/wiki/Bidirectional-DSHOT-and-RPM-Filter

Hydroculture, 2019-02-20:
@Ipoc78 oh if the man could read, thank you for explaining.
Testing with motor_poles = 12

Hydroculture, 2019-02-20:
Juhu!
correct motor_poles setting did it!
Just flashed latest bf build #1450, 
- Escs start up synced (10x on/off @3s, 10x on/off @4s)
- did two flights with lots of throttling up/down, agressive manouveurs, all fine

Ipoc78, 2019-02-20:
Fantastic! Happy flying

sskaug, 2019-02-20:
Awesome - great to hear!

Hydroculture, 2019-02-20:
@sskaug, @joelucid, @lpoc78
- flashed latest BLheli_32 testcode (x.5)
- esc start synced (10x on/off)
- works like a charm with BHt1 1407/3600, BH t1 1407/3600kv , 3br 1105 6000kv,
   hglrc xjb f4 with hglrc dinoshot 40a on all three quads.
- motorpoles means magnetcount! got it. Twelve, not nine!

Thank you all for your inputs and insisting on reading the wiki again.
- a (too small) donation is on its paypal way

best
Holger Landwehr

so i got them trustworth again, flying like falconxs ;)
[image]

closed 2019-02-20 by Hydroculture