betaflight/betaflight #7615: Relocate startup files

hydra, 2019-02-20:
we need to do a tidy up of the system initialisation files.

things that have been really annoying when working on FC/MCU startup code:
1) files with duplicate names - e.g. src/main/target/system_stm32f7xx.c != src/main/drivers/system_stm32f7xx.c
2) `src/main/target/link` is in the wrong place.  should NOT be in target folder.  It's super annoying to have to scroll half-way down a list of targets in a directory to get to the linker scripts.
3) `src/main/target/stm32*` and `src/main/target/system_stm*.h` should NOT be in the target folder
suggest relocating 2) and 3) into src/main/startup

When your dealing with system configuration, having related files spread all over the place is not helpful, neither is having commonly referenced files in directories that contain many other files.
(i'm looking at src/main/target and src/main/drivers in particular here)

Now, I know I'm partly to blame for this, much of the things I talk about date back to the very early days of Cleanflight - e.g. 3bc9dc7121c60005c4c3e851758f560821e4e05f

Anyway, if others, who have worked on early startup code, like @sambas @jflyper @blckmn would care to chime in on this with their thoughts please do.

blckmn, 2019-02-20:
@hydra https://github.com/betaflight/betaflight/pull/7616

blckmn, 2019-02-20:
https://github.com/betaflight/betaflight/pull/7618

blckmn, 2019-02-20:
I have moved a couple of them for you :)

hydra, 2023-04-11:
See also #12663 and #12664.