betaflight/betaflight #7628: LED strip random flashes

klutvott123, 2019-02-20:
**Describe the bug**
LED strip flashes white(or leds go dark. depends on color and brightness it seems) randomly. Looks like it started with build 1418. This happens with AIKONF4. Does not happen on MATEKF722.
First thing i tried was a diode in series with 5V to the leds. No difference. 
Think I have tracked it down to: https://github.com/betaflight/betaflight/blob/6e28a50bc413f4f3a3fe6b5f8aa253fd80609dd6/src/main/drivers/light_ws2811strip_stdperiph.c#L182
If i set the DMA mode back to normal it starts working as expected again.

**To Reproduce**
Flash build 1418 or later on AIKONF4 or other F4 board. Observe leds. It can take some time between each time this happens. 

**Expected behavior**
Expected behaviour is no random flashing.

**Flight controller configuration**
```
# diff

# version
# Betaflight / AIKONF4 (AIK4) 4.0.0 Feb 20 2019 / 11:04:21 (6e28a50bc) MSP API: 1.41

board_name AIKONF4
manufacturer_id 

# feature
feature -OSD
feature TELEMETRY
feature LED_STRIP

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB

# map
map TAER1234

# serial
serial 0 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

# led
led 0 4,4::CT:7
led 1 4,5::C:0
led 2 4,6::C:0
led 3 4,7::C:13
led 4 7,7::C:13
led 5 7,6::C:0
led 6 7,5::C:0
led 7 7,4::CT:7

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 3 1900 2100 0 0
aux 2 15 3 1400 1600 0 0
aux 3 26 1 1400 1600 0 0
aux 4 36 2 1900 2100 0 0

# rxfail
rxfail 3 h

# master
set gyro_sync_denom = 2
set acc_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set serialrx_provider = FPORT
set serialrx_inverted = ON
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1550
set ibat_lpf_period = 5
set pid_process_denom = 2
set thrust_linear = 50

# profile
profile 0

set vbat_pid_gain = ON
set f_yaw = 60

# rateprofile
rateprofile 0

set roll_rc_rate = 223
set pitch_rc_rate = 223
set yaw_rc_rate = 210
set roll_expo = 80
set pitch_expo = 80
set yaw_expo = 70
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0
```

**Setup / Versions**
Aikon F4
XSR receiver, Taranis Q X7 transmitter, Racerstar 35A 4 in 1 ESC, 
No special wiring. Led strip wired to led strip pad on FC. 5v and GND wired to 5v/GND pads right next to it.

mikeller, 2019-02-21:
@klutvott123: This is weird - I was not able to reproduce this with an AIKONF4 running your exact configuration:

https://photos.google.com/share/AF1QipOqC8_49UBuH25XSlZ50LqxZqCEuucd2U36GppvC-XFyajBVYH6tMfCm0buDI_Hig/photo/AF1QipOTuMq-ZCqtlSBjOGmeeUWXwDoVvundlEXEKwo3?key=V1ZEYlRla0hTVVE0cWFNUUZBbDJCb1E3RkZYZ3J3

Just an out there thought, could this be your RX doing weird things, and occasionally getting a stray frame hitting the LEDLOW range? Can you test again with the LEDLOW range removed?

klutvott123, 2019-02-21:
@mikeller Thanks for testing this yourself. It can sometimes take minutes before this happens.
I already tested without the LEDLOW range. I flashed it with build 1450, enabled the led strip and did this as the only changes:
```
led 0 4,4::CT:7
led 1 4,5::C:0
led 2 4,6::C:0
led 3 4,7::C:13
led 4 7,7::C:13
led 5 7,6::C:0
led 6 7,5::C:0
led 7 7,4::CT:7
```
Also tried setting all of them to red. Still does it. Seems like it's less frequent with this minimal diff, but it still happens. Then I set the DMA mode back to normal again, and it stops doing it. I have repeated this test several times, and the results are consistent. Did you try it with the whole diff? 

I have another one of these leds. I will try it with my fortini F4 to see if i can reproduce it. Will report back.

klutvott123, 2019-02-21:
Was able to reproduce on FF_FORTINIF4. 
Flashed build 1450 from configurator with full chip erase. Pasted diff which is pretty much the same as the one i use for the AIKONF4. Disconnected usb, turned on transmitter, connected power to the stack. It shows the same behaviour as with the AIKONF4. Then i tried enabling just the led strip and setting the colors as in my above post. Can't see any random flashing(but it did do it on the AIKONF4).

Then i compiled 1450 with DMA mode set back to normal. Flashed, pasted the same diff as before, disconnected usb, turned on transmitter, connected power to the stack. I have been observing it for a while now and haven't seen any random flasing yet. 

Here's the diff I'm using for this board:
```
# diff

# version
# Betaflight / FF_FORTINIF4 (FORT) 4.0.0 Feb 20 2019 / 11:19:30 (6e28a50bc) MSP API: 1.41

board_name FF_FORTINIF4
manufacturer_id 

# feature
feature LED_STRIP

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB

# map
map TAER1234

# serial
serial 2 0 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led
led 0 4,4::CT:7
led 1 4,5::C:0
led 2 4,6::C:0
led 3 4,7::C:13
led 4 7,7::C:13
led 5 7,6::C:0
led 6 7,5::C:0
led 7 7,4::CT:7

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 3 1900 2100 0 0
aux 2 15 3 1400 1600 0 0
aux 3 26 1 1400 1600 0 0
aux 4 36 2 1900 2100 0 0

# rxfail
rxfail 3 h

# master
set gyro_sync_denom = 2
set acc_hardware = NONE
set min_check = 1005
set max_check = 1995
set serialrx_provider = FPORT
set serialrx_inverted = ON
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set bat_capacity = 1550
set ibata_scale = 300
set pid_process_denom = 2

# profile
profile 0

set vbat_pid_gain = ON
set iterm_relax = RPY
set p_pitch = 54
set i_pitch = 45
set d_pitch = 34
set f_pitch = 20
set p_roll = 47
set i_roll = 40
set d_roll = 32
set f_roll = 20

# rateprofile
rateprofile 0

set roll_rc_rate = 223
set pitch_rc_rate = 223
set yaw_rc_rate = 210
set roll_expo = 80
set pitch_expo = 80
set yaw_expo = 70
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0
```

evnngsky, 2019-02-21:
I have a same case, But my problem produced by a DC-DC converter.
I needed to attach 40 LEDs, So I need an external 5V equipment.
When the voltage from the external DC-DC converter wasn't match to 5v from an FC, the problem had been reproduced.

hydra, 2019-02-21:
yeah, sounds like a noise issue on the LED strip data line.  Does your FC output 5v or 3v data signals?  If 3v3 and there is too much difference between LED power and LED data it wont work - Try dropping the voltage to the LED's to about 4.3V using an inline diode.

hydra, 2019-02-21:
See docs:

```
If you have LEDs that are intermittent, flicker or show the wrong colors then drop the VIN to less than 4.7v, e.g. by using an inline diode on the VIN to the LED strip. The problem occurs because of the difference in voltage between the data signal and the power signal. The WS2811 LED's require the data signal (Din) to be between 0.3 * Vin (Max) and 0.7 * VIN (Min) to register valid logic low/high signals. The LED pin on the CPU will always be between 0v to ~3.3v, so the Vin should be 4.7v (3.3v / 0.7 = 4.71v). Some LEDs are more tolerant of this than others.
```

klutvott123, 2019-03-04:
So I have been looking at the code trying to figure out why this happens. Everytime the transfer complete interrupt occurs, the stream is disabled manually in the interrupt routine. When using circular mode the DMA might already be transfering the data again before we have time to disable the stream. What happens when disabling the stream in the middle of a transfer? Does it stop right there or does it complete the transfer before stopping?

mikeller, 2019-03-05:
@klutvott123: That sounds reasonable. Do you want to open a pull request to fix this by only using circular DMA if `USE_WS2811_SINGLE_COLOUR` is defined?

klutvott123, 2019-03-05:
@mikeller I'm on it! 👍

hydra, 2019-03-06:
@klutvott123 good investigation work!

klutvott123, 2019-03-06:
@hydra Thanks! 👍

PR #7720 (2019-03-06): LED STRIP Use circular DMA for single colour mode only
Use circular DMA for single colour mode only. Use normal DMA for status mode to avoid unpredictable behaviour. Fixes #7628.
closed 2019-03-06 by mikeller