betaflight/betaflight #7644: Matek F722SE only connects if the connect button is pressed right after detection

McGiverGim, 2019-02-23:
**Describe the bug**
 I need to press the connect button in the configurator right after the FC is connected and detected, if not an error of opening serial port is shown in the log. The same happens to me flashing firmware, I need to press the flash button right after the FC is connected.

**To Reproduce**
Connect the FC (at least MATEK F722SE), wait some seconds and press connect. 

**Expected behavior**
The FC must connect to the configurator always, including some seconds after the FC is connected.

**Flight controller configuration**
No need of this. Defaults values are ok.

**Setup / Versions**
 - FC: Matek F722SE
 - Other components: Frsky r-xsr, GPS RadioLink TS100, all the other elements are off when connected by USB

**Additional context**
I have tried to disect to find the culprit, but I have not been able. It seems to me that it happens from a long time ago but the time needed to give the serial connection error has been reduced. For example, build 1399 seemed ok, but in a second round I reproduced the error with it too.

etracer65, 2019-02-23:
I can't replicate this. Flashed a bare MatekF722SE with current master and I can connect every time. Also flashing was no problem - other than the occasional boot wedge immediately after flash that happens on F722's that's fixed by power cycle. This was tested on MacOS.

McGiverGim, 2019-02-23:
Maybe is configurator related? My tests were under Windows. I think @fujin had the same issue. What SO are you using @fujin?

klutvott123, 2019-02-24:
MATEKF722 here. Don't know if it's related, but something similar happens on linux. I haven't tried pressing "connect" immediately after the FC is detected, but if i press it a bunch of times it will eventually connect. Doesn't happen on windows.

haka002, 2019-02-24:
I didn't have any problem with F722SE target and F722-mini board on windows 10.
Was the click may be too early? I mean if you restart the board you need 1 or 2 seconds until it is booted well. Before this, the FCs won't connect.

McGiverGim, 2019-02-24:
Is exactly the opposite. If I wait more than 2 or 3 seconds I get the error. I need to press the button right after the SO detects it. If I give some more time then it does not connect.

ijustwannafly, 2019-03-02:
Sounds more like driver issues on the pc than betaflight.  The configurator has an option to auto connect on detection.  Is that not working?

McGiverGim, 2019-03-03:
Today I have been trying to reproduce it again without luck. I don't know if some update of the firmware fixed it of simply is random bug, but by the moment I close this...

closed 2019-03-03 by McGiverGim