betaflight/betaflight #7647: IN-flight adjustments add "RATE_PROFILE_6" function

pezking, 2019-02-23:
Inflight Adjustments.md states "RATE_PROFILE - Switch between 3 rate profiles using a 3 position switch" and indeed something like "adjrange 11 3 3 900 2100 12 3 0 0" does assume 3 positions.  There are now 6 rate profiles in Betaflight, so it would be great to be able to use them all in flight.  

I ran into this need by flying a micro brushless that is fun to fly in both angle and acro so it makes sense to use different rates for each.   My plan was to mix a 2 position and a 3 position switch onto the same aux and use the 2 position to switch between flight modes and the 3 position would get me 3 rates each.  One could also use the 3 way switch to switch between angle acro and horizon and have 2 rates each.

I understand changing the current RATE_PROFILE function would result in 3 position users ending up with profile 1, 3, and 6, that is why I suggest a new function

mikeller, 2019-02-23:
@pezking: The default is for rate profile switching to assume a 3 position switch is used, and switch between rate profiles 1 - 3.
if you set `rate_6pos_switch = on` in CLI this will change to assuming a 6 position switch is used, and switch between rate profiles 1 - 6.

pezking, 2019-02-24:
Great, thanks!

closed 2019-04-02 by stale[bot]