betaflight/betaflight #7648: VTX vtxtable not configured

haka002, 2019-02-23:
I've just update to the latest 4.0 build from the configurator and I see 'Please configure vtxtable'.

I set this from another issue before: 
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

Matek F722-mini with se build.
VTX noname with 25/100/200 (maybe 0 (PIT) too)

Di I need to set more something for the vtxtable?

Thanks!

etracer65, 2019-02-24:
You should set the power levels appropriately for your VTX based on its documentation. Set the bands and frequencies however you want as long as the VTX supports the frequencies you specify. You can read more on the PR: https://github.com/betaflight/betaflight/pull/7540

etracer65, 2019-02-24:
And note that you should only configure channels and power levels that are legal to use in your country.

YairHH, 2019-02-24:
But how does one actually set this up in betaflight? (Newbie) I tried to find the command on CLI with no success

haka002, 2019-02-24:
> And note that you should only configure channels and power levels that are legal to use in your country.

Always! ;)

haka002, 2019-02-24:
I tried to configure everything well based on my documents:
I tried to change it to powerlevels 3 and without 0 (PIT), but the result is the same error message always.
Here is the document for the VTX: https://drive.google.com/file/d/1Zvw7gQy2e-PKUWXlBZLqGQVZNUfZekeE/view?usp=sharing

vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B b 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E C 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK d 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEB__1 E 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 RACEB__2 F 5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 4
vtxtable powervalues 0 25 100 200
vtxtable powerlabels 0 25 100 200

Any idea?

etracer65, 2019-02-24:
It looks like there's a logic problem with the way the `PLEASE CONFIGURE VTXTABLE` message is displayed. Should be fixed in https://github.com/betaflight/betaflight/pull/7651

Your setup looks correct except for you shouldn't set up 0 as a power level. That's what pit mode is for (if your VTX supports it).

haka002, 2019-02-24:
Ok I will check that ticket, thanks!

closed 2019-04-02 by stale[bot]