betaflight/betaflight #7653: Crash flip motor directions reversed whether in 3D mode or not.

AlbertPhan, 2019-02-24:
I have 3d enabled as a switch. When the quad is in 3d mode OR not 3D mode, the crash flip mode motor directions is reversed of what it should be. 

To reproduce:
Have quad upside down. Arm in crash flip mode and NOT 3d mode. Move cyclic to flip and the motors are reversed. Repeat with 3d mode switch enabled and same result.

Using an F3 board on my Furibee X140.
4 in 1 BLheli ESC
2 motors are reversed.

Using Omnibus firmware 3.5.6 latest as of this writing. 
And Dshot600 for the escs.
ESCs are set to bidirectional and bi reversed as needed for motors.

```name Furibee 140

feature -RX_PPM
feature -DYNAMIC_FILTER
feature RX_SERIAL
feature MOTOR_STOP
feature 3D
feature AIRMODE
beeper -ON_USB
beeper -CRASH FLIP
beacon RX_LOST
beacon RX_SET
serial 2 64 115200 57600 0 115200
aux 0 0 4 1700 2100 0 0
aux 1 1 0 1700 2100 0 0
aux 2 1 5 1700 2100 0 0
aux 3 2 0 1300 1700 0 0
aux 4 13 2 1700 2100 0 0
aux 5 29 1 1400 2100 0 0
aux 6 35 3 900 1300 0 0
set baro_hardware = NONE
set fpv_mix_degrees = 45
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set vbat_warning_cell_voltage = 33
set current_meter = NONE
set small_angle = 180
set deadband = 2
set yaw_deadband = 2
set osd_warn_batt_not_full = OFF
set osd_vbat_pos = 468
set osd_flymode_pos = 2477
set osd_throttle_pos = 2305
set osd_vtx_channel_pos = 449
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2380
set osd_battery_usage_pos = 456
set osd_stat_battery = ON
set rcdevice_init_dev_attempts = 4
profile 1

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set anti_gravity_threshold = 350
set feedforward_transition = 100
rateprofile 0
```

etracer65, 2019-02-24:
Crash flip mode is not supported when the 3D feature is enabled.

AlbertPhan, 2019-02-24:
I thought this was fixed in #7026? Is there a specific branch of firmware I need to use?

etracer65, 2019-02-24:
Sorry, forgot that we fixed it. Never mind.

Make sure your ESC's have an up to date firmware that supports DSHOT commands for motor reversing. Try testing with 3D mode disabled (turn it off on the Configuration tab) to see if the motors reverse like they should when crash-flip is enabled.

AlbertPhan, 2019-02-24:
Yes I just tried it with 3d mode disabled in the config tab it works normally. 
As an aside: I just tried flying 3D today and my drone was acting quite erratically when i was transitioning to flying upside down and I crashed it pretty bad. I broke a motor unfortunately.

AlbertPhan, 2019-03-27:
In the meantime, I have reverted back to non 3d mode to have the crash flip functionality. If I can help further investigate this issue let me know.

closed 2019-05-03 by stale[bot]