betaflight/betaflight #7654: Reference for Hardware Design

xen0nnn, 2019-02-24:
Hello
I am currently designing my own F4 flight controller. Sadly i'm having lots of unanswered questions about the supported hardware and recommended wiring of these parts. 
Is there some kind of reference design to follow to make sure everything is supported with the betaflight software?  
Thanks in advance!
Luca

mikeller, 2019-02-24:
@xen0nnn: There already are some open source schematics for flight controllers, probably most notably at https://alienflight.com/. One notable thing about designing flight controllers is that in order to get a design that will work in real world conditions, sound electrical engineering skills are required, and it is often non-obvious things like thermal problems or insufficient input voltage filtering that make the difference between a flight controller that tends to fail mid-flight and one that works.

Probably join our Slack (registration at https://slack.betaflight.com/) and discuss your plans with the Betaflight members in there.

atomiclama, 2019-03-07:
@xen0nnn There is also a FB group that have a lot of experience with FC design that may be of use: -
https://www.facebook.com/groups/1019540631468076/

closed 2019-03-08 by xen0nnn