betaflight/betaflight #7666: Gps-rescue decrease altidute

methisto, 2019-02-26:
Hi! When i switch on gps rescue, the quad dont try climb on the altidute i Set, he Just flys in my direction very slow and altidute is sinking! Which Parameter will cause this??

SchonRichtig, 2019-02-26:
How far away are you when you flip the switch?

methisto, 2019-02-27:
Ok, checked it out today...
When the battery is full first he loses a few meters, then he start to climb very slowly... with half-full battery he just sinks slowly

# get gps
gps_provider = UBLOX
Allowed values: NMEA, UBLOX
Default value: NMEA

gps_sbas_mode = AUTO
Allowed values: AUTO, EGNOS, WAAS, MSAS, GAGAN

gps_auto_config = ON
Allowed values: OFF, ON

gps_auto_baud = OFF
Allowed values: OFF, ON

gps_ublox_use_galileo = OFF
Allowed values: OFF, ON

gps_rescue_angle = 30
Allowed range: 0 - 200
Default value: 32

gps_rescue_initial_alt = 50
Allowed range: 20 - 100

gps_rescue_descent_dist = 100
Allowed range: 30 - 500
Default value: 200

gps_rescue_ground_speed = 1300
Allowed range: 30 - 3000
Default value: 2000

gps_rescue_throttle_p = 150
Allowed range: 0 - 500

gps_rescue_throttle_i = 20
Allowed range: 0 - 500

gps_rescue_throttle_d = 50
Allowed range: 0 - 500

gps_rescue_velocity_p = 80
Allowed range: 0 - 500

gps_rescue_velocity_i = 20
Allowed range: 0 - 500

gps_rescue_velocity_d = 15
Allowed range: 0 - 500

gps_rescue_yaw_p = 40
Allowed range: 0 - 500

gps_rescue_throttle_min = 1200
Allowed range: 1000 - 2000

gps_rescue_throttle_max = 1600
Allowed range: 1000 - 2000

gps_rescue_throttle_hover = 1280
Allowed range: 1000 - 2000

gps_rescue_sanity_checks = RESCUE_SANITY_OFF
Allowed values: RESCUE_SANITY_OFF, RESCUE_SANITY_ON, RESCUE_SANITY_FS_ONLY
Default value: RESCUE_SANITY_ON

gps_rescue_min_sats = 5
Allowed range: 0 - 50
Default value: 8

frsky_gps_format = 0
Allowed range: 0 - 1

osd_gps_speed_pos = 2104
Allowed range: 0 - 3071
Default value: 234

osd_gps_lon_pos = 234
Allowed range: 0 - 3071

osd_gps_lat_pos = 234
Allowed range: 0 - 3071

osd_gps_sats_pos = 2081
Allowed range: 0 - 3071
Default value: 234

SchonRichtig, 2019-02-27:
you can adjust thes settings:

gps_rescue_throttle_max = 1600 
Allowed range: 1000 - 2000
gps_rescue_throttle_hover = 1280
Allowed range: 1000 - 2000

Seems to be a really low powerd quad.

TonyBlit, 2019-03-25:
@methisto have you tried increasing gps_rescue_throttle_max and gps_rescue_throttle_hover as suggested in the last comment? If that has worked, please close this issue.

closed 2019-05-01 by stale[bot]