betaflight/betaflight #7668: Runaway Bug

T3flon, 2019-02-26:
**Describe the bug**
After I lifted off with the copater, I made a role. Then came the message Runway and I was crashed

**To Reproduce**
https://youtu.be/uxFgUyFcE9g

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**

**Setup / Versions**
 -Betaflight / OMNINXT7 (ONX7) 4.0.0 Feb 24 2019 / 10:05:09 (b5fba14cb) MSP API: 1.41
   #1461

**Additional context**
Add any other context about the problem here.

McGiverGim, 2019-02-26:
Take a look to the configuration, maybe you need to change something https://github.com/betaflight/betaflight/wiki/Runaway-Takeoff-Prevention

mikeller, 2019-02-26:
Agreed with @McGiverGim, most likely your `runaway_takeoff_deactivate_throttle_percent` is set too high.

etracer65, 2019-02-26:
Also you didn't give the feature any time to determine that you were actually flying.

closed 2019-02-26 by T3flon