betaflight/betaflight #7669: SP F3 Evo Lua not working on Betaflight 3.5.5 (Telem works)

bckenneth, 2019-02-26:
I had few SP F3 Evo in few quads with XSR Sbus/S.Port
on BT 3.2 everything works as expected. Lua PID on Radio Rate changes from radio to FC fine.

Just upgraded 2 quads to lastest BT 3.5.5 with all Xsr firmware up to date and X9D on 2.2.3 and Lua 1.1.0.
.both have same issues: Telemetry works. but PID Lua no data shows, I had also tried downgrade it back to BT 3.2 everything works as 100% expect. (btw I also delete all sensors and re discover ..all Telemetry found ok just Lua not working).

I had a quad with BF F3 Fc updated to BT 3.5.5 also works no issue with my X9D all Lua and Telemetry shows. 

Can anyone tell me what happens? did I miss some setting?

klutvott123, 2019-02-27:
It looks like MSP over telemetry has been disabled for the SPRACINGF3EVO target to make the firmware fit in flash, which means that lua won't work for this target. 
https://github.com/betaflight/betaflight/blob/7625f8b2d263401453343eccea2636c3f1cd5959/src/main/target/SPRACINGF3EVO/target.h#L40

It's still enabled for the BETAFLIGHTF3 target, which explains why that one is working for you.

bckenneth, 2019-02-27:
> It looks like MSP over telemetry has been disabled for the SPRACINGF3EVO target to make the firmware fit in flash, which means that lua won't work for this target.
> [betaflight/src/main/target/SPRACINGF3EVO/target.h](https://github.com/betaflight/betaflight/blob/7625f8b2d263401453343eccea2636c3f1cd5959/src/main/target/SPRACINGF3EVO/target.h#L40)
> 
> Line 40 in [7625f8b](/betaflight/betaflight/commit/7625f8b2d263401453343eccea2636c3f1cd5959)
> 
>  #undef USE_MSP_OVER_TELEMETRY 
> It's still enabled for the BETAFLIGHTF3 target, which explains why that one is working for you.

Thx klutvott123 
Now I know what happens. no more googling around for answers.
btw you have any idea what version of BF firmware Lua support up to? 

With many thanks

bckenneth, 2019-02-27:
I tried few version of BF FW tonight, the only FW lua script still support up to 3.4.1. Any FW after 3.4.1  Lua is disable.

mikeller, 2019-02-27:
@bckenneth: Yep, looks about right: https://github.com/betaflight/betaflight/commit/08294d21d8cad8c3fac524fab7efd25644106999

klutvott123, 2019-02-27:
Looks like it, yes. 
@bckenneth This should work for you:
[betaflight_3.5.7_SPRACINGF3EVO_msp_over_telemetry.zip](https://github.com/betaflight/betaflight/files/2910382/betaflight_3.5.7_SPRACINGF3EVO_msp_over_telemetry.zip)
This has msp over telemetry enabled to make lua work again. As a bonus it also has iterm relax enabled. 
Keep in mind it has acro trainer and servos disabled. Hope you don't need those 🙂

bckenneth, 2019-02-27:
> Looks like it, yes.
> @bckenneth This should work for you: [...]

for sure is a Great Boun. Many Thanks to you guys. Many of us don't care LED or Acro trainer or servos on Quads. Using Lua from radio is the key feature to keep this kind of F3 FC running🚀.

bckenneth, 2019-02-28:
Installed FW on all SP F3 board Lua Scripts works Thanks.

closed 2019-04-06 by stale[bot]