betaflight/betaflight #7670: Rename motor_poles command to something less confusing

jedimasterben, 2019-02-27:
Hi, was recommended by fujin to create an issue here.

In testing the new RPM filter, you have to set the number of motor poles, however the command is actually asking for the number of magnets in the bell. It is confusing (got me with it!) as it currently is named. Would it be possible to change it to something like motor_bell_magnets or motor_bell_magnet_count instead? Would make it more clear to the end user :)

Thank you for all you guys do!

etracer65, 2019-02-27:
Poles are part of the standard terminology for a motor's framework. For example many of the motors we use will be in a 12N14P configuration. This means 12 stator windings and 14 magnetic poles. It's very common to see this type of configuration description in the specs for a motor.

jedimasterben, 2019-02-27:
Okay, that makes sense then. Can the wiki be updated, at least, to denote that poles would be the magnet count of the bell? Would clear it up for people like me that are comfortable changing to use some of the experimental features but are missing just that last little bit to make it all click. :)

etracer65, 2019-02-27:
Sure, feel free to update the wiki. We encourage contributors. All you need is a GitHub account and clearly you have that already. 😉

jedimasterben, 2019-02-27:
Done! Also edited more of it for clarity, all the info is there just worded a bit better IMHO. :)  Thanks again!

closed 2019-02-27 by jedimasterben

mikeller, 2019-02-27:
Thanks for updating the wiki, @jedimasterben!