betaflight/betaflight #7678: 4.0.0 #1470 Enable LED-STRIP on OMNIBUS F3 and USB communication stops working

0crap, 2019-02-28:
I flashed the latest Jenkins build # 1470 on my OMNIBUS F3 board with full chip erase.
After flash BF configurator connection via USB is working fine.

Enable LED-strip en do a save-reboot.
BF Configurator is not able to connect to the flight controller any more.
Is reproducable every time.

Only way to connect again is use boot-switch on flight controller and reflash the board.

TheIsotopes, 2019-02-28:
I can confirm this on a matek f405 ... @mikeller pls can you see whats wrong

Barux, 2019-03-01:
Same here, HobbyWing G2 (OmnibusF4SD target) when enabling led_strip, USB comunication interrupt and only flash again connecting in DFU can restore the comunication

JTRiver, 2019-03-01:
Confirm same issue on a HGLRC F4 Flame V2 (omnibusF4 target) from #1469 to #1474.
Ok on #1468

Barux, 2019-03-01:
Tried latest #1474 where it says "set usbDetectPin to input pull down" but nothing to do, problem persists

PR #7680 (2019-03-01): Fixed LED_STRIP and TRANSPONDER.
Oh the fun of global variables used to track local state...

Fixes #7678.
closed 2019-03-01 by mikeller