betaflight/betaflight #7688: Turtle mode spins diagonal motors

exocode, 2019-03-01:
When I activate turtle mode, the **diagonal motors are spinning** _(in the right "reversed" direction, like expected)_. The only problem is, that they are diagonal (front right and left rear/front left and right rear). In terms of Betaflight 1+4 or 2+3. 

Have the most recent stable Betaflight version as well the most recent available ESC firmware for my ESCs.

All other behavior is working as expected. No motor settings are reversed.

[image]
(fourth one has the same as the three above)

etracer65, 2019-03-01:
Please provide the information requested when you opened the issue.

exocode, 2019-03-01:
Sorry didn't see that (opened a "configuration issue" / seems the issue template is not providing such information)

**Describe the bug**
When I activate turtle mode, the **diagonal motors are spinning** _(in the right "reversed" direction, like expected)_.

**To Reproduce**

- Place the quad upside down
- enter turtle mode
- yaw left or right with sticks

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
```
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# Betaflight / FF_FORTINIF4 (FORT) 3.5.6 Feb 16 2019 / 12:59:25 (c28ca22f9) MSP API: 1.40

board_name FF_FORTINIF4
manufacturer_id 

# name
name Lisam 210

# resources

# mixer

# servo

# servo mix

# feature
feature -DYNAMIC_FILTER
feature GPS
feature TELEMETRY
feature LED_STRIP
feature AIRMODE

# beeper

# beacon
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led
led 0 6,15::AW:0
led 1 7,15::G:0
led 2 8,15::G:0
led 3 9,15::AW:1

# color

# mode_color
mode_color 6 0 5
mode_color 6 1 2

# aux
aux 0 0 0 1500 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 46 7 1700 2100 0 0
aux 4 13 5 1700 2100 0 0
aux 5 15 6 1300 1700 0 0
aux 6 19 5 1300 1700 0 0
aux 7 35 2 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_calibration = -11,28,-3
set rssi_channel = 8
set fpv_mix_degrees = 20
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1500
set vbat_max_cell_voltage = 42
set small_angle = 45
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_initial_alt = 30
set gps_rescue_descent_dist = 60
set gps_rescue_ground_speed = 1500
set gps_rescue_min_sats = 0
set pid_process_denom = 2
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2241
set osd_rssi_pos = 2209
set osd_remaining_time_estimate_pos = 2508
set osd_flymode_pos = 2048
set osd_anti_gravity_pos = 85
set osd_crosshairs_pos = 2317
set osd_ah_pos = 2190
set osd_current_pos = 384
set osd_mah_drawn_pos = 2528
set osd_craft_name_pos = 458
set osd_gps_speed_pos = 2060
set osd_gps_sats_pos = 2072
set osd_home_dir_pos = 2106
set osd_home_dist_pos = 2065
set osd_compass_bar_pos = 74
set osd_altitude_pos = 2054
set osd_warnings_pos = 2153
set osd_avg_cell_voltage_pos = 439
set osd_battery_usage_pos = 2536
set osd_disarmed_pos = 2282
set osd_nheading_pos = 2522
set osd_esc_tmp_pos = 440
set osd_core_temp_pos = 440
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set osd_stat_bb_no = OFF
set vcd_video_system = PAL

# profile
profile 0

set dterm_notch_cutoff = 0
set feedforward_transition = 10

# rateprofile
rateprofile 0

```
**Setup / Versions**
MacOS, Chrome: 67.0.3396.62, Configurator: 10.4.0
```
$ version
# Betaflight / FF_FORTINIF4 (FORT) 3.5.6 Feb 16 2019 / 12:59:25 (c28ca22f9) MSP API: 1.40
```

 - **Flight controller** Fortini F4
 - **Other components** 
    - ESC Racerstar RS20A x4  v16.7
 - **BlHeli** `ESC G-L-30 16.7`

**Additional context**

etracer65, 2019-03-01:
> When I activate turtle mode, the **diagonal motors are spinning** _(in the right "reversed" direction, like expected)_.

Can you explain further? Do you mean the diagonal motors start spinning immediately when you arm or when you move the sticks?

The only way diagonal motors can spin is if you induce yaw. So maybe that's what you're doing? Or maybe you need to check your yaw channel centering and/or increase your yaw deadband. Or possibly it's a misunderstanding about how turtle mode works. Normal usage is with the roll/pitch stick and these will spin 2 motors on one side. The special case is yaw and it's meant to spin diagonal motors to allow you to "shake" the quad out of a tree for example.

exocode, 2019-04-01:
I sold the drone now. I think it's a damaged F4 controller, which caused that issue.

closed 2019-04-01 by exocode