betaflight/betaflight #7699: Generic targets dont show timer and dmaopt pin list on dump or diff

MJ666, 2019-03-02:
Reproduced this with STM32F405 and STM32F7X2 targets and AlienFlight F4 and AlienFlightNG F7 boards. In the dump or diff the list of configured timers and the dmaopt pin list is not shown. You can get the related information with "timer list" and dmaopt all" cli commands. If I use the native AlienFlight targets the information is present as expected in diff and dump commands.

diff all:
`# diff all

# version
# Betaflight / STM32F405 (S405) 4.0.0 Mar  2 2019 / 07:01:01 (29db27584) MSP API: 1.41

board_name ALIENFLIGHTF4
manufacturer_id 
mcu_id 003400343033470b32353839
signature 

# reset configuration to default settings
defaults nosave

# name

# resources
resource BEEPER 1 C13
resource MOTOR 1 B08
resource MOTOR 2 B09
resource MOTOR 3 A00
resource MOTOR 4 A01
resource MOTOR 5 C06
resource MOTOR 6 C07
resource MOTOR 7 C08
resource MOTOR 8 C09
resource PPM 1 A08
resource PWM 1 A08
resource PWM 2 B00
resource PWM 3 B01
resource PWM 4 B14
resource PWM 5 B15
resource LED_STRIP 1 B15
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 4 C10
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 4 C11
resource INVERTER 2 C15
resource I2C_SCL 1 B06
resource I2C_SDA 1 B07
resource LED 1 C12
resource LED 2 D02
resource RX_BIND_PLUG 1 B02
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 C02
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 C03
resource SPI_MOSI 3 B05
resource ESCSERIAL 1 A08
resource ADC_BATT 1 C00
resource ADC_RSSI 1 C04
resource ADC_CURR 1 C01
resource ADC_EXT 1 C05
resource SDCARD_CS 1 B10
resource SDCARD_DETECT 1 B11
resource FLASH_CS 1 B12
resource GYRO_EXTI 1 C14
resource GYRO_CS 1 A04

# timer

# dmaopt
dmaopt SPI_TX 2 0
# SPI_TX 2: DMA1 Stream 4 Channel 0
dmaopt ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature MOTOR_STOP

# beeper

# beacon

# map
map TAER1234

# serial
serial 1 64 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set align_mag = CW180FLIP
set mag_bustype = I2C
set mag_i2c_device = 1
set baro_bustype = I2C
set baro_i2c_device = 1
set serialrx_provider = SPEK2048
set spektrum_sat_bind = 5
set blackbox_device = SDCARD
set dshot_burst = ON
set battery_meter = ADC
set beeper_inversion = ON
set beeper_od = OFF
set sdcard_detect_inverted = ON
set sdcard_mode = SPI
set sdcard_spi_bus = 2
set flash_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW270
set i2c1_pullup = ON

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save`

timer list:
# timer list
timer A08 0
timer B00 1
timer B01 1
timer B14 0
timer B15 0
timer B08 0
timer B09 0
timer A00 1
timer A01 1
timer C06 1
timer C07 1
timer C08 1
timer C09 1``

mikeller, 2019-03-02:
Thanks for reporting this, @MJ666. I am aware of this, fixing up the CLI display for resource management is on my burndown list for 4.0-RC1.
Also note that I'll rename `dmaopt` into `dma`, and give it similar syntax to how `resource` is done now.

TheIsotopes, 2019-03-02:
I have a similar problem ... diff command shows timer and dmaopt pin list even though they have not been changed ... can you check this also @mikeller

MJ666, 2019-03-02:
@TheIsotopes @mikeller I can confirm this for the native AFF4 and AFNGF7 targets.

mikeller, 2019-04-01:
This has been fixed in #7012.

closed 2019-04-01 by mikeller