betaflight/betaflight #7703: Furious VTX Power levels

dgrat, 2019-03-02:
Via "set vtx_power = 4" I cannot set the power level to 600 mW.
Actually the whole scale is incorrect. 0 and 4 is ignored. 3 leads to 200 mW. Via OSD I can activate 600 mW. 
Validated by power measurement. Maybe changing the level to 1-5 instead of 0-4 would already fix this issue. 

vtx_power = 4
Allowed range: 0 - 4
Default value: 1

jflyper, 2019-03-02:
@dgrat 
Does this happen with CMS, too?

And please provide the version of firmware at least.

dgrat, 2019-03-03:
Verion: Betaflight / OMNIBUSF4SD (OBSD) 3.5.5 Jan 15 2019 / 10:14:41 (88921670b) MSP API: 1.40
Problem also occured before. 

In CMS I am able to set the tx power up to 700 mW if "vtx_low_power_disarm" is off. Some steps are ignored (e.g. 100 mW) as explained.
The main problem I see is, that you cannot currently use "vtx_low_power_disarm" with 25mW/700mW.

jflyper, 2019-03-03:
@dgrat 
This issue hasn't been reported for the original Tramp, so it's probably an isolated case with Furious's unit. To be definite, we have to find someone to repeat the procedure and see if it reproduces for the original Tramp, too.

If the case is isolated case with Furious's unit, the issue will most likely NOT fixed, because it is a Furious's problem who made the "almost compatible" product.

If the case is global, situation is unclear, as the first RC for 4.0 is coming out in any minute; I'm not sure if there will be another maintenance release of 3.5. Then, the issue will not be fixed in 4.0 unless it is identified and fixed by 4.0 release date. Further more, VTX drivers and CMS are scheduled to receive a big change in how frequencies and power levels are configured in 4.1, the issue itself might go away.

dgrat, 2019-03-03:
@jflyper 
It's fine. Probably Tramp support is better. I just don't own one.

dgrat, 2019-03-04:
@jflyper 
If the Furious should be Tramp compatible. Why it seems to have just 5 instead of 6 power levels in CLI?

In vtx_common.h:

// Tramp "---", 25, 100, 200, 400, 600 mW
#define VTX_TRAMP_POWER_OFF       0
#define VTX_TRAMP_POWER_25        1
#define VTX_TRAMP_POWER_100       2
#define VTX_TRAMP_POWER_200       3
#define VTX_TRAMP_POWER_400       4
#define VTX_TRAMP_POWER_600 5

jflyper, 2019-03-05:
I think CLI value is offset by 1.

closed 2019-04-11 by stale[bot]