betaflight/betaflight #7704: Adc Vbat broken on build #1480 and later on furyf4osd and possibly more.

grahamwho, 2019-03-02:
4**Describe the bug**
ADC Vbat is broken on furyf4osd (at least) on build #1480 and later.

**To Reproduce**
Flash #1480+, check vbat.

**Expected behavior**
Vbat should work but it doesn't, I believe I test build #1479 and it worked but #1480 and later didn't

**Flight controller configuration**
Happens on a fresh flash on defaults.

**Setup / Versions**
 -Diatone mamba using the furyf4osd target

**Additional context**
I'm guessing the adc buffer stuff that was added in build 1480 broke adc vbat, may have broke adc current I don't remember seeing current when flying so I can't be sure about this. I'm guessing this is affecting all targets but if not, let me know if you want me to do/try anything else. I did go back to build #1479 and adc vbat works again.

mikeller, 2019-03-03:
@grahamwho: Should be fixed in #7687, please try again with the latest build.

klutvott123, 2019-03-03:
#7687 broke it. Will be fixed with #7702. My mistake. Sorry for the inconvenience.

laz2wiringpi, 2019-03-13:
OMNIBUSSF4 - i had same issue in 1480 and can confirm its now fixed

mikeller, 2019-03-14:
Closing this as fixed.

closed 2019-03-14 by mikeller