betaflight/betaflight #7706: DSHOT 600 not working KAKUTE F7 - No boot

jamesbroom, 2019-03-03:
**Describe the bug**
When flashing any 4.0 greater than 1468 then selecting DSHOT 600 I get the three tones from the ESC but the FC never boots so no fourth or fifth tone. 1467 and earlier work as expected.

**To Reproduce**
Flash BF 4 > 1468 onto Kakute F7.  Using Mamba 40A ESC.

**Expected behavior**
FC should boot when applying power to ESC.

**Flight controller configuration**
`# diff

# version
# Betaflight / KAKUTEF7 (KTF7) 4.0.0 Feb 27 2019 / 10:30:53 (9f8ad1aa4) MSP API: 1.41

board_name KAKUTEF7
manufacturer_id 

# name
name CHECK6

# resources
resource MOTOR 1 E11
resource MOTOR 2 E09
resource MOTOR 3 B01
resource MOTOR 4 B00

# dmaopt

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature LED_STRIP

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 1 16384 115200 57600 0 115200
serial 6 0 115200 57600 0 115200

# led
led 0 4,7::C:12
led 1 5,7::C:12
led 2 6,7::C:12
led 3 7,7::C:12
led 4 8,7::C:12
led 5 9,7::C:12

# color

# mode_color

# aux
aux 0 0 1 900 1425 0 0
aux 1 19 0 1300 1725 0 0
aux 2 35 0 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set dyn_lpf_gyro_max_hz = 435
set acc_calibration = 38,10,97
set mag_hardware = NONE
set rssi_channel = 7
set serialrx_provider = SRXL
set serialrx_halfduplex = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set bat_capacity = 1500
set small_angle = 180
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_rssi_alarm = 45
set osd_cap_alarm = 1100
set osd_vbat_pos = 2433
set osd_rssi_pos = 2081
set osd_tim_2_pos = 2102
set osd_flymode_pos = 37
set osd_g_force_pos = 51
set osd_throttle_pos = 2441
set osd_vtx_channel_pos = 2095
set osd_crosshairs_pos = 2253
set osd_current_pos = 2448
set osd_mah_drawn_pos = 2456
set osd_craft_name_pos = 44
set osd_altitude_pos = 2085
set osd_pidrate_profile_pos = 361
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 353
set osd_disarmed_pos = 2410
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_fft = ON
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 1
set vtx_freq = 5769

# profile
profile 0

set p_pitch = 36
set p_roll = 36
set p_yaw = 40
set i_yaw = 70
set d_yaw = 20

# rateprofile
rateprofile 0

set roll_srate = 78
set pitch_srate = 78
set yaw_srate = 78
set tpa_rate = 60
set tpa_breakpoint = 1750`

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
Kakute F7, Diatone Mamba 40A BLHELI_S ESC (16.7)
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
Atlatl HV V2 VTX on UART 1
Runcam Micro Swift 3 on UART 2
Spektrum 4649T on UART 6
 - how are the different components wired up.

**Additional context**

Following is from 1469 flashed to FC.

`# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: USB_DETECT
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 4
B01: MOTOR 3
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: SERIAL_TX 2
D06: SERIAL_RX 2
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 2
E10: FREE
E11: MOTOR 1
E12: FREE
E13: FREE
E14: FREE
E15: FREE`

`# dma

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE`

mikeller, 2019-03-03:
Confirmed. From testing with a unified target, it seems to be that the board is wedging when BARO is configured, and then Dshot is enabled.
Testing with a non-unified target with BARO compiled out confirms this.

@jamesbroom: If you feel like experimenting, you can load the latest build of the STM32F745 target onto your board, and then apply the following config _before applying your own config_:

```
# Betaflight / KAKUTEF7 (KTF7) 4.0.0 Mar  3 2019 / 20:23:18 (b1211ad6e) MSP API: 1.41

board_name KAKUTEF7
manufacturer_id HBRO

# resources
resource BEEPER 1 D15
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11
resource MOTOR 5 C09
resource MOTOR 6 A03
resource PPM 1 E13
resource LED_STRIP 1 D12
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 D06
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 E07
resource I2C_SCL 1 B06
resource I2C_SDA 1 B07
resource LED 1 A02
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 4 E02
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 4 E05
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 4 E06
resource ESCSERIAL 1 E13
resource ADC_BATT 1 C03
resource ADC_RSSI 1 C05
resource ADC_CURR 1 C02
resource SDCARD_CS 1 A04
resource SDCARD_DETECT 1 D08
resource OSD_CS 1 B12
resource GYRO_EXTI 1 E01
resource GYRO_CS 1 E04
resource USB_DETECT 1 A08

# timer
timer E13 0
timer B00 0
timer B01 1
timer E09 0
timer E11 0
timer C09 1
timer A03 1
timer D12 0

# dmaopt
dmaopt SPI_TX 1 1
# SPI_TX 1: DMA2 Stream 5 Channel 3
dmaopt ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dmaopt pin E13 1
# pin E13: DMA2 Stream 6 Channel 6
dmaopt pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dmaopt pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dmaopt pin E09 2
# pin E09: DMA2 Stream 3 Channel 6
dmaopt pin E11 1
# pin E11: DMA2 Stream 2 Channel 6
dmaopt pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dmaopt pin A03 0
# pin A03: DMA1 Stream 1 Channel 6
dmaopt pin D12 0
# pin D12: DMA1 Stream 0 Channel 2

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature OSD

# serial
serial 20 1 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# master
# Re-enable these once I2C has been fixed on F7
#set baro_bustype = I2C
#set baro_i2c_device = 1
#set baro_i2c_address = 0
set serialrx_provider = SBUS
set adc_device = 1
set dshot_burst = OFF
set motor_pwm_protocol = ONESHOT125
set current_meter = ADC
set battery_meter = ADC
set beeper_inversion = ON
set beeper_od = OFF
set sdcard_detect_inverted = ON
set sdcard_mode = SPI
set sdcard_dma = OFF
set sdcard_spi_bus = 1
set system_hse_mhz = 8
set max7456_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 4
set gyro_1_sensor_align = CW270
```

(N.b. this config will have to be re-applied whenever the board is reset to defaults.)

jamesbroom, 2019-03-03:
@mikeller Flashed STM32F745, rebooted FC by disconnecting and reconnecting USB, able to connect via Betaflight 10.4.0.  Applied your board specific config, rebooted.  Able to connect.  Changed motor protocol to DSHOT 600 and Save/Reboot.  Unable to connect to FC.  Blue activity light on FC stays on solid instead of blinking as normal.  Only way to recover is to switch to DFU and reflash.  Also tried switching off BARO and rebooting before switching to DSHOT 600.

mikeller, 2019-03-03:
@jamesbroom: Sorry, my earlier comment was premature. I thought I had the wedging pinned down to the I2C baro combined with Dshot, but turns out they were two independent conditions that both cause wedging.

Anyhow, Dshot is fixed in #7707.

jamesbroom, 2019-03-03:
@mikeller Feedback.  Flashed 1483 and Kakute F7 + DSHOT 600 issue appears resolved.  FC boots and operates normally.

PR #7707 (2019-03-04): Fixed Dshot for hal (F7).
Fixes #7706.
closed 2019-03-04 by mikeller