betaflight/betaflight #7722: BF 4.0 Motor #4 not spinning with Dshot on CLracing F7

goodnerdboy, 2019-03-05:
**Describe the bug**
Motor #4 does not spin with any Dshot on CLracing F7 in builds [#1469](https://ci.betaflight.tech/job/Betaflight/1469/)+, last version tested is [#1487](https://ci.betaflight.tech/job/Betaflight/1487/) and it is still broken.
 
**To Reproduce**
Just install [#1469](https://ci.betaflight.tech/job/Betaflight/1469/)+ on a CLracing F7 and try to spin the motors, either through configurator/motors or trying to arm without USB connection. Motors #1, #2 and #3 spin but not motor #4.

**Expected behavior**
All motors spin the same way with Dshot.

**Flight controller configuration**
[diffCLracingF7.txt](https://github.com/betaflight/betaflight/files/2932084/diffCLracingF7.txt)

**Setup / Versions**
 - Flight controller CLracing F7 bought from DroneFpvRacer;
 - ESC T-Motor 55A 4in1, does the same with racerstar 50A;
 - wired through connector.

**Additional context**
First posted this as a comment to  #7366 but figured the post and the versions related were probably too old to be the same issue. If it is related and this is a double post, i appologies.
Thanks!!

jdd1777, 2019-03-05:
Im using the talon f7 with the 1488 clracingf7 build and have the same issue.

dadoribeiro, 2019-03-06:
There was an issue like the one you described when the RPM filters were being tested on F7s. Here is the list (https://github.com/betaflight/betaflight/wiki/Bidirectional-DSHOT-and-RPM-Filter). The solution was to use the LED pad.

mikeller, 2019-03-07:
I only own an early version of the board that has different motor pins, so I cannot test with the current firmware.

Maybe @bnn1044 can help?

bnn1044, 2019-03-07:
@mikeller. I will test it tonight.

goodnerdboy, 2019-03-08:
Can I help for the tests in any way?

bnn1044, 2019-03-10:
comfired the MOTOR 4 doesn't work. 

here is the DMA list

# dma
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin B08 0
# pin B08: DMA1 Stream 7 Channel 2
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5

motor 4 don't have a DMA assignment. that may be the problem. since the board was setup to use Dshot_burst before. now the DMA didn't assign to B09.

mikeller, 2019-03-10:
@bnn1044: In that case, resetting the board to defaults (which enables `dshot_burst`) should make it work, or not?

bnn1044, 2019-03-10:
@mikeller that is the defaults right from the new flash. looks like default setting the DMA is not in for motor 4.

mikeller, 2019-03-10:
@bnn1044: So does motor 4 work after flashing with 'Full chip erase' or not? The individual pin DMA settings are not used if `dshot_burst` is on.

bnn1044, 2019-03-10:
@mikeller no, motor 4 it doesn't work after the flashing.

mikeller, 2019-03-10:
@bnn1044: This is odd - I can't test here as I don't have the hardware, but I'll need a `diff all` and a `resource show all` to be able to tell more.

bnn1044, 2019-03-10:
@mikeller # diff all
# version
# Betaflight / CLRACINGF7 (CLR7) 4.0.0 Mar  9 2019 / 20:07:32 (9c4e750f3) MSP API: 1.41
board_name CLRACINGF7
manufacturer_id 
mcu_id 003b00363137511130343533
signature 
# reset configuration to default settings
defaults nosave
# start the command batch
batch start
# name
# resources
# timer
# dma
# mixer
# servo
# servo mix
# feature
# beeper
# beacon
# map
# serial
# led
# color
# mode_color
# aux
# adjrange
# rxrange
# vtx
# rxfail
# display_name
# master
set motor_pwm_protocol = DSHOT600
# profile
profile 0
# profile
profile 1
# profile
profile 2
# restore original profile selection
profile 0
# rateprofile
rateprofile 0
# rateprofile
rateprofile 1
# rateprofile
rateprofile 2
# rateprofile
rateprofile 3
# rateprofile
rateprofile 4
# rateprofile
rateprofile 5
# restore original rateprofile selection
rateprofile 0
# save configuration
save

resource show all
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: PREINIT 2
B00: LED 1
B01: FREE
B02: RX_BIND
B03: CAMERA_CONTROL
B04: BEEPER
B05: SPI_MOSI 3
B06: MOTOR 1
B07: MOTOR 2
B08: MOTOR 3
B09: MOTOR 4
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D02: SERIAL_RX 5

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

mikeller, 2019-03-10:
@goodnerdboy, @bnn1044: Can you try #7754? This should fix it.

goodnerdboy, 2019-03-10:
I’ll test that in an hour or two

goodnerdboy, 2019-03-10:
@mikeller this works, motor #4 spins normally now.
Thanks!!

mikeller, 2019-03-10:
Thanks for testing, @goodnerdboy!

riveraj082, 2019-04-03:
What was done to fix this issue i am confused? I have the same issue with motor 4.

mikeller, 2019-04-04:
@goodnerdboy: If this has been fixed, can you close the ticket please?

@riveraj082: See https://github.com/betaflight/betaflight/issues/7722#issuecomment-471273816.

closed 2019-04-04 by goodnerdboy

goodnerdboy, 2019-04-04:
@riveraj082 this was fixed in PR 7754 (see https://github.com/betaflight/betaflight/issues/7722#issuecomment-471273816)
If you have a new issue you should report it and send all the information needed.

smreutzel, 2019-07-05:
Does #7554 fix the issue mentioned in https://github.com/betaflight/betaflight/wiki/Bidirectional-DSHOT-and-RPM-Filter#SupportedTargets for the CLRacing F7 that states "Motor 4 doesn't work. Use the LED pad instead"?