betaflight/betaflight #7729: BF 4.0 motor remap not working on MATEKF722

ExaflopS, 2019-03-06:
**Describe the bug**

BF 4.0 #1482 was last build that worked right, and all builds from #1483 dont initialize ESC after remapping motors.

**To Reproduce**
Steps to reproduce the behavior:
- Flash BF 4.0 #1483 or newer
- choose Dshot 600, or Dshot1200 protocol
- remap motors

**Expected behavior**
After connecting battery, ESCs dont initialize.

**Flight controller configuration**
# diff all

# version
# Betaflight / MATEKF722 (MKF7) 4.0.0 Mar  5 2019 / 14:57:03 (dc4866437) MSP API: 1.41

board_name MATEKF722
manufacturer_id 
mcu_id 002400323237510b34363730
signature 

# reset configuration to default settings
defaults nosave

# name
name CHAMELEON TI

# resources
resource MOTOR 1 C09
resource MOTOR 2 C08
resource MOTOR 3 C07
resource MOTOR 4 C06

# timer
timer A03 2
timer C06 1
timer C07 1
timer C08 1
timer C09 1
timer B01 1
timer A08 0
timer B08 0
timer A02 1
timer A00 1
timer A01 1
timer A15 0

# dmaopt
dmaopt pin A03 NONE
dmaopt pin C06 1
# pin C06: DMA2 Stream 2 Channel 7
dmaopt pin C07 1
# pin C07: DMA2 Stream 3 Channel 7
dmaopt pin C08 1
# pin C08: DMA2 Stream 4 Channel 7
dmaopt pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dmaopt pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dmaopt pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dmaopt pin B08 0
# pin B08: DMA1 Stream 7 Channel 2
dmaopt pin A02 0
# pin A02: DMA1 Stream 0 Channel 6
dmaopt pin A00 0
# pin A00: DMA1 Stream 2 Channel 6
dmaopt pin A01 0
# pin A01: DMA1 Stream 4 Channel 6
dmaopt pin A15 0
# pin A15: DMA1 Stream 5 Channel 3

# mixer

# servo

# servo mix

# feature

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST

# map
map TAER1234

# serial
serial 2 32 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 2 1750 2100 0 0
aux 2 13 2 1300 1600 0 0
aux 3 26 0 1300 2100 0 0
aux 4 35 3 1725 2100 0 0
aux 5 49 3 1250 1600 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h
rxfail 6 s 1500

# display_name

# master
set acc_calibration = 8,25,-4
set rssi_channel = 6
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 10
set align_board_yaw = 180
set bat_capacity = 1000
set vbat_warning_cell_voltage = 340
set yaw_motors_reversed = ON
set pid_process_denom = 1
set sdcard_dma = ON
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 55
set osd_cap_alarm = 1000
set osd_rssi_pos = 2049
set osd_current_pos = 2368
set osd_mah_drawn_pos = 2402
set osd_altitude_pos = 2071
set osd_warnings_pos = 2058
set osd_stat_tim_1 = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 4
set vtx_freq = 5769

# profile
profile 0

set vbat_pid_gain = ON
set anti_gravity_gain = 6000
set p_pitch = 41
set d_pitch = 44
set f_pitch = 90
set p_roll = 39
set d_roll = 42
set f_roll = 90
set p_yaw = 25
set i_yaw = 110
set d_min_roll = 0
set d_min_pitch = 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 195
set pitch_rc_rate = 195
set yaw_rc_rate = 195
set roll_expo = 70
set pitch_expo = 70
set yaw_expo = 70
set roll_srate = 62
set pitch_srate = 62
set yaw_srate = 62

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

**Setup / Versions**
MATEKF722 STD(MKF7)

 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
R9 mini RX /connested sbus and s-port -flex fw 181228 
TBS unify Pro HV + smart audio connected VTX

mikeller, 2019-03-07:
Tested on MATEKF722 with build 1491, the above config works just fine:

[image]

ExaflopS, 2019-03-07:
Tahnk you. I have tried build 1491 and now it works fine. 
I had to remove timer and dmaopt before pasting diff_all settings from 1482 build into 1491.

closed 2019-03-07 by ExaflopS