betaflight/betaflight #7738: diff hardware - understanding ask

TheIsotopes, 2019-03-08:
`diff hardware` only lists `baro_hardware` but not `mag_hardware` ... Is it wanted like that?
imo we should also use `mag_hardware` or remove `baro_hardware` from the list.

@mikeller You introduced this with #7221 ... what is your opinion?

mikeller, 2019-03-08:
@TheIsotopes: Good point. In general, I think that only settings that _need_ to be specified for a board to work should be included in the `dump hardware` list - this is meant to be the bare minimum, and is intended as a tool to make it easier for manufacturers and users to convert their custom firmware into a target configuration.

Using `baro_hardware = AUTO` should work fine for all boards, so this should be removed from the list.

Do you want to open a pull request for this?

TheIsotopes, 2019-03-08:
pull request was created now

closed 2019-03-09 by TheIsotopes