betaflight/betaflight #7740: Betaflight 4.0 RC1 & DalRC F722

Strangedave, 2019-03-09:
Just installed 4.0 RC1 and used a diff from 4.0.0 Feb 28 2019 / 17:14:10 (8483b09a5) to set it up. When i arm only the number 4 motor is spinning. If i try to increase throttle the Quad flips over. 

**To Reproduce**
Load 4.0 rc1 and arm the quad

**Expected behavior**
i have been using a build from this page for a few days was working well. https://github.com/joelucid/rpmfilterf7

**Flight controller configuration**
 
```diff```

# version
# Betaflight / DALRCF722DUAL (DLF7) 4.0.0 Mar  8 2019 / 03:05:53 (37dd39659) MSP API: 1.41

board_name DALRCF722DUAL
manufacturer_id 

# start the command batch
batch start

# name
name WIN 5R

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 1 16384 115200 57600 0 115200
serial 2 2 115200 9600 0 115200
serial 3 32 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1125 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1300 1700 0 0
aux 4 31 3 1900 2100 0 0
aux 5 35 2 1700 2100 0 0
aux 6 49 3 1300 1700 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 0
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 200
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 0
set acc_calibration = -26,-8,34
set mag_hardware = NONE
set baro_bustype = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 16
set blackbox_p_ratio = 64
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 16000
set failsafe_switch_mode = KILL
set ibata_scale = 139
set ibata_offset = -500
set yaw_motors_reversed = ON
set small_angle = 35
set gps_provider = UBLOX
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2081
set osd_rssi_pos = 2093
set osd_tim_2_pos = 2103
set osd_flymode_pos = 2414
set osd_crosshairs_pos = 2286
set osd_current_pos = 2086
set osd_mah_drawn_pos = 2097
set osd_craft_name_pos = 2381
set osd_gps_speed_pos = 2401
set osd_gps_sats_pos = 2425
set osd_warnings_pos = 2346
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 4
set vtx_freq = 5880
set vcd_video_system = NTSC

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 7000
set feedforward_transition = 5
set smart_feedforward = ON
set p_pitch = 43
set i_pitch = 79
set d_pitch = 32
set f_pitch = 65
set p_roll = 45
set i_roll = 70
set d_roll = 34
set f_roll = 65
set p_yaw = 55
set i_yaw = 125
set d_yaw = 5
set f_yaw = 59
set launch_trigger_allow_reset = OFF
set launch_angle_limit = 35

# rateprofile
rateprofile 0

set roll_srate = 71
set pitch_srate = 71
set yaw_srate = 69
set tpa_breakpoint = 1850

**Setup / Versions**

5"  WIN Frame
TMotor F-40 Pro v2 2400kv
Runcam microeagle
R-XSR receiver
Tramp HV
Lumenier AXII LHCP
DALRC Rocket 50A 4-in-1 BLHeli32
HQProp DP 5x4.8x3 PC V1S
DALRC F722
Buzzer from bangood
Microphone from bangood
battery 4s 75c 1550mah

butterSnake, 2019-03-09:
If you power cycle the quad does it get come good?

JRuscavage, 2019-03-09:
I loaded RC1 yesterday on my Foxeer F7(it looks like a copy of the Dal F722 and something with the esc arming is broken.  ESC's wont arm with any Dshot protocol.  I can get esc to arm using oneshot125 but  then only one motor will spin up and will twitch without any input(motor #2).  I flashed back to build 1431 which I knew worked and all is good again.

Using a Foxeer F7 with Aikon AK32 4 in 1 esc, Frsky xm+ receiver.  Let me know if any additional info is needed.

Strangedave, 2019-03-09:
> If you power cycle the quad does it get come good?

I tried three times no change. I have not yet loaded the last working firmware. I think i will revert back to the previous version for now. Pretty sure this is an actual bug in the Firmware. Just hoping to let the Devs know.

Thanks for the help either way.

Dave

djtaotao, 2019-03-09:
The same issue here, DALRCF722DUAL.
With joelucid hex all motors are spinning.

Strangedave, 2019-03-09:
Just verified test hover with joelucid hex also. Guess I'm waiting for rc2

djtaotao, 2019-03-09:
Here is the solution for now: https://github.com/betaflight/bidircfg/blob/master/DALRCF722DUAL.cf

Strangedave, 2019-03-09:
Will try this later tonight. I see that you are dshot 600. I am currently set for dshot1200 and I was getting twitches at 8/8k so I set for 4/4k and if the rain stops I’m going to try that. Do you know that 8/8 will work with dshot 600 for sure?

Strangedave, 2019-03-09:
Thanks for the help by the way.

mikeller, 2019-03-09:
@Strangedave:

> set dshot_bidir = ON

Try `dshot_bidir = off`. If everything works then you will need a special resource configuration for bidirectional Dshot to work, and this is a known limitation of the current implementation of bidirectional Dshot, and this issue should be closed.

Strangedave, 2019-03-09:
Just making sure that I’m doing my part in the testing of this firmware. As long as it’s. Know problem I will go ahead and close this issue.

closed 2019-03-09 by Strangedave