betaflight/betaflight #7745: BF4 - MATEK F722 STD motor 4 not spinning

ExaflopS, 2019-03-09:
**Describe the bug**
After installing BF4.0 dev build 1500 and blheli FW 32.6.7 , with bidirectional dshot and rpm filters enabled, motor 4 / esc 4 sometimes dont initialize. There is needet to power cycle the quadcopter, sometimes even 3 or 4 times power cycling in a row, the 4th motor dont initialize.
Note this bug is present only with bidirectional dshot enabled.

**To Reproduce**
install BF4.0 dev version 1500 and blheli FW 32.6.7 , enable bidirectional dshot and rpm filters.

**Expected behavior**
After connecting battery, esc 4 dont initialize, so after arming motor 4 does not spin.

**Flight controller configuration**

```# diff all

# version
# Betaflight / MATEKF722 (MKF7) 4.0.0 Mar  8 2019 / 11:37:03 (52c4e96c8) MSP API: 1.41

board_name MATEKF722
manufacturer_id 
mcu_id 002400323237510b34363730
signature 

# reset configuration to default settings
defaults nosave

# start the command batch
batch start

# name
name CHAMELEON TI

# resources
resource MOTOR 1 C09
resource MOTOR 2 C08
resource MOTOR 3 C07
resource MOTOR 4 C06

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper
beeper -ON_USB

# beacon

# map
map TAER1234

# serial
serial 2 32 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 2 1750 2100 0 0
aux 2 13 2 1300 1600 0 0
aux 3 26 0 1300 2100 0 0
aux 4 35 3 1725 2100 0 0
aux 5 49 3 1250 1600 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h
rxfail 6 s 1500

# display_name

# master
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 0
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_min_hz = 200
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 0
set acc_calibration = 8,25,-4
set rssi_channel = 6
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 10
set align_board_yaw = 180
set bat_capacity = 1000
set vbat_warning_cell_voltage = 340
set vbat_lpf_period = 100
set ibat_lpf_period = 25
set yaw_motors_reversed = ON
set pid_process_denom = 1
set thrust_linear = 40
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 55
set osd_cap_alarm = 1000
set osd_rssi_pos = 2049
set osd_current_pos = 2368
set osd_mah_drawn_pos = 2402
set osd_altitude_pos = 2071
set osd_warnings_pos = 2058
set osd_stat_tim_1 = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 4
set vtx_freq = 5769

# profile
profile 0

set dyn_lpf_dterm_max_hz = 300
set dterm_lowpass2_hz = 0
set iterm_rotation = ON
set iterm_relax_type = GYRO
set p_pitch = 42
set i_pitch = 110
set d_pitch = 36
set f_pitch = 95
set p_roll = 38
set i_roll = 105
set d_roll = 33
set f_roll = 90
set p_yaw = 30
set i_yaw = 110
set f_yaw = 40
set abs_control_gain = 0
set d_min_roll = 0
set d_min_pitch = 0
set d_min_boost_gain = 40
set d_min_advance = 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 195
set pitch_rc_rate = 195
set yaw_rc_rate = 195
set roll_expo = 70
set pitch_expo = 70
set yaw_expo = 70
set roll_srate = 62
set pitch_srate = 62
set yaw_srate = 62
set tpa_rate = 80
set tpa_breakpoint = 1850

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```
**Setup / Versions**
 - Flight controller MATEK F722 STD
 - Other components 
   RX R9mini,
   VTX: TBS Unify PRO HV, 
   ESC: HOLYBRO TEKKO 32 35A, FW: 32.6.7

mikeller, 2019-03-09:
@ExaflopS:

> set dshot_bidir = ON

Try `dshot_bidir = off`. If everything works then you will need a special resource configuration for bidirectional Dshot to work, and this is a known limitation of the current implementation of bidirectional Dshot, and this issue should be closed.

ExaflopS, 2019-03-09:
Thank You for response- yes, when I turn off dshot bidir, everything works.

closed 2019-03-09 by ExaflopS

ExaflopS, 2019-03-11:
@mikeller 
I did testing and when using 4k4k with dshot1200 or 4k4k with dshot600 everything is ok.
Also, I have checked multiple times status in the cli while connected battery and when using 8k8k gyro rate seemd to be sometimes more than 11% off, like it should be 8000 ideally, but sometimes it is dropping to 7100ish.
when using 4k4k, gyro rate stays most time at 4000 and only sometimes it is 3968 or 4016, which is less than 1% error