betaflight/betaflight #7748: Altitude estimation biased towards GPS in BARO+GPS configurations

qavpilot, 2019-03-09:
**Describe the bug**
Current calculateEstimatedAltitude() call (3.5.6) handling altitude estimation in position.c seem to be biased towards GPS based altitude instead of preferring barometer. The bias is due to high gpsTrust value (0.3 - 0.9) used to compute estimatedAltitudeCm in scenario where both barometer and GPS altitude are available. As a result FC uses rapidly changing altitude (HDOP value reported makes gpsTrust change quite rapidly even in good sat. reception conditions) despite quad flying almost level.
This impacts both overall altitude reporting as well as GPS rescue operation.

**To Reproduce**
1. Get quad with baro+GPS HW
2. Enable Altitude debug (set debug_mode = ALTITUDE) and enable blackbox
3. Power up quad with baro+gps HW on a ground and let it get good fix
4. Arm the quad
5. Place quad on 2m column and let is stay there for 60s
6. Open the trace with blackbox explorer and observe abrupt changes in gpsTrust values

**Expected behavior**
1. Using either VDOP from GSA (NMEA) or vAcc from NAV-POSLLH (uBLOX) to compute gpsTrust
2. Making gpsTrust at most 10% of estimated alt. value or dropping GPS altitude from alt. calculation altogether due to its fluctuations (unless there is a desire to use much more sophisticated algorithm to fuse GPS and Baro readings like http://www.crocodile.org/lord/baroaltitude.pdf)

**Flight controller configuration**
```# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.5.6 Feb 16 2019 / 13:29:46 (c28ca22f9) MSP API: 1.40

board_name OMNIBUSF4SD
manufacturer_id 

# name
name RAF220

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS

# beeper
beeper -DISARMING
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# beacon

# map

# serial
serial 0 2 115200 115200 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 3 1950 2050 0 0
aux 1 1 1 950 1050 0 0
aux 2 2 1 1425 1575 0 0
aux 3 46 0 1950 2050 0 0
aux 4 13 5 1950 2050 0 0
aux 5 28 1 1950 2050 0 0
aux 6 31 2 1950 2050 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_trim_pitch = 6
set baro_hardware = BMP280
set rssi_channel = 11
set serialrx_provider = IBUS
set rx_min_usec = 996
set min_throttle = 1019
set motor_pwm_protocol = ONESHOT125
set motor_pwm_rate = 2000
set failsafe_throttle = 1120
set failsafe_procedure = GPS-RESCUE
set vbat_max_cell_voltage = 42
set vbat_scale = 112
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_rescue_descent_dist = 50
set gps_rescue_ground_speed = 700
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set osd_vbat_pos = 2103
set osd_rssi_pos = 2137
set osd_flymode_pos = 2476
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_gps_speed_pos = 2233
set osd_gps_lon_pos = 2497
set osd_gps_lat_pos = 2529
set osd_gps_sats_pos = 2520
set osd_home_dir_pos = 2093
set osd_home_dist_pos = 2094
set osd_altitude_pos = 2198
set osd_warnings_pos = 2545
set osd_battery_usage_pos = 49
set osd_disarmed_pos = 2442
set osd_stat_max_alt = ON
set debug_mode = ALTITUDE
set vtx_band = 1
set vtx_power = 2
set vtx_freq = 5865
set vcd_video_system = PAL

# profile
profile 0

# rateprofile
rateprofile 0
```

**Setup / Versions**
OMNIBUSF4SD  (Omnibus F4 Pro V3)

laz2wiringpi, 2019-03-16:
i have made a test build . to log  HDOP VACC GPSALT BAROALT in blackbox.
 
and while your correct in the test on a bench . its not quite the same in flight.
i do have a concern with the logic for offset when armed  however its simple to see the alt is off in the osd and if the pilot has a concern... then land. 
 
Note: found  the baro drops do to wind pressure ? when arming.

qavpilot, 2019-03-22:
Can you please share your blackbox log ?
Wanted to compare dynamic behavior on altitude with my case. I have baro sensor covered with fluffy material to avoid direct wind getting into sensor hole, so for me even a active case the GPS provides mode jittery readings

laz2wiringpi, 2019-03-22:
yea , i have to dig them up ...

with an alt its not just wind , but also temp .. as you sit there it warms up .. and when you take off it cools down i dont think the hardware / software takes this into account. 

what kind of issue are you having .  my logs will not help with your readings .. 
and the hack i did was just that a hack on a junk test quad i dont mind losing if i messed up the code .

currently im using baro  and not gpsalt in my test build . while i figure out a improvement to the current code .. i have some ideas but little time.....

do you use beta flight slack ?

laz2wiringpi, 2019-03-22:
https://www.youtube.com/watch?v=vfu-APt6Wd8&feature=youtu.be

my son did a test flight for me.
vtx is bad ..   test junk quad of spare parts... 
the debug values are
HDOP VACC GPSALT BAROALT - GPS BN-180 ublox , nema has bad  support for VACC  
the alt in the top right is the baro NOT gpsalt

qavpilot, 2019-03-23:
You are right on all these corrections (temp & wind) but gps error alone is order of magnitude larger than theses correction, plus it has a tendency to cause abrupt changes to altitude due to sats appearing/leaving the area.
This is major concern - the altitude isn't consistent (I don't expect it to be centimeters accurate but current mix seem to be couple of meters off).
Thanks for the video - it shows exactly same jumps and discrepancy between GPS and BARO I see.
Need to build private version that uses baro as main altitude as I didn't see runtime option to switch.
Will look into papers on fusing Baro with GPS, there are examples of good implementations alread :)
Thanks for confirming my observations

laz2wiringpi, 2019-03-23:
look at inav .. you may be reinventing the wheel ...

i don't think there using gps alt unless its acc is 2.5m  reported  i think....  never used it myself i  did look at the code .. 
but even then you can get garbage data and they do a lot of code  to detect garbage .. 

rescue is  just to get the quad back . and its still early in development .. 

Im also looking at cleaning up the alt ,  i think it's better to use the baro if you have it . an ignore the gpsalt

buniekwawa, 2019-04-24:
i've noticed similar issue.
today i found that place in code and change logic little bit. https://github.com/buniekwawa/betaflight/commit/e923916f259bc8c50b47d1718b5be31dfdbed352
i am going to deploy it to my kakute f7 and test it.

if everything will be fine i will create pull request.

laz2wiringpi, 2019-05-02:
@qavpilot @buniekwawa  did a option  for you 
i think you can close 
look at 
https://github.com/betaflight/betaflight/issues/5890

closed 2019-05-02 by qavpilot