betaflight/betaflight #7750: Smart Audio does not work in 4.0 RC1

KamilKaczy, 2019-03-09:
**Describe the bug**
Hello! That's my first post on Github, so hello everyone, You are doing amazing job :) I don't know anything about programming, but i hope that by letting you know about that bug will contrubute to fixing it. Smart Audio refuses to work with new Betaflight 4.0 Release Candidate 1, when previously it was working flawlessly with BF 4.0 1440 Build. I tried everything.  I was not changing anything in hardware or wiring, between the updates. My FC is Omnibus F4 pro v2, VTX is Holybro Atlatl HV V2, connected via hardware UART (TX3). I flashed back to previous firmware (BF 4.0 1440 build) and also tried 1442 build, and after i paste my "diff all" in CLI, on both these builds it works right away. But when i flashed again to 4.0 RC1, same thing, on OSD, VTX SA menu there is only "stats" entry, and only "sent" packets count goes from 1 to 10, and goes back to 1.
On UART1 i have RX(SBUS/Frsky R9 Slim+)
on UART 3 SmartAudio
on UART 6 GPS M8N
on Softserial 1 SmartPort telemetry

UPDATE:
OK, i spend an hour trying to flash different firmwares, and my conclusion is: on firmware 4.0 build 1444 and below (all older) Smart Audio is working with no problems, on my setup at least. And on firmware 4.0 build 1446 and above, all the way to Release Candidate 1 - is not working at all. There is only stats submenu available. So it's either 1445 or 1446 build that had some code change that is causing this. I suppose it's 1445. I hope that helps. 

**To Reproduce**
Flash RC1 on OMNIBUSF4SD and configure SA using TX3 and Atlatl HV V2 VTX

**Expected behavior**
Being able to change power and frequency via OSD

**Flight controller configuration**
``` # diff all

# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Feb 17 2019 / 03:11:01 (da7608f51) MSP API: 1.41

board_name OMNIBUSF4SD
manufacturer_id 
mcu_id 002800393037470439333631
signature 
# reset configuration to default settings
defaults nosave

# name
name GT 220 R

# display_name

# resources
resource MOTOR 5 NONE
resource SONAR_TRIGGER 1 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 A01

# dmaopt

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature TELEMETRY

# beeper
beeper -RX_LOST
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 2 115200 38400 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1600 2100 0 0
aux 1 1 1 1675 2100 0 0
aux 2 2 1 1250 1675 0 0
aux 3 46 2 1125 1275 0 0
aux 4 13 2 1725 1925 0 0
aux 5 35 2 1925 2100 0 0

# adjrange

# rxrange

# vtx

# vtxtable

# rxfail
rxfail 2 s 1450
rxfail 3 s 1225
rxfail 5 s 1900
rxfail 7 s 1000

# master
set gyro_lowpass2_type = BIQUAD
set gyro_lowpass2_hz = 150
set gyro_calib_duration = 200
set dyn_lpf_gyro_max_hz = 580
set acc_calibration = 3,51,210
set mag_hardware = NONE
set min_check = 1005
set max_check = 1995
set rssi_channel = 8
set fpv_mix_degrees = 30
set serialrx_provider = SBUS
set airmode_start_throttle_percent = 35
set rx_min_usec = 876
set rx_max_usec = 2120
set blackbox_p_ratio = 64
set min_throttle = 1008
set dshot_idle_value = 330
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 25
set failsafe_off_delay = 20
set failsafe_throttle = 1170
set failsafe_throttle_low_delay = 666 ###CORRUPTED CONFIG###
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1300
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 340
set ibata_scale = 203
set ibata_offset = -400
set beeper_dshot_beacon_tone = 4
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_angle = 45
set gps_rescue_initial_alt = 40
set gps_rescue_descent_dist = 30
set gps_rescue_ground_speed = 2500
set gps_rescue_throttle_min = 1180
set gps_rescue_throttle_max = 1500
set gps_rescue_throttle_hover = 1230
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_use_mag = OFF
set deadband = 1
set pid_process_denom = 1
set thrust_linear = 50
set osd_warn_arming_disable = OFF
set osd_warn_batt_not_full = OFF
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 47
set osd_cap_alarm = 950
set osd_alt_alarm = 130
set osd_tim2 = 769
set osd_vbat_pos = 14411
set osd_rssi_pos = 14502
set osd_link_quality_pos = 2248
set osd_tim_2_pos = 14451
set osd_g_force_pos = 468
set osd_throttle_pos = 14592
set osd_crosshairs_pos = 270
set osd_current_pos = 14483
set osd_mah_drawn_pos = 14515
set osd_motor_diag_pos = 2339
set osd_craft_name_pos = 488
set osd_gps_lon_pos = 2497
set osd_gps_lat_pos = 2465
set osd_gps_sats_pos = 14469
set osd_home_dir_pos = 14639
set osd_home_dist_pos = 14670
set osd_altitude_pos = 14678
set osd_pid_roll_pos = 414
set osd_pid_pitch_pos = 447
set osd_pid_yaw_pos = 479
set osd_debug_pos = 224
set osd_power_pos = 114
set osd_warnings_pos = 14730
set osd_avg_cell_voltage_pos = 14443
set osd_battery_usage_pos = 458
set osd_nvario_pos = 14711
set osd_esc_rpm_pos = 355
set osd_stat_max_dist = ON
set osd_stat_min_batt = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_min_link_quality = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 2
set vtx_channel = 5
set vtx_freq = 5809
set rcdevice_init_dev_attempts = 4

# profile
profile 0

set dyn_lpf_dterm_max_hz = 270
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set anti_gravity_threshold = 20
set iterm_relax = RPY
set iterm_relax_type = GYRO
set yaw_lowpass_hz = 120
set throttle_boost = 18
set acro_trainer_angle_limit = 80
set p_pitch = 44
set f_pitch = 80
set p_roll = 40
set f_roll = 80
set i_yaw = 90
set f_yaw = 90
set level_limit = 61

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 131
set pitch_rc_rate = 131
set yaw_rc_rate = 131
set roll_expo = 21
set pitch_expo = 21
set yaw_expo = 21
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 72
set tpa_rate = 0
set tpa_breakpoint = 1400

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 2

# save configuration
save
#  ```

**Setup / Versions**
FC- Omnibus F4 Pro V2
Vtx- Holybro Atlatl HV V2

podgeme, 2019-03-10:
lol, ive noticed this issue for the last few weeks with both my quads that have a holybro atlatl where smart audio suddenly broke when i moved from BF 3 to BF 4.0 but i hadn't got round to opening an issue and locating which nightly build it stopped on. ive just finished spending an hour locating the build and then some more time testing various uart arrangements and them find KamilKaczy has already located the exact build in the last 24 hours since i last looked! well done by the way i know having done it myself that its quite tedious!

anyways i have tested using 2 different holybro atlatl v2.3 vtx's. i can confirm that there was no issue on nightly build 1444 and then the issue developed on build 1445. all of my tests were done on stock betaflight with a full chip erase when moving between fw versions then i simply setup the receiver so that i could navigate to the osd and obviously the smartaudio protocol itself but apart from that i touched nothing. i tested using an omnibusf4sd on uart 1 where it worked on build 1444 but not 1445 (when i say not working it simply shows "statx" in the osd and any channel changes done through the command line fail to be applied to the vtx), i then also tested using an omnibusf4V6 connecting the holybro atlatl smart audio to uart 1,2,3 and 6 all of which displayed the same result (working on 1444, broken on 1445). i suspect this issue is isolated to the atlatl specifically but clearly there was something changed in betaflight that caused the problem.

to summarise-
holybro atlatl v2 (both of mine labelled v2.3) smart audio not communicating on build 1445 onwards
tested on omnibusF4SD uart 1 and omnibusF4V6 all available TX uarts, all produce the same issue.

hope this helps, great job as always betaflight devs, loving the rpm filter its awesome!
if anything was unclear or there's any more info i can provide I'm more than happy to do so.

podgeme, 2019-03-10:
ive also just tested the atlatl v2.3 vtx on a dys f4 pro v1 with it also not working on build 1445 and above for all uarts (1,3, and 6).

laz2wiringpi, 2019-03-12:
#7603 
Send 0x00 bit(s) (always) even with Soft Serial. #7603

fujin, 2019-03-13:
Could someone test if the problem presents on 3.5.7 as well?

SebiWolze, 2019-03-13:
I confirm that bug with 4.0 RC1 
Bardwell F4 V2 + Unify Race V2 MMCX here.

podgeme, 2019-03-13:
not sure what you mean by betaflight 3.5.7? Just tested for the bug on build 1052 which as far as i can tell is the last betaflight 3.5.X version. it was not present SA working fine. the bug is also not there on BF 3.5.6 stable release. as far as i can tell the bug appears on build 1445 and is still present up to now (build 1514). i also tried using a pull down resistor on the SA line in case the problem is similar to that in #7603 but with no success.

laz2wiringpi, 2019-03-14:
can someone test i do not have the vtx
https://github.com/laz2wiringpi/betaflight/tree/test_smartaudio/obj

PatrykSokol, 2019-03-14:
I've just tested on Holybro Kakute F4 v2 and Holybro Atlatl V2 HV and it works just fine :)

podgeme, 2019-03-14:
laz2wiringpi-
tested your build on omnibusF4SD and Atlatl v2.3, no bug working perfectly.

laz2wiringpi, 2019-03-14:
@PatrykSokol it woks fine with what my hex ,latest 4.0 ?

laz2wiringpi, 2019-03-14:
@podgeme dam ... 
that may going to be an issue . in production
@mikeller  you have more knowledge than i do with the internals  any ideas of how this may be resolved i hate to see another branch just for broken vtx's again :(

PatrykSokol, 2019-03-14:
@laz2wiringpi  - It works fine with your binary

Sil20, 2019-03-15:
Hi!
I have the same issue using a Hobbywing XRotot F4 and a Atalatl HV V2.2. I did several Tests:
Betaflight 3.5.0 => OK
Betaflight 3.5.6 => OK
Betaflight 4.0.0 First build => KO
**Custom hex 4.0.0 from @laz2wiringpi=> OK but the first time the VTX information display whas wrong** See attached snapshot
[image]
[image]
[image]
[image]

jamesbroom, 2019-03-17:
Just confirmed this issue with Kakute F7 and Holybro Atlatl V2 HV v2.3.

4.0.0#1444 can see the VTX.  Can configure VTX from OSD and from Spektrum radio.  Telemetry back to Spektrum radio correctly shows current VTX settings.

4.0.0#1445 and up (tried #1524) cannot see the VTX at all.

3.5.8 sees the VTX and everything works as expected.

codecae, 2019-03-17:
Please see #7823 for a possible workaround.  Would like to get some testing input.

Sil20, 2019-03-20:
> Please see #7823 for a possible workaround. Would like to get some testing input.
Hi, I think  this specific issue  is not related to soft serial. My previous tests are done using a hardware UART.

codecae, 2019-03-20:
@sil20 there are also fixes for this issue as well.

Sil20, 2019-03-22:
OK I tried using My setup, and the SA is not working either. So far only the custom hex 4.0.0 from @laz2wiringpi is working for me.

laz2wiringpi, 2019-03-22:
@Sil20  sorry i had to pull the stale builds .  i found an issue with my build machine on win 10 and shared windows file system. 

@codecae 
it got to be as i ref before the line  where another 0x00 was added in the lite fix

laz2wiringpi, 2019-03-23:
@fujin i dont think this it will work. you need the x00 correct?
@Sil20  
or anyone with an issue with SA . broke in 4.0.0 

https://github.com/laz2wiringpi/BetaflightSA/releases

does it work .. and what vtx - FC do you have

KamilKaczy, 2019-03-24:
@laz2wiringpi
I tested your build and it works, Smart Audio is fully functional. My FC is Omnibus F4 Pro V2, VTX is Holybro Atlatl V2, specifically V2.3.
Is that possible that Smart Audio and Smart Audio Lite protocols are mutually uncompatible, at least on certain VTXs? Perhaps user could choose using CLI command if send that extra 0x00 bit or not, depending whether user is using SA Lite or not. Or would that be too complicated?

Sil20, 2019-03-24:
> @fujin i dont think this it will work. you need the x00 correct?
> @Sil20
> or anyone with an issue with SA . broke in 4.0.0
> 
> https://github.com/laz2wiringpi/BetaflightSA/releases
> 
> does it work .. and what vtx - FC do you have

I juste tried And it works!!
I have a Hobbywing XRotor F4 (omnibusF4SD) and a ATALATL HV V2 (V2.2 on the VTX) Smart audio is connneced to a hardware  UART n°6

Few points though
* How do I know if the VTX uses SA lite or regular?
* It takes ~30seconds for the smart audio to work
* There is a serie of number on the SA OSD page:
[image]
* On the SA state OSD page I see the status goind automatically to OnlyV2. Is there a way to force it to Only V2 to save Time?

laz2wiringpi, 2019-03-24:
@KamilKaczy this has been a long standing issue with tx from way back. see branches for akk_vtx_fix 3.3,3.4,3.5 - however all my akk's have worked and howver i dont need the fix and it works in both builds . do to a fix by the manufacture.

IMO,  you may be out of luck . until 4.1 or higher without a custom build.,  The  serial would need  a rework  .  i  am just   attempting to find out just how many cloned tbs SA ore effected  as the genuine tbs protocol like the  lite now requires this to my understanding.

How to get a the cli command readable at this function's memory and not cause other issues .  i have seen quite a few pr's but none have been approved. some need a pull down , some need a pull after frame. 

 The manufacture's have made quite a mess  to clean up after them. (:

have you contacted Holybro ? maybe you can get firmware for the vtx that is fixed , like akk did  on new units .

Sil20, 2019-03-26:
I tried to contact Holybro few month ago regarding the delay for the SA to  work and did not get a clear answer. I contact hem also mentionning this thread few days ago,no answer so far.

laz2wiringpi, 2019-03-26:
this has  been reverted  fixed in 4.0.0 rc4
build 1542+
 thank @fujin he is sacrificing his vtx .. lite code  , hopefully he can get it  fixed in 4.1
please test and ask to be closed if it works

mikeller, 2019-03-26:
@KamilKaczy, others: Can you please test to see if this has been fixed with 4.0.0-RC4?

jamesbroom, 2019-03-26:
Works fine on my Atlatl V2.

RC4 with configurator 10.5.

Visually verified that the OSD stick commands do indeed change the VTX settings.

Sil20, 2019-03-27:
Tired this morning: 4.0.0.RC4 working for me !!
Thanckyou!

dcool8, 2019-03-27:
Tried it today on two quads and it worked fine for me.

KamilKaczy, 2019-03-27:
Smart Audio in 4.0 RC4 works for ma also, thank you very much @fujin, @laz2wiringpi, @mikeller and all others for your commitment and for making Betaflight such an amazing piece of software!

closed 2019-03-28 by KamilKaczy

binlan, 2019-04-07:
was working for me in RC4 as well, but in RC6 is broken again.
OMNIBUSF4SD
TBS Unify Pro