betaflight/betaflight #7756: Values of Roll, Pitch, Yaw, Throttle does not change using MSP but AUX1-4 changes

monomik, 2019-03-10:
I am using Multiwii Serial Protocol to use my keyboard as remote controller. When I use MSP_SET_RAW_RC to write values of RC Channels (Roll, Pitch, Yaw, Throttle, AUX1, AUX2, AUX3, AUX4) values for AUX1-4 are changed but values for Roll, Pitch, Yaw, Throttle remain unchanged. The same method works fine with Baseflight. 

To send RC Channel values using Python 3:
```python3
import serial
import struct

ser = serial.Serial("/dev/ttyUSB0", 115200) #establishing serial connection

data = [1200, 1300, 1400, 1500, 1330, 1440, 1550, 1660] #RC channel values to be set

csum = 0 #checksum
new_data = [b'$', b'M', b'<', 16, 200] + data #complete data to be sent
for i in struct.pack('<2B8H', *new_data[3:len(new_data)]):
    csum = csum ^ i #bitwise XOR for calculating checksum
new_data.append(csum)
ser.write(struct.pack('<3c2B8HB', *new_data))
```

To receive RC Channel values using Python 3:
``` python3
import serial
import struct

ser = serial.Serial("/dev/ttyUSB0", 115200)

# requesting RC channel values
data = [] 
csum = 0
new_data = [b'$', b'M', b'<', 0, 105] + data
for i in struct.pack('<2B', *new_data[3:len(new_data)]):
    csum = csum ^ i
new_data.append(csum)
ser.write(struct.pack('<3c3B', *new_data))

# receiving the response
while True:
    header = ser.read()
    if header == b'$': 
        header = header + ser.read(2)
        dlength = struct.unpack('<b', ser.read())[0] #size of payload = 36(here) AETR + 1-14AUX
        code = struct.unpack('<b', ser.read())[0] #type of payload, 105 for MSP_RC
        data = ser.read(dlength) #payload
        temp = struct.unpack('<18H', data)
        print(temp) 
        break
```

I am using NAZE 32 which is directly connected to computer via USB (/dev/ttyUSB0). Customized version of Betaflight 4.0.0 was flashed for NAZE using [https://github.com/betaflight/betaflight/blob/master/docs/Customized%20Version.md](url) in order to enable Serial_RX and RX_MSP. The problem also persists with Cleanflight 2.1.0 (default version). 

Output for `diff`
```
# diff
# Betaflight / NAZE (AFNA) 4.0.0 Mar  9 2019 / 17:17:46 () MSP API: 1.41

feature -RX_PPM
feature -AIRMODE
feature -ANTI_GRAVITY
feature -DYNAMIC_FILTER
feature MOTOR_STOP
feature RX_MSP
serial 1 64 115200 57600 0 115200
aux 0 0 0 1400 1600 0 0
set min_throttle = 1150
set max_throttle = 1850
set motor_pwm_protocol = OFF
set pid_process_denom = 2
profile 0

rateprofile 0
```

mikeller, 2019-03-10:
@monomik: NAZE is unsupported in Betaflight. Try doing this with a supported flight controller - RX_MSP works fine on supported flight controllers.

monomik, 2019-03-10:
@mikeller: I have also ordered an Omnibus F4, will let you know the results as soon as I get it.

monomik, 2019-03-22:
@mikeller: I found the solution to this. In **Failsafe > Channel Fallback Settings > Roll/Pitch/Yaw/Throttle** is set to **AUTO**. When we change it to HOLD, it works fine. I also tried this on Omnibus F4 and it worked.

I have another question:
I am trying to control my quadcopter using an on-board computer that communicates with the flight controller (Omnibus F4) using serial connection. Everything is done via MSP, there is no physical receiver connected to the FC. When I try to arm the quad by using AUX1, it does not arm. Is there a way to disable any precautionary check (likely: absence of receiver) and arm the quad as soon as AUX1 is in defined range?

mikeller, 2019-03-23:
@monomik: Looks like you are triggering a failsafe event before the flight controller is receiving valid MSP channel settings. Since RX_MSP is designed for testing only as it lacks most of the safeguards that all of the proper RX protocols implement this should not be an issue - just keep testing with failsafe on.

closed 2019-04-29 by stale[bot]