betaflight/betaflight #7758: Compiling BF for SPRACINGF3 adding GPS

u3dreal, 2019-03-10:
Hello all

I'm trying to compile BF with GPS omitting dshot and others.

i added 

#undef USE_DSHOT
#undef USE_OSD
#undef USE_ESC_SENSOR
#undef USE_ESC_TELEMETRY
#undef USE_RANGEFINDER  
#undef USE_PPM
#undef USE_SERIALRX_CRSF

#define USE_GPS
#define USE_GPS_UBLOX

to the target.h header

things compile fine and the size is small enough for the F3

But in BF i can't get GPS working.
Configured on UART2 i can make the GPS dialog show up in the gui.
The GPS tab is also there.

But the GPS icon on the top doesn't not want to come on.
And no data in GPS tab.

I tried everything every UART every baudrate , port + softserial.
The GPS works fine in u-center and i configured it to use 57600 baud as default.

The GPS does not show up.

Did i miss some important defines to make it working ?

help would be greatly apprechiated.

regards 

u3dreal

mikeller, 2019-03-10:
@u3dreal:

You supplying the information you are asked to provide when opening an issue would be greatly appreciated.

u3dreal, 2019-03-10:
@mikeller 

Compiling betaflight git with target SPRACINGF3.

Sorry what did i miss ?? This is my first time here Sorry.

mikeller, 2019-03-10:
@u3dreal:

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

u3dreal, 2019-03-10:
OK here is the diff

```
# diff
# Betaflight / SPRACINGF3 (SRF3) 4.0.0 Mar 10 2019 / 18:15:33 (9c4e750f3) MSP API: 1.41 / FEATURE CUT LEVEL 4

batch start
name DrQ
resource MOTOR 7 NONE
resource SERIAL_TX 2 A02

feature -RX_PPM
feature -RSSI_ADC
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
map RETA1234
serial 1 64 115200 57600 0 115200
serial 30 2 115200 57600 0 115200
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 16
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set min_throttle = 1000
set use_unsynced_pwm = ON
set motor_pwm_protocol = MULTISHOT
set motor_pwm_rate = 16000
set align_board_yaw = 90
set bat_capacity = 1400
set current_meter = VIRTUAL
set vbat_scale = 101
set ibatv_scale = 350
set ibatv_offset = 60
set gps_provider = UBLOX
set deadband = 2
set yaw_deadband = 3
profile 0

rateprofile 0
```

mikeller, 2019-03-11:
Tested with SPRACINGF3, with your target configuration, and the following settings work just fine for me with the GPS on UART3:

```
# Betaflight / SPRACINGF3 (SRF3) 4.0.0 Mar 11 2019 / 23:11:42 (2eb0a578a) MSP API: 1.41 / FEATURE CUT LEVEL 4
batch start
defaults nosave
mcu_id 001b00335832570a20363436

feature -RX_PPM
feature GPS
serial 2 2 115200 57600 0 115200
set gps_provider = UBLOX
```

Maybe try to reset the settings on your GPS unit to defaults?

u3dreal, 2019-03-11:
Hi Mike. I already have the GPS reset and set like in the wiki.

OK I'll try again on UART 3 

If not my board is toasted.

Thanks alot !!

u3dreal, 2019-03-11:
OK found that TX 3 was also toasted. TX 2 was as well.
But thanks to betaflight there is remapping.

Now works on Motor 7 + 8 

Thanks again for the help sorting this out

closed 2019-03-11 by u3dreal