betaflight/betaflight #7759: BF 4.0 - CLRacing F4 motors 1 and 2 not spinning

RadomirJ, 2019-03-10:
**Describe the bug**
After installing BF4.0 dev build 1504 and RC1 too and blheli FW 32.6.7 , with bidirectional dshot and rpm filters enabled, motor 1 and motor 2 doesn`t spin. 
When I disable  dshot_bidir all motors spin correctly.

**To Reproduce**
install BF4.0 dev version 1504 and blheli FW 32.6.7 , enable bidirectional dshot and rpm filters.

**Expected behavior**
After arming motor 1 and 2 does`t spin.

**Flight controller configuration**
```
# diff all

# version
# Betaflight / CLRACINGF4 (CLR4) 4.0.0 Mar 10 2019 / 10:18:21 (95c6b52c8) MSP API: 1.41

board_name CLRACINGF4
manufacturer_id 
mcu_id 005000295848500f20323852
signature 

# reset configuration to default settings
defaults nosave

# start the command batch
batch start

# name
name chamTi

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature TELEMETRY

# beeper

# beacon

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1600 2100 0 0
aux 1 13 1 1600 2100 0 0
aux 2 35 3 1600 2100 0 0
aux 3 49 2 1600 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_max_hz = 575
set acc_calibration = -7,87,-56
set rssi_channel = 16
set serialrx_provider = SBUS
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set ibata_scale = 188
set yaw_motors_reversed = ON
set deadband = 4
set yaw_deadband = 4
set pid_process_denom = 1
set thrust_linear = 40
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 40
set osd_cap_alarm = 1100
set osd_vbat_pos = 2465
set osd_rssi_pos = 2081
set osd_vtx_channel_pos = 2103
set osd_current_pos = 2486
set osd_mah_drawn_pos = 2476
set osd_battery_usage_pos = 2089
set osd_stat_rtc_date_time = ON
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_channel = 8
set vtx_freq = 5880

# profile
profile 0

set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set iterm_rotation = ON
set iterm_relax_type = GYRO
set p_pitch = 42
set i_pitch = 110
set d_pitch = 36
set f_pitch = 95
set p_roll = 38
set i_roll = 105
set d_roll = 33
set f_roll = 90
set p_yaw = 30
set i_yaw = 110
set f_yaw = 40
set d_min_roll = 27
set d_min_pitch = 0
set d_min_boost_gain = 35

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 140
set pitch_rc_rate = 140
set yaw_rc_rate = 120
set roll_expo = 19
set pitch_expo = 19
set yaw_expo = 25
set roll_srate = 74
set pitch_srate = 74
set yaw_srate = 72
set tpa_rate = 80
set tpa_breakpoint = 1850

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

**Setup / Versions**

Flight controller CL racing F4 AIO
Other components
RX R9mini,
VTX: Tramp HV,
ESC: Spedix GS30A, FW: 32.6.7

netzwergracer, 2019-03-11:
I can confirm this exact same issue on CLRACINGF4 board. Motor 1 and 2 are not spinning on BLHeli32 bidir.

I would like to add:
When reading settings in BLHeliSuite32, Motor 1 and 2 show "input motor protocol: unknown" whereas Motor 3 and 4 show "input motor protocol: DShot" as it should.

mikeller, 2019-03-11:
@RadomirJ: Can you please test if this problem exists with bidirectional Dshot disabled? If not, then this should be moved into https://github.com/betaflight/bidircfg/issues.

netzwergracer, 2019-03-11:
I'm not RadomirJ, but the problem does not exist with dshot_bidir=off on CLRACINGF4.
With bidirectional Dshot disabled, all 4 motors are spinning.

RadomirJ, 2019-03-11:
I confirm comment from netzwergracer , All motors spinnig correctly after enable command dshot_bidir=off.

supiiik, 2019-03-11:
Same problem here :) CLracing F4s
With set dshot_burst=off motors 1 and 2 are not visible even in blheli
with set dshot_burst=on is everything ok (but bidirectional DSHOT not working)

supiiik, 2019-03-14:
Seems to be working fine now with BF 4.0 RC2

netzwergracer, 2019-03-14:
Yes I can confirm that, the new snippet works on latest BF 4.0 build.

joelucid, 2019-03-15:
Thanks guys that is good news!

joelucid, 2019-03-15:
@mikeller this is happening again with RC2. It looks like all configuration is correct but two motors don't spin. Here's output of timer:

```
# timer
timer B09 1
timer B00 1
timer B01 1
timer A03 0
timer A02 0
```

Here are the individual timer options:

```
# timer b00 list
# 0. TIM1 CH2
# 1. TIM3 CH3
# 2. TIM8 CH2

# timer b01 list
# 0. TIM1 CH3
# 1. TIM3 CH4
# 2. TIM8 CH3

# timer a03 list
# 0. TIM2 CH4
# 1. TIM5 CH4
# 2. TIM9 CH2

# timer a02 list
# 0. TIM2 CH3
# 1. TIM5 CH3
# 2. TIM9 CH1
```

This is the exact same as the static config which is known to work:

```
const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {

    DEF_TIM(TIM11, CH1,  PB9, TIM_USE_CAMERA_CONTROL,      0, 0), //   CAMERA_CONTROL_PIN

    DEF_TIM(TIM3,  CH3,  PB0, TIM_USE_MOTOR,               0, 0), // M1 - D2_ST6
    DEF_TIM(TIM3,  CH4,  PB1, TIM_USE_MOTOR,               0, 0), // M2_OUT D2_ST2
    DEF_TIM(TIM2,  CH4,  PA3, TIM_USE_MOTOR,               0, 1), // M3_OUT D1_ST6
    DEF_TIM(TIM2,  CH3,  PA2, TIM_USE_MOTOR,               0, 0), // M4_OUT D1_ST1

//    DEF_TIM(TIM3,  CH1,  PB4, TIM_USE_BEEPER,              0, 0), // BEEPER PWM
//    DEF_TIM(TIM4,  CH3,  PB8, TIM_USE_MOTOR | TIM_USE_LED, 0, 0), // LED & MOTOR5 D1_ST7

};
```

The dshot code needs to enable the used channel and how that's done differs by channel type (N channel different from normal channel). Could it be that the timer config commands don't propagate that info correctly?

joelucid, 2019-03-15:
Looking at the "normal" config for clracingf4:

```
const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {

    DEF_TIM(TIM11, CH1,  PB9, TIM_USE_CAMERA_CONTROL,      0, 0), //   CAMERA_CONTROL_PIN

    DEF_TIM(TIM1,  CH2N, PB0, TIM_USE_MOTOR,               0, 0), // M1 - D2_ST6
    DEF_TIM(TIM8,  CH3N, PB1, TIM_USE_MOTOR,               0, 0), // M2_OUT D2_ST2
    DEF_TIM(TIM2,  CH4,  PA3, TIM_USE_MOTOR,               0, 1), // M3_OUT D1_ST6
    DEF_TIM(TIM2,  CH3,  PA2, TIM_USE_MOTOR,               0, 0), // M4_OUT D1_ST1

    DEF_TIM(TIM3,  CH1,  PB4, TIM_USE_BEEPER,              0, 0), // BEEPER PWM
    DEF_TIM(TIM4,  CH3,  PB8, TIM_USE_MOTOR | TIM_USE_LED, 0, 0), // LED & MOTOR5 D1_ST7

};
```

This does show that two N channels get remapped to normal channels in the snippet. If these timers still have the N channel flag after importing the snippet one would expect two motors not to spin.

joelucid, 2019-03-15:
Looked at the code a bit. I'm having trouble understanding this:

```
const timerHardware_t *timerGetByTagAndIndex(ioTag_t ioTag, unsigned timerIndex)
{
    if (!ioTag) {
        return NULL;
    }

#if TIMER_CHANNEL_COUNT > 0
    uint8_t index = 1;
    for (unsigned i = 0; i < TIMER_CHANNEL_COUNT; i++) {
        if (TIMER_HARDWARE[i].tag == ioTag) {
            if (index == timerIndex || timerIndex == 0) {
                return &TIMER_HARDWARE[i];
            }
            index++;
        }
    }
#else
    UNUSED(timerIndex);
#endif

    return NULL;
}
```

Seems like this will return the first timer instance whether timerIndex is 0 or 1. Tried the obvious fixes but they made things even worse. Paging @mikeller ...

mikeller, 2019-03-15:
@joelucid: Not my code, I think this was added by @jflyper. But the timer index in `timerIOConfig()` is 1 based (0 being the default 'unconfigured'), so 0 should never be supplied as `timerIndex`. I wasn't sure if the original author had any intention of using 0 as 'give me any timer for this pin', so I left the `|| timerIndex == 0` in when extending this.

joelucid, 2019-03-15:
@mikeller does that mean that the index in the "timer pin" command should also start with 1?

mikeller, 2019-03-15:
No, it is offset before being read / written in CLI: 

https://github.com/betaflight/betaflight/blob/2db432a2b73dac7478324d5ac2a322a2e0ee6768/src/main/cli/cli.c#L5235

https://github.com/betaflight/betaflight/blob/2db432a2b73dac7478324d5ac2a322a2e0ee6768/src/main/cli/cli.c#L5163

jflyper, 2019-03-16:
I'm trying to recall what `timerGetByTagAndIndex` should do ...

mikeller, 2019-03-16:
@jflyper: It used to be `timerGetByTag()` I think, and then I refactored it to make it possible to get timers that are not registered in `timerIOConfig()`. So in retrospect the `timerIndex == 0` probably never made sense in the original form, as 
 `timerIOConfig()` is 1 based, and timer assignments with `index` 0 are not stored in it.

SebiWolze, 2019-03-16:
Contact me in Slack (sebiwolze) if you need anything tested. Got the Bardwell F4 AiO V2.

SebiWolze, 2019-03-18:
Just for your interest: It´s still not working with the bardwell F4 V2 - The target CLRacing F4 is the same, but something isn´t like in the normal Target.

supiiik, 2019-03-18:
Oh no, I just found, that it's still not working on CLracingF4 too. I somehow set it up with bidir=on and all motors was spining, but I forgot to check, if also RPM is received. Today I found, that there is 100% invalid packets from DSHOT :(

rabsfpv, 2019-03-19:
I had a problem with the same CLRACINGF4 target using the older Bardwell board (v1.1 it is called I think).  In order to get bidir working for it I had to free the resource B04 (beeper) and B08 (which I had on softserial SERIAL TX 11).  Once that was done I was able to successfully validate using timer/dma snippet. My ESC hex was 32.6.7 for Airbot_Wraith32_ST_Multi_32_67. 

Here's my current resource:
# resource
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource INVERTER 1 C00
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B09
resource CAMERA_CONTROL 1 B09
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C03
resource ADC_CURR 1 C01
resource SDCARD_CS 1 B12
resource SDCARD_DETECT 1 B07
resource FLASH_CS 1 B03
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05

and rpm bidir status: 
# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.30V, Core temp=41degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3017, Max available config: 16384
GYRO=MPU6000, ACC=MPU6000
System Uptime: 57 seconds, Current Time: 2019-03-19T01:14:39.920+00:00
CPU:21%, cycle time: 289, GYRO rate: 3460, RX rate: 110, System rate: 9
Voltage: 1552 * 0.01V (4S battery - OK)
I2C Errors: 0
SD card: None inserted
Arming disable flags: CLI
Dshot reads: 798916
Dshot invalid pkts: 0
Dshot irq micros: 3
Dshot RPM Motor 0: 244
Dshot RPM Motor 1: 252
Dshot RPM Motor 2: 246
Dshot RPM Motor 3: 253
7 7 7 7 7 7 7 7 7 7 7 7 7 7 7 7 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

gvhlaw, 2019-03-19:
I can confirm that I had the same problem on Bardwell’s CLRacingF4 V2 - Motors 1 and 2 did not spin with bidir on and burst off.  This was on 32.67.  I flashed RC2, pasted my diff all, and then applied the Snippet - everything works great with bidir on or off.  We discovered that a there is a reset all function in a “diff all” such that pasting it after making the timer and dma changes causes problems.  I also reverted to 32.6 without saving configurations, reconfigured the ESC’S, then flashed 32.65.  No more problems.  Other pilots experienced the same problem when pasting a diff after pasting the snippet - and 32.67 may be part of the problem as well.

supiiik, 2019-03-19:
Still not working. I have also did free B04 and B08, but the result is same
```
# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.27V, Core temp=46degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3017, Max available config: 16384
GYRO=MPU6000, ACC=MPU6000
System Uptime: 55 seconds, Current Time: 2019-03-19T08:50:48.090+00:00
CPU:14%, cycle time: 249, GYRO rate: 4016, RX rate: 33, System rate: 9
Voltage: 1181 * 0.01V (3S battery - OK)
I2C Errors: 0
SD card: None inserted
Arming disable flags: CLI
Dshot reads: 25
Dshot invalid pkts: 25
Dshot irq micros: 2
Dshot RPM Motor 0: 0
Dshot RPM Motor 1: 0
Dshot RPM Motor 2: 0
Dshot RPM Motor 3: 0
7 7 7 7 14 14 14 14 14 14 7 7 7 7 14 14 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
```

gvhlaw, 2019-03-19:
@supiik - try plugging the battery into the quad before you connect it to Betaflight and see if that makes a difference.  Keep your props off.

SebiWolze, 2019-03-20:
Has somebody found a real workaround yet?

netzwergracer, 2019-03-20:
On my Bardwell F4 V2 it works.
Set up BF 4.0 from scratch, then added the snippet.

On the Wiki page for 4.0 it reads in large and bold letters: "IMPORTANT NOTE : SET 4.0 up from scratch! DO NOT PASTE OLD DUMPS INTO THE CLI!"

supiiik, 2019-03-20:
Wirking if I use sniped and free BEEPER 1 resource (beeper is constantly beepeng now)
```
# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.27V, Core temp=50degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3017, Max available config: 16384
GYRO=MPU6000, ACC=MPU6000
System Uptime: 36 seconds, Current Time: 2019-03-20T07:58:30.417+00:00
CPU:15%, cycle time: 248, GYRO rate: 4032, RX rate: 33, System rate: 10
Voltage: 1179 * 0.01V (3S battery - OK)
I2C Errors: 0
SD card: None inserted
Arming disable flags: CLI
Dshot reads: 463425
Dshot invalid pkts: 0
Dshot irq micros: 3
Dshot RPM Motor 0: 53
Dshot RPM Motor 1: 61
Dshot RPM Motor 2: 65
Dshot RPM Motor 3: 69
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

# resource
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B08
resource LED_STRIP 1 B08
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource INVERTER 1 C00
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B09
resource CAMERA_CONTROL 1 B09
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C03
resource ADC_CURR 1 C01
resource SDCARD_CS 1 B12
resource SDCARD_DETECT 1 B07
resource FLASH_CS 1 B03
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05

# timer
timer B09 1
timer B00 1
timer B01 1
timer A03 0
timer A02 0

# dma
dma SPI_TX 2 0
# SPI_TX 2: DMA1 Stream 4 Channel 0
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin A03 1
# pin A03: DMA1 Stream 6 Channel 3
dma pin A02 0
# pin A02: DMA1 Stream 1 Channel 3

# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 4.0.0 Mar 14 2019 / 11:05:31 (360afd96d) MSP API: 1.41

# start the command batch
batch start

board_name CLRACINGF4
manufacturer_id 

# name

# resources
resource BEEPER 1 NONE

# timer
timer B00 1
timer B01 1

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 150
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 0
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set beeper_frequency = 0
set pid_process_denom = 1
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON

# profile
profile 0

set dterm_lowpass2_hz = 0
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 0

# end the command batch
batch end
```

netzwergracer, 2019-03-20:
On my quad BiDir RPM filter and beeper both work.
No need to free beeper 1 resource.

1. Set up BF 4.0 from scratch
2. set beeper_frequency = 0
3. paste snippet into CLI

supiiik, 2019-03-20:
It seems, that original snipet is working fine. Maybe problem was, that I plugged USB first.
Only LEDs are not working together with dshot bidir...

SebiWolze, 2019-03-20:
I had sooo much problems with RPM Filter and now that worked:

Set up BF 4.0 from scratch
set beeper_frequency = 0
paste snippet into CLI

Thank you all!

Arming disable flags: CLI
Dshot reads: 157380
Dshot invalid pkts: 0
Dshot irq micros: 2
Dshot RPM Motor 0: 0
Dshot RPM Motor 1: 0
Dshot RPM Motor 2: 0
Dshot RPM Motor 3: 0

nyxfpv, 2019-04-03:
I too had to set set beeper_frequency = 0 to get motors 1 & 2 to spin.  now I'm not getting any RPM data from the motors.

Dshot reads: 729976
Dshot invalid pkts: 0
Dshot irq micros: 3
Dshot RPM Motor 0: 0
Dshot RPM Motor 1: 0
Dshot RPM Motor 2: 0
Dshot RPM Motor 3: 0
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

nyxfpv, 2019-04-03:
# version
# Betaflight / CLRACINGF4 (CLR4) 4.0.0 Apr  1 2019 / 11:07:16 (88665a678) MSP API: 1.41

# start the command batch
batch start

board_name CLRACINGF4
manufacturer_id 

# name

# resources

# timer
timer B00 1
timer B01 1
timer B04 NONE
timer B08 NONE

# dma
dma pin B04 NONE
dma pin B08 NONE

# mixer

# servo

# servo mix

# feature
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 5 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1050 0 0
aux 1 13 2 900 1050 0 0
aux 2 35 1 900 1100 0 0

# adjrange
adjrange 0 0 3 900 1025 12 3 0 0
adjrange 1 1 3 1450 1550 12 3 0 0
adjrange 2 2 3 1975 2100 12 3 0 0

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_type = PT1
set gyro_lowpass2_hz = 150
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 0
set acc_hardware = NONE
set min_check = 1010
set serialrx_provider = SBUS
set dshot_idle_value = 450
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set vbat_scale = 111
set beeper_frequency = 0
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set pid_process_denom = 1
set tlm_inverted = ON
set osd_tim2 = 30721
set osd_vbat_pos = 2465
set osd_rssi_pos = 2478
set osd_tim_2_pos = 2487
set osd_throttle_pos = 2456
set osd_vtx_channel_pos = 2049
set osd_mah_drawn_pos = 2433
set osd_warnings_pos = 2441
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set feedforward_transition = 30
set iterm_rotation = ON
set smart_feedforward = ON
set iterm_relax = RPY
set f_pitch = 100
set f_roll = 100
set abs_control_gain = 10
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 2

set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75

# end the command batch
batch end