betaflight/betaflight #7761: 4.0.0-RC1 STM32F745 unified target hard faults during startup in HID code on Nucleo-F746ZG

jflyper, 2019-03-11:
4.0.0-RC1, Nucleo-F746ZG, loaded with `STM32F745` unified target hard faults in HID code during first boot after flashing.
Host machine is OSX 10.13.6 (High Sierra).

```
(gdb) w
#0  0x0022127e in microsISR () at ./src/main/drivers/system.c:93
#1  0x002212dc in micros () at ./src/main/drivers/system.c:107
#2  0x0022130e in delayMicroseconds (us=us@entry=1000)
    at ./src/main/drivers/system.c:128
#3  0x00221322 in delay (ms=47, ms@entry=50) at ./src/main/drivers/system.c:161
#4  0x0022f708 in stopMotors () at ./src/main/flight/mixer.c:542
#5  0x002220f4 in HardFault_Handler () at ./src/main/fc/hardfaults.c:135
#6  <signal handler called>
#7  0x00252ada in USBD_HID_SendReport (pdev=0x2001dcb0 <USBD_Device>, 
    report=0x2000ffb0 "", len=<optimized out>)
    at ./lib/main/STM32F7/Middlewares/ST/STM32_USB_Device_Library/Class/HID/Src/usbd_hid.c:405
#8  0x00000000 in ?? ()
```

mikeller, 2019-03-11:
Just tested with an OMNIBUSF7V2 running the STM32F745 target, and it works just fine after enabling `usb_hid_cdc`. Tested moving the 'sticks' on RX_MSP, and the 'joystick' moves as expected.

Maybe the wedging is caused by the differences between 745 and 746?

DieHertz, 2019-03-11:
F746 has a slightly different ISR vector layout.

closed 2019-04-17 by stale[bot]