betaflight/betaflight #7765: RPM filter motor grinding

robert-hoffmann, 2019-03-11:
With the laters build of BF (betaflight_4.0.0_MATEKF405-1509) and BLHELI32 (Furling32_Multi_32_67), on StingerSwarm 2207 2150KV motors.

I get slight grinding noise on the motors after following the setup guide (to the letter)
https://github.com/betaflight/betaflight/wiki/Bidirectional-DSHOT-and-RPM-Filter

Using 4k/4k

..i did revert gyro_lowpass_hz to the default value of 150, but nothing changed.

This does not happen when not using the RPM filter, and even more wierd, i get no grinding noise when spinning the motors via the BLHELI interface, only from within the configurator.

I also lowered D a little, but that did not help.

Also, once when i spun the motors in BLHELIsuite, and then went straight to BF configurator, the grinding dissapeared, then i unplugged everything (usb+batt ..i do this systematically after each change) and it came back.

It's not horribly bad ..but any idea if this could simply be a setting that could be tweaked, or something else ?

Thanks for any feedback !

`# version
# Betaflight / MATEKF405 (MKF4) 4.0.0 Mar 11 2019 / 11:58:49 (0deca9c6f) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF405
manufacturer_id 
mcu_id 001b00323436510639333434
signature 

# name
name Chameleon

# resources
resource MOTOR 5 NONE
resource SERIAL_TX 11 A15

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 4 2048 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1950 2050 0 0
aux 1 13 0 1950 2050 0 0
aux 2 35 2 1950 2050 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_type = PT1
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 150
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 0
set acc_hardware = NONE
set acc_calibration = 18,30,-23
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 8
set airmode_start_throttle_percent = 20
set blackbox_device = NONE
set dshot_idle_value = 400
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set ibata_scale = 167
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set yaw_deadband = 1
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_rssi_pos = 2138
set osd_tim_2_pos = 2081
set osd_current_pos = 2496
set osd_mah_drawn_pos = 2520
set osd_warnings_pos = 2505
set osd_avg_cell_voltage_pos = 2104
set osd_stat_max_fft = ON
set scheduler_optimize_rate = ON
set vtx_band = 3
set vtx_channel = 8
set vtx_power = 3
set vtx_freq = 5945
set vcd_video_system = PAL

# profile
profile 0

set dyn_lpf_dterm_max_hz = 300
set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 4000
set feedforward_transition = 45
set iterm_rotation = ON
set iterm_relax_type = GYRO
set d_pitch = 33
set f_pitch = 65
set d_roll = 30
set f_roll = 60

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set thr_mid = 30
set thr_expo = 60
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 20
set roll_srate = 85
set pitch_srate = 85
set tpa_rate = 80
set tpa_breakpoint = 1750

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save`

bizmar, 2019-03-11:
If you reset the filters to the defaults and add the RPM filter without disabling anything, does it still make a grinding sound?
  
The grinding sound is indicative of too much noise coming into the d-term. So maybe the system has too much inherent noise that doesn't allow you to disable as much filtering. 
  
A blackbox log with `set debug_mode = gyro_scaled` would be helpful in diagnosing the problem, and maybe one with `set debug_mode = rpm_filter` to verify if it's working. A short props off, log is enough.
  
As far as I know (not being a dev) dining the motors through BL_HELI or BF configurator should be the same, so that's strange and may indicate a deeper problem. But we have to start somewhere.

robert-hoffmann, 2019-03-11:
Yes i've done that, still grinds
Also my sdcard hoder broke, so no more BB for me ^^

Thanks anyways :-)

theSkonk, 2019-03-12:
try setting your dynamic min to 100hz

dadoribeiro, 2019-03-12:
Did you change any settings (besides motor direction) on BLHeliSuite?

robert-hoffmann, 2019-03-12:
@dadoribeiro Yes, i'm on `timing auto, 48khz pwm, and sine mode`, but i tested toggling each setting ...will try again 👍 

Also i'll try a stock BF with just these, and then work my way through

```
set dshot_burst=off
set scheduler_optimize_rate=on
set dshot_bidir=on
set gyro_sync_denom = 2
```

and report back afterwards later tonight ..right now it's time to spend the day flying ^^

butterSnake, 2019-03-12:
Set your timing to between 19-23° auto is not always the best. Way less chance of desyncs with a bit more timing

robert-hoffmann, 2019-03-12:
@butterSnake Ok, noted 👍 
They are 2207 stingerswarm motors @2150kv / 14 poles. I set auto usually cause they said it's usually smoother. Not looking for performance here, just smooth acro flying.

dadoribeiro, 2019-03-12:
Without BB I would set everything to default, BLHeli including. And then go changing 1 setting at a time, starting in BF.

butterSnake, 2019-03-12:
Blheli isn’t gona have a big impact this is a filtering/frame resonance issue

robert-hoffmann, 2019-03-12:
@butterSnake 
The grinding is not flying, but just spinning the motors up slowly without props connected to the pc via BF configurator.

For example when it hits 1450 i'll hear a slight grinding, then towards 1470 it will dissapear, and i'll hear it again at 15xx or whatever ...it's like holes of smooth with bumps of grinding, but usually in the range of 1350-1700 throttle

butterSnake, 2019-03-12:
Dude I’m one of the test pilots..  it’s not blheli32 it’s feedback I’ve had to deal with it in the development when u pushed my filtering too hard. And or your having gyro issues. A log would be better other wise we are pretty much taking pot shots in the dark. Have you got another fc you can try. Any crash hard enough to remove your SD card holder could easily damage the gyro. I had my roll axis go spazzy on me after a bit so bad crash racing a few weeks ago.. took me a week to work out it was shot

butterSnake, 2019-03-12:
Flipping back on the glpf filtering could help. But it it helps it’s only masking the issue

robert-hoffmann, 2019-03-12:
@butterSnake 

It could definately be gyro related ..it's a original matek f405 aio with the sensitive gyro (the one they discontinued). That said, BF4.0 nightly (and even before) flies super smooth already. Never had any twitching issues and basically no propwash.

I'll just try stock everything, and then go through each setting 1 at a time to see what  gives 👍 

Ok, noted for the gLPF filter

butterSnake, 2019-03-12:
It might be too sensitive for rpm.. it really needs a clean airframe and a stable gyro. It really leaves very little filtering on the quad

robert-hoffmann, 2019-03-12:
We'll see :-) Like i said, BF already flies perfectly fine for my taste ..it's just, i hear ppl saying RPM is the best thing since sliced bread, so can't help myself from testing, since i'm a software dev too ^^

bexamous, 2019-03-15:
Am I imssing something..  

> The grinding is not flying, but just spinning the motors up slowly without props connected to the pc via BF configurator.

On bench like that you get grinding?  I think we're getting sidetracked with filtering disucssion then.

If you toggle dshot_bidir setting on/off (power cycle fc&escs after changing setting) the grinding go way and come back?

> Also, once when i spun the motors in BLHELIsuite, and then went straight to BF configurator, the grinding dissapeared, then i unplugged everything (usb+batt ..i do this systematically after each change) and it came back.

This seems odd, right?  Heh I didn't even know blhelisuite had a way to spin up motors.  But anyways when esc is initialized fc sends something so it knows to use bidir mode.  Maybe if you spun it up in blheli it initialized without bidir mode, then if you went straight to bfconfigurator w/o unplugging everything esc is sitll not in bidir mode and then worked?

It would be pretty nice if just toggling dshot_bidir mode made issue reproduce or not..  well I mean what do I know.  Those F3 ESCs also seem a little unusual, wonder how many others have tried them.

joelucid, 2019-03-15:
This is really interesting. It seems to imply that the mere sending of telemetry data by the esc leads to motor irregularities. 
@sskaug any ideas?

sskaug, 2019-03-17:
I have tried spinning a motor with and without bidirectional dshot active, and can not find a difference. There will always be spots in the rpm range where the motor runs a tiny bit rough - this is generally where harmonics of the comutation frequency meets the pwm frequency.

robert-hoffmann, 2019-03-31:
Sorry for the delay but got around to fixing my quad and redoing a test

This is with latest build #1551 on `Furling32_Multi_32_68.Hex`
and just these settings :
```
set dshot_burst = OFF
set scheduler_optimize_rate = ON
set dshot_bidir = ON
```

The noise happens on all 4 motors, even though the video only shows 1
https://youtu.be/Sr64Hf_kAAg