betaflight/betaflight #7766: 4.0.0RC1 KAKUTEF7 doesn't report DSHOT RPM Motor counts

RubberDuck00, 2019-03-12:
Hardware:  KAKUTEF7 (not the all in one), Holybro metal 4in1
KAKUTEF7: BF 4.0.0RC1
ESC:  TEKKO32_F3_4in1 Rev 32.67 - Multi

Followed process described by Bardwell to plug in battery prior to plugging in FC to PC.  I tried build #1505 and got RPM motr counts to increase but one motor had issues spinning (count zero).  The rest appeared to get an error shortly after starting and didn't respond as expected.  DSHOT error counts increased.  Flashed to 4.0.0 RC1 and got good response from motors (they spinned as expected) but the RPM motor counters are not increasing.

BF Configurator 10.5.0
# status
MCU F745 Clock=216MHz, Vref=3.30V, Core temp=57degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3061, Max available config: 32768
GYRO=ICM20689, ACC=ICM20689, BARO=BMP280
System Uptime: 31 seconds, Current Time: 2019-03-12T00:56:21.236+00:00
CPU:9%, cycle time: 251, GYRO rate: 3984, RX rate: 33, System rate: 10
Voltage: 1502 * 0.01V (4S battery - OK)
I2C Errors: 4
SD card: None inserted
Arming disable flags: RXLOSS CLI MSP
Dshot reads: 393676
Dshot invalid pkts: 0
Dshot irq micros: 2
Dshot RPM Motor 0: 0
Dshot RPM Motor 1: 0
Dshot RPM Motor 2: 0
Dshot RPM Motor 3: 0
7 7 7 7 7 14 7 7 7 7 14 14 7 14 14 14 0 0 13 1 15 2 17 4 18 6 0 8 2 9 3 11 
0 0 7 0 0 0 7 0 0 8 7 7 8 8 7 8 

**Flight controller configuration**
```
# diff

# version
# Betaflight / KAKUTEF7 (KTF7) 4.0.0 Mar  8 2019 / 03:18:31 (37dd39659) MSP API: 1.41

board_name KAKUTEF7
manufacturer_id 

# start the command batch
batch start

# name

# resources
resource MOTOR 5 NONE
resource PINIO 1 C09

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature TELEMETRY
feature ESC_SENSOR

# beeper

# beacon

# map

# serial
serial 0 2048 115200 57600 0 115200
serial 2 16384 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 5 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1400 2100 0 0
aux 1 1 0 1400 1500 0 0
aux 2 13 0 1100 1200 0 0
aux 3 35 0 1500 1600 0 0
aux 4 40 3 1250 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_max_hz = 575
set acc_calibration = 76,-18,48
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set current_meter = ESC
set battery_meter = ESC
set small_angle = 180
set pid_process_denom = 1
set osd_vbat_pos = 2058
set osd_vtx_channel_pos = 2456
set osd_current_pos = 2081
set osd_mah_drawn_pos = 2049
set osd_pid_roll_pos = 295
set osd_pid_pitch_pos = 327
set osd_pid_yaw_pos = 359
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 2090
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set pinio_box = 40,41,42,43

# profile
profile 0

set dterm_lowpass2_hz = 0
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 0
```

dadoribeiro, 2019-03-12:
Did you try the conf snippet for the Kakute F7? What motors are you using? Are you sure they are 12 poles?

RubberDuck00, 2019-03-12:
I used the default snippet because there isn't a snippet for KAKUTEF7.  Yes, 12 motor poles.  I'm using F40 pro II 2400kv.

dadoribeiro, 2019-03-12:
The motors are 12N14P, so 14 poles, and you are using 12 poles on the diff. The link for the snippet https://github.com/betaflight/bidircfg/blob/master/KISSFCV2F7.cf

RubberDuck00, 2019-03-13:
Thank you!  I didn't catch that poles are the magnet count.  Corrected it.  I'm unable to reproduce the #1505 results.  I'm just getting zero for the DSHOT RPM counts.  You believe this is an ESC firmware issue and not a FC firmware issue?  I did some quick searches and couldn't find a related issue for BLheli_32.  Can you share the link or reference so I can track when it's fixed?

RubberDuck00, 2019-03-13:
BLHeli_32 32.6.7 issue #344.  Closing this one.  Thanks!

closed 2019-03-13 by RubberDuck00