betaflight/betaflight #7767: gyro_to_use = both

Ralfde, 2019-03-12:
BF 4.0.0 RC1 does not accept the CLI command. 
set gyro_to_use = BOTH
for Omnibus F7NXT.
After setting it does still show gyro_to_use = first or second.
As the iflight FCs does use both gyros, this would be also a really nice feature for the 
Omnibus F7NXT and F7V2.
Especially as these FCs do use two same or two different gyros (users choice which gyro box they use) this should be an even bigger advantage as with the iflight FCs.
Please add this feature for the official release of BF4.

Asizon, 2019-03-12:
Tested in OmnibusF4V6 with dual gyro and working!
# version
# Betaflight / OMNIBUSF4V6 (OBV6) 4.0.0 Mar  8 2019 / 03:28:54 (37dd39659) MSP API: 1.41

board_name OMNIBUSF4V6
manufacturer_id 

# start the command batch
batch start

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_to_use = BOTH

# profile
profile 0

Asizon, 2019-03-12:
NOTE: Both gyro only works with two same gyros installed.
Example: Gyro1-MPU6000/Gyro2-MPU6000

mikeller, 2019-03-12:
@ralfde: Looks like what you are using is a poor quality clone that does not function correctly. Return it and get your money back, then buy an original that works.

Ralfde, 2019-03-12:
I didn't make batch start. I will test tomorrow. Thanks for the info.
I'm using original Airbot Omnibus F7NXT, but 
first gyro: MPU6000
second gyro: ICM-20601
I would like to keep this configuration of gyros, because they work both really good, if I just use first or second gyro.
But if I need to use two of the same gyros, I can also test this tomorrow.

etracer65, 2019-03-12:
That is as designed. If the gyros are not the same part number then the option `BOTH` cannot be used and will be reset to `FIRST`.

Ralfde, 2019-03-12:
Ok. Thanks. Could this maybe be changed to allow also two different gyros? Could I maybe change it myself? In which part of the software (file) could I try to change it? Or could you maybe make a custom build for testing for me?

mikeller, 2019-03-12:
@Ralfde: Feel free to open a pull request to implement this.

etracer65, 2019-03-12:
No, it would require substantial refactoring of the gyro initialization code in Betaflight to do this properly. There is only one data structure describing the initialization parameters for the gyros and no ability to have the differentiated settings needed different gyro devices.

Ralfde, 2019-03-12:
thanks. So I hope I will find a MPU6000 gyro box for testing.

Ralfde, 2019-03-13:
I tested this morning Omnibus F7V2 with 2 MPU6000.
I did the CLI commands:
batch start
set gyro_to_use = BOTH
save

Now the setting seem to be stored correctly, but the copter behaviour in the setting view seem to be strange. The gyro seem to make the half travel and the accel does then correct it. Is this just a viewing problem in the 10.4.0 configurator?

Ralfde, 2019-03-13:
@mikeller: It looks like I can't do pull requests. Could you please do this for me?
Would be really nice to allow also two different types of gyros.

etracer65, 2019-03-13:
@Ralfde 

> Now the setting seem to be stored correctly, but the copter behaviour in the setting view seem to be strange. The gyro seem to make the half travel and the accel does then correct it. Is this just a viewing problem in the 10.4.0 configurator?

You need to set the alignment manually for each gyro in the CLI.
```
gyro_1_sensor_align = CW270
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

gyro_2_sensor_align = CW90
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP
```
You'll need to figure out which values are correct for your boards. Test each one separately by using `set gyro_to_use = FIRST` and `SECOND` to make sure the alignment is correct. Don't try to use the configurator the change these settings as it's not dual-gyro aware and will reset both values.

Lastly, if it turns out that both gyros are not offset by 90 degrees then it really makes no sense to use both gyros at the same time. They'll just both be averaging the same axes and you'll be accomplishing nothing other than using more CPU cycles.

Ralfde, 2019-03-13:
That's why I would see the MPU6000 together with the ICM-20601 in the Omnibus F7NXT.

Different gyros with different vibrations and also different orientation of the axes.

mikeller, 2019-03-13:
@Ralfde:

> @mikeller: It looks like I can't do pull requests. Could you please do this for me?
Would be really nice to allow also two different types of gyros.

You should be able to do this. But if it does not work for you, just send me your proposed code changes and I will open a pull request for you.

> That's why I would see the MPU6000 together with the ICM-20601 in the Omnibus F7NXT.

> Different gyros with different vibrations and also different orientation of the axes.

Can you elaborate a bit on what you mean by 'different vibrations', and how do you come to assume that using different gyro models would imply they have a different orientation of the axes?

Ralfde, 2019-03-13:
Can you elaborate a bit on what you mean by 'different vibrations', and how do you come to assume that using different gyro models would imply they have a different orientation of the axes?

At Omnibus F7NXT 
Gyro 1 MPU6000 is CW0
Gyro 2 ICM-20601 is CW270

If you set the ICM-20601 also to 16g and 2000dps for the gyro, the calcuations would be the same, only the init routines would be different for them.
But MPU60000 and ICM-20601 should have different vibration behaviour.

gyro_1_sensor_align = CW0
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

gyro_2_sensor_align = CW270
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

etracer65, 2019-03-13:
As I've already stated, to correctly support different gyro types simultaneously would require a substantial refactoring of the gyro data structures and initialization functions. There's no quick fix. Feel free to work on these changes if you want.

At this point there's also no evidence that utilizing multiple gyros (even of the same type) provides any benefit of advantage. And this lack of advantage is offset by increasing the processing needed to interface two gyro sensors.

Ralfde, 2019-03-13:
Sorry, I'm not a programmer, only advanced user...
I can maybe change a single line, if you tell me which one..., but it seem to be not that simple.
It doesn't need to be a quick fix. Could be maybe done for BF 4.1 or 4.2.

mikeller, 2019-03-13:
@Ralfde:

> It doesn't need to be a quick fix. Could be maybe done for BF 4.1 or 4.2.

Are you going to do it by 4.1 or 4.2?

Because if not, then there is absolutely nothing that gives you the right to make statements about what needs or doesn't need to be done in Betaflight. Betaflight is a volunteer project, and the volunteers who contribute their time get to decide what they work on, nobody else. Please stop expressing an attitude of you having a say in what gets done in Betaflight in here right now.

Ralfde, 2019-03-13:
Sorry, it was only a suggestion.

Ralfde, 2019-03-16:
I tested now Omnibus F7NXT with two MPU6000. 
In the angle mode, the angle seem to be doubled and in the rate mode the rates seem to be doubled, too. 

# diff all

# version
# Betaflight / OMNINXT7 (ONX7) 4.0.0 Mar 11 2019 / 12:07:22 (0deca9c6f) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNINXT7
manufacturer_id 
mcu_id 003300283137510f30343533
signature 

# name

# resources
resource MOTOR 1 B00
resource MOTOR 3 B05

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature ESC_SENSOR

# beeper

# beacon

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 1 8 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1300 2100 0 0
aux 1 1 0 1000 1700 0 0
aux 2 26 3 1800 2000 0 0
aux 3 30 2 1000 1200 0 0
aux 4 35 2 1800 2000 0 0
aux 5 39 8 1800 2000 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_lowpass_type = PT1
set gyro_to_use = BOTH
set acc_calibration = 106,-23,-357
set mag_declination = 2420
set min_check = 1010
set rssi_channel = 16
set rssi_scale = 125
set rssi_offset = -12
set fpv_mix_degrees = 35
set serialrx_provider = SUMD
set blackbox_p_ratio = 128
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT1200
set current_meter = ESC
set ibata_scale = 200
set ibata_offset = -3500
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_angle = 20
set gps_rescue_initial_alt = 20
set gps_rescue_descent_dist = 30
set gps_rescue_ground_speed = 1000
set gps_rescue_throttle_max = 1500
set gps_rescue_sanity_checks = RESCUE_SANITY_OFF
set deadband = 10
set yaw_deadband = 10
set pid_process_denom = 1
set osd_vbat_pos = 2100
set osd_rssi_pos = 2134
set osd_vtx_channel_pos = 2497
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2080
set osd_mah_drawn_pos = 2092
set osd_gps_speed_pos = 225
set osd_gps_sats_pos = 193
set osd_home_dir_pos = 238
set osd_home_dist_pos = 258
set osd_altitude_pos = 2262
set osd_pidrate_profile_pos = 2467
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set cpu_overclock = 240MHZ
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 3
set vtx_freq = 5732

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 90
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set iterm_relax = OFF
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 100
set iterm_limit = 500
set pidsum_limit = 600
set pidsum_limit_yaw = 600
set yaw_lowpass_hz = 150
set throttle_boost = 0
set acro_trainer_angle_limit = 40
set p_pitch = 45
set d_pitch = 34
set f_pitch = 100
set p_roll = 40
set d_roll = 30
set f_roll = 100
set p_yaw = 55
set abs_control_gain = 0
set d_min_roll = 0
set d_min_pitch = 0
set d_min_boost_gain = 30

# profile
profile 1

set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set yaw_lowpass_hz = 150
set throttle_boost = 0
set acro_trainer_angle_limit = 40
set p_pitch = 45
set d_pitch = 34
set p_roll = 40
set d_roll = 30
set p_yaw = 55

# profile
profile 2

set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set iterm_relax = RPY
set yaw_lowpass_hz = 150
set throttle_boost = 0
set acro_trainer_angle_limit = 40
set p_pitch = 45
set d_pitch = 34
set p_roll = 40
set d_roll = 30

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60
set tpa_rate = 20
set tpa_breakpoint = 1100
set tpa_mode = PD

# rateprofile
rateprofile 1

set tpa_rate = 10
set tpa_breakpoint = 1150

# rateprofile
rateprofile 2

set tpa_rate = 10
set tpa_breakpoint = 1150

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

etracer65, 2019-03-16:
> In the angle mode, the angle seem to be doubled and in the rate mode the rates seem to be doubled, too.

That would be extremely unlikely. Have you confirmed with any data such as a blackbox log or is this just a subjective opinion? Capture a blackbox log and use `set debug_mode = DUAL_GYRO` to capture some data from both gyros. If the combined gyro rates track close to the setpoint traces then there is no problem.

Ralfde, 2019-03-16:
Don't need a log. I adjusted now the half rates in the configurator and now they are like before.

etracer65, 2019-03-16:
Well if you're not going to provide a log I don't see how we can evaluate if there's a problem. And we certainly won't waste effort researching otherwise.

Ralfde, 2019-03-16:
Before it rains again, here the logfile and diff all and dump all:
[Logfiles.zip](https://github.com/betaflight/betaflight/files/2974132/Logfiles.zip)

etracer65, 2019-03-16:
The problem is that you second gyro is not working.

Ralfde, 2019-03-16:
When I do:
set gyro_to_use = SECOND, then the second gyro works:
[image]
Could you please give me a tip in which file for the gyro support for both gyro is included?

etracer65, 2019-03-16:
src/main/sensors/gyro.c

Ralfde, 2019-03-16:
@Asizon: did you use two MPU6000 with Omnibus F4V6?
Did I miss maybe anything (a command), because it's not working with the Omnibus F7NXT with two MPU6000?

mikeller, 2019-03-16:
@Ralfde:

> Don't need a log. I adjusted now the half rates in the configurator and now they are like before.

You seriously need to adjust the attitude you are showing in here. Nobody in here does anything at your service, and if they ask you to provide more information for problems you are reporting, then brushing them off is not the right thing to do.

If you want to get help in a way that you can make requests, and others have to help you, can I recommend you leave here, and find a place where you can pay others to provide help for you?

Ralfde, 2019-03-17:
The Omnibus F4V6 with two MPU6000 does show the same problem. 
Maybe I missed a needed setting.
[Logfiles.zip](https://github.com/betaflight/betaflight/files/2975208/Logfiles.zip)

Asizon, 2019-03-17:
What is to be doubled?what is exactly your problem with gyro both mode?

Ralfde, 2019-03-17:
The rate and angle is doubled. etracer65 told me, that the second gyro did not work, which does cause the problem. So I would like to find out, why the second gyro does not work correctly in the gyro_to_use = BOTH mode.
But when I set gyro_to_use = second, the second gyro does work correct and in the set gyro_to_use = first mode the first gyro does work correct.
I tried OmnibusF4V6 with RC2 and the Omnibus F7NXT with the firmware shown in the textfile.
I use two MPU6000 and I used the cli commands:
batch start
set gyro_to_use = BOTH

I'm also useing the bidirectional DSHOT.
set dshot_burst=off
set scheduler_optimize_rate=on
set dshot_bidir=on
set motor_poles=14

But I believe the problem is already there before I set the bidir mode. So I believe I'm missing a needed command or I need a different firmware build to make also the second gyro to work correct.

Ralfde, 2019-03-28:
I saw, that some changes where did for the BF4.0.0 RC4, but the problem with the OmnibusF4V6 and OmnibusF7NXT is still the same. At configuration of BOTH gyros (MPU6000), only one seem to work.

Ralfde, 2019-04-05:
BF4.0.0 RC6 with Omnibus F7NXT.
set gyro_to_use = BOTH
The Accel sensor does not work any more:
[image]

Should it be like this?

# diff all

# version
# Betaflight / OMNINXT7 (ONX7) 4.0.0 Apr  3 2019 / 14:18:18 (22b9f3453) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNINXT7
manufacturer_id 
mcu_id 004300293137511230343533
signature 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map
map TAER1234

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_to_use = BOTH
set acc_calibration = 67,37,-168
set rc_interp = MANUAL
set rc_interp_int = 10
set rc_smoothing_type = INTERPOLATION
set deadband = 10
set yaw_deadband = 10

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

mikeller, 2019-04-07:
Tested with Betaflight 4.0.0-RC6 on a SPRACINGF7DUAL (I don't have an OMNIBUSF7 with dual gyros), and ACC is working (the scale on the graph seems to be off by a factor of 0.5, but it's moving):

[image]

Ralfde, 2019-04-07:
The OmnibusF7NXT doesn't show this accel values. Should I try to organize a FC sample?

mikeller, 2019-04-07:
I think @jflyper should have one - getting Airbot samples to me in New Zealand usually takes months.

Can you try resetting to defaults, and re-applying the config, but **without the** `set acc_calibration = 67,37,-168`?

Asizon, 2019-04-07:
Ok, confirmed, I have same issue with both, no accel works on OmnibusF4 V6

Asizon, 2019-04-07:
could be a MPU problem @mikeller ? Because you are testing ICM on SpracingDual right?we are testing with mpu6000

Asizon, 2019-04-07:
Lol!Tested with 3.5.4 it works fine!
[image]

Asizon, 2019-04-07:
@mikeller testing 4.0 builds, I can confirmed that the problem is caused by commit: e44f75b, the build #1556 seems ok, acc working

mikeller, 2019-04-07:
@Asizon: Weird. Can you confirm that this only applies when `gyro_to_use = both`, but not when `gyro_to_use = first` or `gyro_to_use = second`?

Asizon, 2019-04-07:
Confirmed!Only with gyro BOTH having this issue, gyro first and second seems Acc OK

mikeller, 2019-04-07:
@Asizon: And with `gyro_to_use = both`, it is both the graph on the 'Tethered Logging' and the model on the 'Setup' tab that do not show any movement? Can you record a blackbox log and check if that one shows ACC movement?

Asizon, 2019-04-07:
in the model of setup, show a movement but wrong movement, when I calibrated acc the quad seems to autostabilize,before doing the calibration the quad makes half movement and not complete. Im going to try to record a log

mikeller, 2019-04-07:
@Asizon: This sounds a lot like the alignment for gyro 1 and gyro 2 are not set up correctly, and are now cancelling out each other. Can set `gyro_to_use = first`, then use the model in `Setup` to adjust `gyro_1_sensor_align` until the model moves in accordance to you moving the board, then use `gyro_to_use = second` to do the same for `gyro_2_sensor_align`, and then try again?

Asizon, 2019-04-07:
Yes this is already checked, 2 gyros have the same moves accordance to the board. On 4.0 build #1556 it works correctly, not problems found

Asizon, 2019-04-07:
[BLACBOXLOGGYROBOTHTEST.zip](https://github.com/betaflight/betaflight/files/3051414/BLACBOXLOGGYROBOTHTEST.zip)

Ralfde, 2019-04-07:
With build #1556 the accel works, but the second gyro of both gyros seem still not to work, because the movement on the screen is like earlier.

Asizon, 2019-04-07:
Yes,the movement is like earlier respect to reality,but for the moment we dont know the reasson

Ralfde, 2019-04-08:
I made a log with #1556 btfl_001.bbl and with #1559 btfl_002.bbl
Maybe it helps.
[Gyro Both.zip](https://github.com/betaflight/betaflight/files/3052886/Gyro.Both.zip)

Asizon, 2019-04-08:
@etracer65 Do yout think if your new DUAL_GYRO_SCALED debug mode can be help to fix it?

etracer65, 2019-04-08:
The debug mode just reports the values to the blackbox log - it can't fix anything.

etracer65, 2019-04-08:
Can I get some clarity on your setups? Do I understand it correctly that you're trying to use some Omnibus flight controller that has an onboard MPU6000 and a separate external gyro box? And then you've replaced the external gyro box (that would normally have an ICM gyro) with a new one that has a MPU6000?

If so, then the default alignment in the target will be correct for gyro 1 (as it hasn't changed), but the default for gyro 2 is unlikely to be correct as it was configured expecting the external box to contain an ICM gyro.

So have you properly set `gyro_2_sensor_align` in the CLI to match the actual alignment of the second gyro? Don't just set it the same as gyro 1 as there's no requirement that they are mounted the same. You need to actually test and figure out the proper alignment. You can do this by using gyro 2 only (`set gyro_to_use = SECOND`) and then observing the model in the Setup tab. If the motion is not matching reality, then try a different value for `gyro_2_sensor_align` - there are 8 different values to try (don't use `DEFAULT`). **Also don't use the sensor alignment options on the Configuration tab to adjust as it's not dual-gyro aware in the currently released Configurator - change the settings in the CLI.**

After you get the alignments correct for **each gyro** then everything should work properly with `gyro_to_use = BOTH`.

I tested on a variety of dual-gyro boards and every one worked properly once I ensured the alignments were correct.

Ralfde, 2019-04-08:
Yes, I use MPU6000 gyro box. In case it does work with ICM gyro, then I would try it, but I believe Gyro 1 and 2 must be the same type to allow gyro_to_use = both setting. 
I tested first and second gyro seperate and set the gyro_2_sensor_align correctly. I angled the flat strap to have different gyro direction. Gyro movement does look better with #1559, but accels do not show any changes in the graph. Could you check my logfiles already?
Would be interesting, if at least the gyro works correctly and if only the accel is a problem right now.

etracer65, 2019-04-08:
While you're in `gyro_to_use = BOTH` mode, what does the output of the `gyroregisters` CLI command show?

etracer65, 2019-04-08:
Also the output of the `status` command.

Ralfde, 2019-04-09:
# get gyro
gyro_hardware_lpf = NORMAL
Allowed values: NORMAL, 1KHZ_SAMPLING, EXPERIMENTAL

gyro_sync_denom = 1
Allowed range: 1 - 32

gyro_lowpass_type = BIQUAD
Allowed values: PT1, BIQUAD

gyro_lowpass_hz = 150
Allowed range: 0 - 16000

gyro_lowpass2_type = BIQUAD
Allowed values: PT1, BIQUAD

gyro_lowpass2_hz = 0
Allowed range: 0 - 16000

gyro_notch1_hz = 0
Allowed range: 0 - 16000

gyro_notch1_cutoff = 0
Allowed range: 0 - 16000

gyro_notch2_hz = 0
Allowed range: 0 - 16000

gyro_notch2_cutoff = 0
Allowed range: 0 - 16000

gyro_calib_duration = 125
Allowed range: 50 - 3000

gyro_calib_noise_limit = 48
Allowed range: 0 - 200

gyro_offset_yaw = 0
Allowed range: -1000 - 1000

gyro_overflow_detect = ALL
Allowed values: OFF, YAW, ALL

gyro_to_use = BOTH
Allowed values: FIRST, SECOND, BOTH
Default value: FIRST

dyn_lpf_gyro_min_hz = 150
Allowed range: 0 - 1000

dyn_lpf_gyro_max_hz = 450
Allowed range: 0 - 1000

gyro_filter_debug_axis = ROLL
Allowed values: ROLL, PITCH, YAW

gyro_cal_on_first_arm = OFF
Allowed values: OFF, ON

gyro_1_bustype = SPI
Allowed values: NONE, I2C, SPI, SLAVE

gyro_1_spibus = 1
Allowed range: 0 - 4

gyro_1_i2cBus = 0
Allowed range: 0 - 4

gyro_1_i2c_address = 0
Allowed range: 0 - 119

gyro_1_sensor_align = CW0
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

gyro_2_bustype = SPI
Allowed values: NONE, I2C, SPI, SLAVE

gyro_2_spibus = 1
Allowed range: 0 - 4

gyro_2_i2cBus = 0
Allowed range: 0 - 4

gyro_2_i2c_address = 0
Allowed range: 0 - 119

gyro_2_sensor_align = CW270
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

gyro_rpm_notch_harmonics = 3
Allowed range: 0 - 3

gyro_rpm_notch_q = 500
Allowed range: 1 - 3000

gyro_rpm_notch_min = 100
Allowed range: 50 - 200

Ralfde, 2019-04-09:
# status
MCU F722 Clock=216MHz, Vref=3.29V, Core temp=49degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3083, Max available config: 16384
GYRO=MPU6000, ACC=MPU6000, BARO=LPS
System Uptime: 111 seconds, Current Time: 0000-01-01T00:00:00.000+00:00
CPU:4%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 9
Voltage: 14 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 2
Arming disable flags: RXLOSS CLI MSP

Asizon, 2019-04-09:
Try with gyroregisters

[12 more replies]

PR #7950 (2019-04-14): Fix gyro/acc initialization when more than one MPU6000 (or MPU9250) sensor is connected
Fixes #7767 

Logic was not expecting there to be multiple MPU6000's connected simultaneously. There are currently no boards manufactured with this config, but it's possible to do using boards that have an onboard MPU6000 and an external gyro box containing a second MPU6000.
closed 2019-04-14 by mikeller