betaflight/betaflight #7768: 4.0.0RC1 - Foxeer F722 Dual does not detect ESCs

GhostMaster9, 2019-03-12:
Ive installed the 4.0.0RC1 but seems like the ESCs aren't detected by the FC, i mean they are detect in blheli32 suite but the initialize sound does not finish. 

Im using Furling32 with 32.6 Blheli32 and Foxeer F722 with 4.0 RC1.

Only oneshot125 seems to work and not with all motors, some get the last beeps of initialization, and some dont. 

Thanks.

mikeller, 2019-03-12:
@GhostMaster9: Please provide the information that you were asked to provide when opening this issue - otherwise there is little chance that anybody will actually look at your issue.

GhostMaster9, 2019-03-12:
> @GhostMaster9: Please provide the information that you were asked to provide when opening this issue - otherwise there is little chance that anybody will actually look at your issue.

Edited the first post, sorry.

dadoribeiro, 2019-03-12:
Was it working before in 3.5.x? Maybe try to run the snippet for DSHOT Bdir, but without the filter portion?

mikeller, 2019-03-12:
@GhostMaster9: So is this correct that none of the motors work when using Dshot600? (You can test this by **taking the props off** and then using the sliders in the 'Motors' tab to spin the motors.)

GhostMaster9, 2019-03-12:
Exactly, nome of the motors work, last build that worked was Betaflight 4.0 Development from 1 March. 
It worked before RC1 for sure, this drone had been flying, i will try the lastest build from today. Thanks guys

eumelOne, 2019-03-12:
I can confirm this issue with the same FC+ESC combination. But i figured out with the dma pin section from the bidir dshot snippet everything works fine and all escs are initialise and the motors are running.

mikeller, 2019-03-14:
Just tested on a FOXEERF722 with 4.0.0-RC1, and I can reproduce that Dshot600 does not result in a valid output signal. I do not know what is causing this.

The good news is I also tested with the latest on `master`, and with this the Dshot600 output signal is good on m1 to m4.

mikeller, 2019-03-14:
@GhostMaster9: Can you try 4.0.0-RC2 please?

GhostMaster9, 2019-03-14:
> @GhostMaster9: Can you try 4.0.0-RC2 please?

Yes its working :D

Thanks

eumelOne, 2019-03-14:
Jup, tested it also and it works.

mikeller, 2019-04-04:
@GhostMaster9: If this is fixed, can you close the issue please?

closed 2019-04-04 by GhostMaster9