betaflight/betaflight #7769: Calculated Flight time gives wrong calculation

cusglider, 2019-03-12:
**Describe the bug**
The first calculated value appears way after one minute into the flight and suggests a wrongly calculated flight time. Landing after 7 minutes still give you a value of 7 min more to go.

**To Reproduce**
Attach battery, arm and take off

**Expected behavior**
Giving me sooner a wrightly calculated flight time, like in version 3.5.6

**Flight controller configuration**
# version
# Betaflight / MATEKF722 (MKF7) 4.0.0 Mar  8 2019 / 03:23:32 (37dd39659) MSP API: 1.41

board_name MATEKF722
manufacturer_id 
mcu_id 004c00233037510f35333131
signature 

# reset configuration to default settings
defaults nosave

# start the command batch
batch start

# name
name CUS Black QAV-R

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper
beeper -ON_USB
beeper -BLACKBOX_ERASE

# beacon
beacon RX_LOST

# map
map TAER1234

# serial
serial 2 32 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1325 2100 0 0
aux 1 1 1 900 1225 0 0
aux 2 13 2 1300 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set mag_hardware = NONE
set rssi_channel = 8
set motor_pwm_protocol = DSHOT600
set failsafe_throttle = 1100
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1500
set pid_process_denom = 1
set osd_warn_crash_flip = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_vbat_pos = 2391
set osd_rssi_pos = 2170
set osd_tim_2_pos = 2487
set osd_remaining_time_estimate_pos = 2520
set osd_flymode_pos = 2105
set osd_throttle_pos = 2232
set osd_vtx_channel_pos = 2081
set osd_crosshairs_pos = 2253
set osd_current_pos = 2508
set osd_mah_drawn_pos = 2496
set osd_craft_name_pos = 2088
set osd_altitude_pos = 2144
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2423
set osd_disarmed_pos = 2378
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_channel = 8
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5880

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 95
set pitch_rc_rate = 95
set yaw_rc_rate = 105

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
MATEKF722 (MKF7) bought at Banggood 2018
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
Frsky R-XSR, Matek VTX

 - how are the different components wired up.
normally soldered
**Additional context**
Add any other context about the problem here.

mikeller, 2019-03-12:
@cusglider: Am I right to assume that what you are talking about is the 'Remaining time estimate' OSD element?

If so, make sure that this setting is correct

> set bat_capacity = 1500

and that your current meter is calibrated properly and reports accurate values, as the remaining time estimate is based on total consumption, current amperage, and total capacity.

cusglider, 2019-03-12:
Hi Mike,

 

thanks for the fast response.

 

Yes, your assumption is right, I am talking about the 'Remaining time estimate' OSD element.

 

I made the suggested entry in the CLI and a test hovering in my living room.

 

It took more than 2 minutes now to get a value.

After a hover time of 3:30 the 'Remaining time estimate' showed 28:14 which is way too high, like the values I got in the field today.

 

Best Regards

 

Carl

 

P.S: Thanks to you and the team for the inspiring work fort he community!!!!

etracer65, 2019-03-12:
Do you even have a current sensor connected? The MatekF722 doesn't have an on-board current sensor. If you have a current sensor, have you calibrated it? The remaining time calculation is based on the current consumption so far vs. the configured battery size. So if the current consumption measurements are not accurate there's no way the remaining time estimate can be either.

cusglider, 2019-03-13:
Good morning Bruce,

 

to my understanding the Matek F7 has a current sensor picking up the data right where you connect the batteries. 

 

I have set it up the way as Matek suggested (see attached file). 

 

Please note that all is working well with Betaflight 3.5.5 and 3.5.6, making me guess there has to be something different between those versions and 4.0 RC1 that is handling the OSD feature remaiing flight time.

 

So the possibly uncorrect settings are not present.

 

Best Regards

 

Carl

etracer65, 2019-03-13:
The MatekF722-STD (the only board using the MATEKF722 target) does not have an on-board current sensor. It has a current sensor connection that must be supplied by either a PDB or 4-in-1 ESC that has a current sensor. So my original point remains. If the board isn't getting accurate current readings then there's no way the remaining time estimate can be correct. So first you need to resolve whether you actually have a current sensor connected and if so calibrate the range so that the current readings are accurate.

Lastly, since you're replying through email there's no way to see any "attached" files or pictures. If you with to attach files please reply directly on Github.

cusglider, 2019-03-13:
Hi Bruce,

Point taken. However this does not explain why the same configuration has been working fine and precise for a year on various betaflight versions, including the latest using the calibration values Matek provided in their manual.

I am currently in the field and will try to resend the file in the evening thru github.

Carl
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/betaflight/betaflight/issues/7769#issuecomment-472370919",
"url": "https://github.com/betaflight/betaflight/issues/7769#issuecomment-472370919",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

cusglider, 2019-03-13:
[image]

cusglider, 2019-03-13:
The Amp meter is correctly calibrated with the value 179 and has been working correctly with many betaflight versions before. Reinstalling Betaflight 3.5.6 gives again correct calculations.

cusglider, 2019-03-13:
[image]

cusglider, 2019-03-13:
First calculated value arrives after 1:25min into the flight, usually around 0:38 min. The consumption is correctly measured, however the calculated remaining flight time makes no sense with 12:39min. Again, putting Betaflight 3.5.6 back on all values are correctwithout having done a change to the quad, FC etc.

etracer65, 2019-03-13:
Ok, it would have helped if you supplied all this info initially when asked about whether you had a current sensor, etc.

One problem I see in your config is that you have not set the default OSD battery capacity alarm (`osd_cap_alarm`) which means it's still the default of 2200mah. The remaining time estimate is based on this alarm. The display shows `--:--` until you consume 10% of this alarm limit (so 220mah for you).  After that it begins to calculate based on usage so far.  Since you seem to have a 1500mah battery I can presume that basing the calculation on an alarm limit of 2200 won't work very well.  Assuming you wanted to consume 80% of your battery (a normal recommendation), then you should set `osd_cap_alarm = 1200`.

cusglider, 2019-03-13:
I learned something. I was not aware of the crossover of the alarm value and the remaining time calculation. Thanks and sorry for causing you some hassle

mikeller, 2019-03-14:
@etracer65: Yeah, this is something I've been meaning to fix for a while now - I can not see why anybody would want to set different battery capacities for acoustical warnings and OSD warnings, so the two should probably be consolidated into one.

adam-ah, 2023-06-30:
@mikeller Late thread, but I was caught up with the same issue.
May I suggest predicting flight-time based on battery voltage decrease over time?
We don't even need to know how the total mAh or current consumption.
Instead, we could use the following information:
- the typical LiPo discharge curve
- the current V
- the flight time so far

Using these we could estimate remaining flight time and I propose it would be more useful than the current approach which requires calibrating the current readings (many people cannot do it properly), and adding the battery's capacity to BF before takeoff (if you use different batteries it can be a pain in the shoe).

I'm happy to help to write the code / review the code / test it.

What do you think?

ledvinap, 2023-07-10:
@adam-ah : You can take blackbox logs, (export them to excel), and try your algorithm ...

adam-ah, 2023-07-10:
Good idea, @ledvinap, just did it

I created a Python notebook to test it out - it seems accurate even on the partial blackbox data I have (I don't have a full 100 to 0 flight recorded, will get one tomorrow).

Had to add a moving average over the last 10 seconds (over 10 data points, only taking every 1000th / 1s sample) to smooth out the data. 

Also tested the Savitzky–Golay filter but the moving average was adequate (and much easier to implement).

Will update this when I have a full flight data.

adam-ah, 2023-07-16:
UPDATE: I'm making better than expected progress with it, the neural net based predictions are excellent (I'll write up what I tested so far, including calculating battery% and current flight times, none of them were great).
However, I would need more test data - where could I find full-flight binary blackbox files (*.BBL) to fine-tune the model? 
@ledvinap @mikeller

closed 2026-09-26 by blckmn