betaflight/betaflight #7770: Micro banking on aileron axis

cusglider, 2019-03-12:
**Describe the bug**
In straight forward flight the quad banks fast and slightly only a few degrees back and forth on the aileron axis. It is not so noticable during turns

**To Reproduce**
Take off, climb and straight level flight forward

**Expected behavior**
Keeping stability like on version 3.5.6 and show no "micro banking"
**Flight controller configuration**
# version
# Betaflight / MATEKF722 (MKF7) 4.0.0 Mar  8 2019 / 03:23:32 (37dd39659) MSP API: 1.41

board_name MATEKF722
manufacturer_id 
mcu_id 004c00233037510f35333131
signature 

# reset configuration to default settings
defaults nosave

# start the command batch
batch start

# name
name CUS Black QAV-R

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper
beeper -ON_USB
beeper -BLACKBOX_ERASE

# beacon
beacon RX_LOST

# map
map TAER1234

# serial
serial 2 32 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1325 2100 0 0
aux 1 1 1 900 1225 0 0
aux 2 13 2 1300 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set mag_hardware = NONE
set rssi_channel = 8
set motor_pwm_protocol = DSHOT600
set failsafe_throttle = 1100
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1500
set pid_process_denom = 1
set osd_warn_crash_flip = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_vbat_pos = 2391
set osd_rssi_pos = 2170
set osd_tim_2_pos = 2487
set osd_remaining_time_estimate_pos = 2520
set osd_flymode_pos = 2105
set osd_throttle_pos = 2232
set osd_vtx_channel_pos = 2081
set osd_crosshairs_pos = 2253
set osd_current_pos = 2508
set osd_mah_drawn_pos = 2496
set osd_craft_name_pos = 2088
set osd_altitude_pos = 2144
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2423
set osd_disarmed_pos = 2378
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_channel = 8
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5880

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 95
set pitch_rc_rate = 95
set yaw_rc_rate = 105

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
MATEKF722 (MKF7)
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
Frsk R-XSR, Matek VTX
 - how are the different components wired up.
normal soldering
**Additional context**
Add any other context about the problem here.

McGiverGim, 2019-03-12:
Can you test setting abs_control to zero? Your problem is different than mine, but maybe origin is the same. A blackbox log will help too.

cusglider, 2019-03-12:
Hi,

I understand to enter this term in the cli. I will do so and come back to you.

Best Regards

Carl
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/betaflight/betaflight/issues/7770#issuecomment-472168266",
"url": "https://github.com/betaflight/betaflight/issues/7770#issuecomment-472168266",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

cusglider, 2019-03-12:
Hi  Míguel,

 

After finding out I cannot set this value using the CLI I guess you mean with „abs_control“ the „Absolute Controll“ in the PID Controller settings.

 

I will give it a try. Sorry, I am not running blackbos, however, if you wish me to do so  can give it a try.

 

Thanks for the support and the work you guys do fort he community!

 

Best Regards

 

Carl

 

Von: Carl-Ulrich Stoltz <CUStoltz@gmx.net> 
Gesendet: Dienstag, 12. März 2019 22:41
An: betaflight/betaflight <reply@reply.github.com>
Betreff: Aw: Re: [betaflight/betaflight] Micro banking on aileron axis (#7770)

 

Hi,

I understand to enter this term in the cli. I will do so and come back to you.

Best Regards

Carl
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

McGiverGim, 2019-03-12:
You are right, the CLI command is set abs_control_gain=0 but doing it in the Configurator as you did is the same.

cusglider, 2019-03-12:
Great help Miguel,

 

I will fly tomorrow and I am curious about what I will find out.

McGiverGim, 2019-03-13:
Another thing you can test is disable D_MIN Maybe the D Min is too low for you. If disabling the abs control does not work, downgrade your D values to the values of version 3.5 (default roll 25 and pitch 27 or any value you tuned and you know is working right) and disable d_min with `set d_min_roll=0` and `set d_min_pitch=0`.

cusglider, 2019-03-13:
Good morning to your side of the world,

 

in case abs control does not work, should I reanable the value to 10, before I try the other suggestions?

 

Will be leaving to the field in one hour taking my notebook with me. Have a great day.

 

Carl

McGiverGim, 2019-03-13:
As you prefer. They are totally different features, will not affect the tests of your problem.

cusglider, 2019-03-13:
Hi Miguel

Results fresh from the field:

1. Only abs control=0
Stronger, almost rolling movements on the aileron axis

2. Abs control=10, disabled D_Min
Movements still there, though less, no "rolling" effect

3. Abs controll 10, disabled D_Min, PID s to suggested Vers. 3.5 values
Smoothness returned, almost not noticable aileron moves, best results so far

What can i tune in which way to eliminate the tiny "rest" movements?

I have DVR footage on the flights.

Best Regards

Carl

McGiverGim, 2019-03-13:
Sorry, I can't say. I hoped some of the new 4.0 features was the culprit, but your tests only say that seem D Term related. Without a Blackbox log I can't say, I suggest to go to the official Betaflight slack (http://slack.betaflight.com) and ask in the #pidtuning channel, there are more people with experience with that.

cusglider, 2019-03-13:
So it would be helpful for you guys to receive a blackbox file?

On which of the flights would you want a file?
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

McGiverGim, 2019-03-13:
I think with defaults will be ok. Trying to show the problem.

cusglider, 2019-03-13:
Where should i send the file to?
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

McGiverGim, 2019-03-13:
You can attach it to the issue, a simply drag and drop I think it will work.

cusglider, 2019-03-13:
I will do it in the evening. Just made 2 log files with the original RC1 settings.

Thanks for your assistance. For the rest i love the flight characteristics. So close to the stick while maintaining a smoothness.
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

cusglider, 2019-03-13:
[LOGS.zip](https://github.com/betaflight/betaflight/files/2962109/LOGS.zip)

cusglider, 2019-03-13:
2 logs of 2 flights with the default settings of Beta RC1

McGiverGim, 2019-03-13:
@cusglider I don't see anything bad in your logs. Can you take a look to them and point to the exact second in which the problem occurs? You said you have the videos, adding them in top of the blackbox will help you to gave us this information.

cusglider, 2019-03-13:
It tried to upload a short video sequence, however, it won't work. Even the zip file is not accepted. Anyway. Take the shorter log file and go to sec 25 to 20 after arming and sec 33-38. If we find a way to upload the video you would be able to see it.

McGiverGim, 2019-03-14:
I'm sorry but looking at the logs I only see very soft movements of sticks that the quad is reacting perfectly to them. The PID Error is more or less zero in both space of time you are saying. Maybe others can find what I can't, so I suggest again maybe is better for you to ask in the Slack chat.

cusglider, 2019-03-14:
It's a "micro movement" maybe better visible on the DVR. However it can be irritating, if you have a constand micro movement bank to the left and as a correction back to the right without any stick input. Unfortunaly uploading does not work with github, can i send you a dropbox link?
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

McGiverGim, 2019-03-14:
You can publish links here, there is no problem.

cusglider, 2019-03-14:
I uploaded the video in Betaflight slack

ctzsnooze, 2019-03-15:
Please make a log in the 3.5 code that seems to work better and upload it also.  I only see a small amount of D noise, nothing else bad.  Also consider disabling the ADC in your taranis to ensure smoother RC inputs.

cusglider, 2019-03-15:
Whats disabling ATC on the Taranis? Never got across this term. I can do the extra 3.5 log, no problem
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

cusglider, 2019-03-15:
With regard to the just released RC2 I will continue on that code
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

vaughnd, 2019-03-31:
I also ran in to this issue on two of my quads with 4.0 RC4 with two different fc+esc stacks and tuning, so I'm suspecting the deadband I removed based on the notes on "Bonus Section: RC Smoothing fix for FrSky Transmitters" https://github.com/betaflight/betaflight/wiki/Bidirectional-DSHOT-and-RPM-Filter is possibly causing it. I'm seeing a slow yaw and/or roll wobble with throttle cut while moving forward and no stick inputs, which I didn't have on 3.5.7 on either quad. Flies perfectly with throttle up. I have an X9D SE with hall sensors and crossfire, but I've usually used 2 on the deadband. The rc commands are quite small, but the effect is quite a bad wobble. If I just hold the sticks off center a tiny bit it flies smoothish with throttle cut. 

set rc_smoothing_input_hz = 0
set rc_smoothing_derivative_hz = 150
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1

I'm going to try again soon with the default rc smoothing set up and deadband = 2 and see what happens. Blackbox logs, dvr, and diff from my 4s quad here: https://www.dropbox.com/sh/486nsiegjdiwntd/AADYPLo4vdTmgGyTeJJGHlBRa?dl=0

log 80 + pict0274.avi are when I got my tune to a decent level and was flying without letting the sticks center themselves. log 69 + 70 and pict0260.avi you'll see the wobble as I fly forwards and drop throttle. Any ideas to test will be much appreciated please.

vaughnd, 2019-04-01:
From another group it's apparently the new abs control default of 10 causing trouble, but then cusglider getting worse movements with it turned off doesn't make sense. I'll haul a laptop to the field next time and experiment.

closed 2019-05-08 by stale[bot]