betaflight/betaflight #7773: LUX F7/MATEKF722SE with ICM20620 gyro selected wont run about 8khz/8khz with 4.0

MrExtraction, 2019-03-13:
**Describe the bug**

Prior to flashing I had no issues but after installing bf 4.0 on my LUX F7 FC (Target MATEKF722SE), changing the set gyro to the ICM20620, and switching the protocol to DSHOT1200 (running DALRC Rocket 45a 4in1 ESC which supports DSHOT1200), anytime I try to enable gyro 32kHz sampling mode and change the settings to 32kHz/32kHz and "save and reboot" it automatically changes the settings back to 8kHz/8kHz with "enable 32kHz sampling mode" disabled. The Lux f7 fc and its ICM20620 gyro should be able to handle this no problem. Resting CPU load at 8/8 is 4%.

**To Reproduce**

Install 4.0 on MATEKF722SE target FC and try to go above 8kHz/8kHz.

**Expected behavior**

Should be able to "Save and Reboot" and have the settings saved to 32kHz/32kHz

**Flight controller configuration**

```# diff

# version
# Betaflight / MATEKF722SE (MF7S) 4.0.0 Mar  8 2019 / 03:23:53 (37dd39659) MSP API: 1.41

board_name MATEKF722SE
manufacturer_id 

# start the command batch
batch start

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map
map TAER1234

# serial
serial 2 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1750 2100 0 0
aux 3 13 2 1700 2100 0 0
aux 4 35 3 1700 2100 0 0
aux 5 39 2 1300 1700 0 0
aux 6 47 3 1300 1700 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_to_use = SECOND
set acc_calibration = -19,-36,21
set mag_hardware = NONE
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT1200
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1500
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_vbat_pos = 2528
set osd_flymode_pos = 2554
set osd_altitude_pos = 23
set osd_warnings_pos = 2409
set vcd_video_system = PAL

# profile
profile 0

set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 200
set feedforward_transition = 30
set iterm_relax_type = GYRO
set i_pitch = 85
set i_roll = 80
set d_roll = 30
set f_yaw = 35
set d_min_roll = 25
set d_min_pitch = 28
set d_min_boost_gain = 30
set d_min_advance = 80

# rateprofile
rateprofile 0

set thr_expo = 58

**Setup / Versions**
 - Flight controller : Luminier Lux F7 Purchased from GetFPV;
 - Other components, TBS Nano, TBS Unify Pro HV, Dalrc Rocket 45a 4in1 ESC

mikeller, 2019-03-13:
This is by design. No benefits to using 32 kHz gyro sampling could be shown, so this mode was removed. See #7567.

closed 2019-03-13 by mikeller

MrExtraction, 2019-03-13:
Oh really? No benefits in terms of improved quad performance or
responsiveness when using 32kHz in 4.0? Or just its been discovered that
overall/just in general that settings above 8kHz show no real benefit and
just waste cpu? Relatively new to the hobby so been reading a ton of
articles about how F7s can be set to 32kHz and its great yada yada, I
assume that info is outdated? Thanks for the quick response.

mikeller, 2019-03-13:
@MrExtraction:

> No benefits in terms of improved quad performance or responsiveness when using 32kHz in 4.0?

The problem is that when running in 32 kHz mode, the noise level of the gyro signal becomes very high, requiring aggressive filtering on the firmware side that more than offsets any potential gains in responsiveness.

MrExtraction, 2019-03-13:
Okay I guess that makes sense, I have a good deal of filtering already
incorporated into my hardware and didn't seem to have any issues running
above 8kHz previously with noise which is why I was curious (but I assume
that may not address the need for filtering on the firmware side that you
mentioned?). I don't know if its out of the cards but I know personally I'd
still love the option to at least try running my quad above 8kHz/8kHz.
Either way, Thanks a bunch for Betaflight and all the work you're doing to
pump out 4.0. I've been loving it.