betaflight/betaflight #7778: BF 4.0 - CLRACINGF4 Buzzer not working

netzwergracer, 2019-03-13:
**Describe the bug**
In BF 4.0 dev 1507 the buzzer is not working at all. It does not output beeps.

**To Reproduce**
Install BF 4.0 onto CLRACINGF4 FC

**Expected behavior**
Buzzer should work for different events. On the same FC the buzzer is working perfectly fine with BF 3.5.6 release version
**Flight controller configuration**
'''# diff all

# version
# Betaflight / CLRACINGF4 (CLR4) 4.0.0 Mar 11 2019 / 12:32:46 (0deca9c6f) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CLRACINGF4
manufacturer_id 
mcu_id 001a00243436471837333638
signature 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -OSD
feature GPS
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 5 2 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 3 900 1300 0 0
aux 2 46 1 1700 2100 0 0
aux 3 13 1 1300 1700 0 0
aux 4 26 2 900 1300 0 0
aux 5 31 2 1700 2100 0 0
aux 6 47 3 1300 1700 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_lowpass_hz = 0
set dyn_lpf_gyro_max_hz = 500
set acc_calibration = 8,14,88
set min_check = 1010
set max_check = 1990
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set dshot_idle_value = 500
set motor_pwm_protocol = DSHOT1200
set failsafe_procedure = GPS-RESCUE
set current_meter = ESC
set vbat_scale = 227
set vbat_divider = 20
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_rescue_initial_alt = 30
set gps_rescue_descent_dist = 50
set gps_rescue_ground_speed = 1000
set gps_rescue_throttle_min = 1100
set gps_rescue_throttle_max = 1400
set gps_rescue_throttle_hover = 1200
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set gps_rescue_min_dth = 50
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 1
set tlm_halfduplex = OFF
set debug_mode = GYRO_SCALED

# profile
profile 0

set dterm_lowpass_hz = 0
set vbat_pid_gain = ON
set feedforward_transition = 30
set iterm_rotation = ON
set smart_feedforward = ON
set iterm_relax_type = GYRO

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 95
set pitch_rc_rate = 95
set yaw_rc_rate = 85
set roll_expo = 31
set pitch_expo = 31
set roll_srate = 81
set pitch_srate = 81
set yaw_srate = 75

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# '''

**Setup / Versions**
CLRACINGF4 
R-XSR
Unify Pro HV
Foxeer Falkor

etracer65, 2019-03-13:
`set beeper_frequency = 0`

The real original CLRACINGF4 has an on-board passive beeper that required a frequency to drive it. You're using a clone that reused the target and has different components so you need to adjust appropriately.

joshuabardwell, 2019-03-13:
Dear etracer. My FC uses the clracing target with the full authorization and permission of Cheng Lin. He was paid to do the original design of v1. Please don't call my FC a clone.

SebiWolze, 2019-03-13:
Hey folks, got the same Problem with the JBF4 V2. 
It´s not a clone of CLRacing...

netzwergracer, 2019-03-14:
Thanks a lot.
`set beeper_frequency = 0` did the trick on Bardwell F4 V2 AIO.

closed 2019-03-14 by netzwergracer