betaflight/betaflight #7781: BF 4.0 - SmartAudio doesn't work on Holybro Kakute F4V2

PatrykSokol, 2019-03-13:
SmartAudio doesn't work on any UART (software or hardware). It isn't hardware issue, because on BF 3.5 it works just fine.

To reprosduce this bug just try to configure SmartAudio on Holybro Kakute F4 V2

```
# diff

# version
# Betaflight / KAKUTEF4V2 (KTV2) 4.0.0 Mar 10 2019 / 19:19:18 (9e715845c) MSP API: 1.41

board_name KAKUTEF4V2
manufacturer_id 

# start the command batch
batch start

# name

# resources
resource LED_STRIP 1 NONE
resource CAMERA_CONTROL 1 C08

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature GPS

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 3 2 115200 9600 0 115200
serial 5 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1500 2100 0 0
aux 1 1 2 900 1100 0 0
aux 2 46 0 1875 2100 0 0
aux 3 35 3 900 1225 0 0
aux 4 39 1 1775 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_type = PT1
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 200
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 0
set acc_calibration = 172,-53,73
set rssi_scale = 200
set serialrx_provider = FPORT
set blackbox_device = SPIFLASH
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = GPS-RESCUE
set current_meter = ADC
set ibata_scale = 294
set 3d_deadband_low = 1500
set 3d_neutral = 1486
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_alt_alarm = 500
set osd_vbat_pos = 2497
set osd_rssi_pos = 2081
set osd_current_pos = 2473
set osd_mah_drawn_pos = 2504
set osd_gps_speed_pos = 2169
set osd_gps_sats_pos = 2105
set osd_home_dir_pos = 2515
set osd_home_dist_pos = 2517
set osd_altitude_pos = 2098
set osd_warnings_pos = 2482
set osd_avg_cell_voltage_pos = 2465
set osd_nvario_pos = 2137
set osd_stat_min_batt = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_max_fft = ON
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_band = 2
set vtx_freq = 5733

# profile
profile 0

set dyn_lpf_dterm_max_hz = 300
set dterm_lowpass2_hz = 0
set smart_feedforward = ON
set p_pitch = 55
set d_pitch = 43
set p_roll = 51
set d_roll = 40
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 0

set roll_rc_rate = 165
set pitch_rc_rate = 165
set yaw_rc_rate = 165
set tpa_rate = 80
set tpa_breakpoint = 1750

# end the command batch
batch end
```
**Setup / Versions**
-FC - Holybro Kakute F4 v2
-VTX - Holybro Atlatl HV v2

It was wired to literally every Tx on FC.

laz2wiringpi, 2019-03-13:
may be 
#7603
Send 0x00 bit(s) (always) even with Soft Serial. #7603
see
https://github.com/betaflight/betaflight/issues/7750

mikeller, 2019-04-04:
@PatrykSokol: Can you try with Betaflight 4.0.0-RC6, and if this is fixed close the issue please?

bop4hell, 2019-04-16:
Betaflight 4.0.0 Stable is out and all of my Kakute stack have the same issues since I test BF 4.0 jenkins.

kleveryar, 2019-04-16:
> Betaflight 4.0.0 Stable is out and all of my Kakute stack have the same issues since I test BF 4.0 jenkins in Mars.

the same issue with stable 4.0. VTX is TBS Unify Pro HV SE. Just push the button (((

bop4hell, 2019-04-16:
Hi kleveryar : did you mean than even real TBS Unify VTX have the same issues ?
Using the push button will not be an solution for every scenario... I buy VTX's with SA for this reasons...

RipperDrone, 2019-04-17:
Should I do a little writeup in the Wiki to avoid so many recurring questions about VTX compatibility? To do that, I would need some clarification from experts, though:

There seems to be different needs to handle the communication / protocols for different VTXs:
- SmartAudio 1.x vs 2.1 vs Lite
- old vs new/compliant AKK/RDQ VTX's
- Holybro (same like old AKK?)

Questions:
- which of them are supported by current BF4?
- what's the need for the others (leading 0x00 on data frame, trailing 0x00, ...??)
- what's the plan for implementation in 4.0->4.1 phase: Switchable protocol, just supporting some, ...?

Maybe once that's clearer, users will have a better idea how to implement their own modded versions and which direction to expect.

Happy to implement it into the Wiki if I can get some hints on the protocol differences and questions mentioned above...

moonlitfpv, 2019-04-19:
Smartaudio doesn't work for my Unify Pro HV Race. FC: Speedybee F4

bop4hell, 2019-04-19:
If you need data I can make tests with my vtx's
TBS Unify pro
Atalatl v1 and 2... "v1 is not smart audio ... oups ^^"
Matek HV

closed 2019-05-26 by stale[bot]