betaflight/betaflight #7786: [Feature Request] Show only SA or IRC Tramp option in OSD menu

JyeSmith, 2019-03-14:
Both Smartaudio and IRC Tramp options are displayed in the OSD menu.  Is it possible to have only one option displayed based on which protocol has been selected via the Ports tab?

I have  2 identical quads and recently replaced a VTx, so now one has TBS and the other IRC.  Im going to forget which is in which quad and this feature will make life easier :)

jflyper, 2019-03-15:
Dynamic CMS menu content is technically possible, at an expense of increased memory usage, which is crucial for flight controllers with memory size constraints. However, as we are going to drop F3 shortly, it could be done, but it is not a priority item.

Short term solution would be to name your rigs differently and show the name on the main OSD. This will also give you a chance to identify your rigs without going into CMS.

functionpointer, 2019-11-11:
Looks like this is implemented with https://github.com/betaflight/betaflight/pull/9134 and https://github.com/betaflight/betaflight/pull/9066

closed 2019-11-11 by JyeSmith