betaflight/betaflight #7787: Hi, I was wondering if it is possible to program an autopilot for a drone using beta flight.

poly12, 2019-03-15:
**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

jflyper, 2019-03-15:
@poly12 Primary focus of Betaflight is flight behaviors, so there are virtually no navigation or flight automation oriented features. Consider iNav or Ardupilot for your platform. For dead reckoning at places you can't use GPS, you may want to consider ultrasonic or RF-TOF based systems. I would start with Ardupilot within Gazebo or similar setup so that I don't have to actually implement (or deploy) the non-GPS dead reckoning system.

poly12, 2019-03-15:
Thank you so much!!

closed 2019-03-15 by mikeller