betaflight/betaflight #7788: OSD: GPS lattitude and longitude as google string

laurentopia, 2019-03-15:
**Is your feature request related to a problem? Please describe.**
Lattitude and longitude take a lot of room in the OSD

**Describe the solution you'd like**
adding the option to display the google GPS string would take up very little room. like in iNav

McGiverGim, 2019-03-15:
Do you have more info about this? Some sample? Documentation? A link to the PR on iNav?

netzwergracer, 2019-03-15:
I would prefer "What 3 Words" coordinate system.

It is easy to remember and pretty exact.
For example Gucci Store in Trump Tower NYC is `shower.speeds.bunny`

Enter these 3 words here https://map.what3words.com and you get the exact location. They also have a good app.

If you lose your quad, you can easily remember the 3 words of the last position and enter them into your cellphone.

McGiverGim, 2019-03-15:
I don't think something that this 3 words will be possible. It starts with a dictionary and the flash in our FC is very limited, so it can't be implemented.

krzysztofmatula, 2019-03-15:
I must say that the "plus code" is very compact and useful.
It is supported in INAV.

McGiverGim, 2019-03-15:
I found information about it: https://en.m.wikipedia.org/wiki/Open_Location_Code

I will look into iNav to see how complex is the transformation, it seems useful.

etracer65, 2019-03-15:
Actually, no. The plus code is not `Q27G+38`. It's `Q27G+38 New York` from google maps. The full plus code is something more like `87G8Q27G+388`. So in the end I'm not sure how useful this is vs. the full numeric coordinates as it's not really any easier to "remember".  Also many people use telemetry in their radios to capture the last known GPS position. This won't support using the plus code.

krzysztofmatula, 2019-03-15:
It's not for remembering but for taking ~2x less space on the OSD screen compared to standard lat+lon.
It's also for ~2x less time for copying from the screen to google maps on your smartphone if in need.

krzysztofmatula, 2019-03-15:
[image]
For reference: screenshot from INAV

ChilledoutUK, 2019-03-18:
Considering most screens crop more than the dvr i would just use a smaller font and push it right up the edge of what your dvr can record. 

Add to that that many transmitters can record the gps coordinates from flights through telemetry that's another route.

Finally in 4.0 you can switch the osd profiles so that when your wanting a better view you can have a trimmed down osd view and when on a long range risky flight you can switch into your gps osd mode.

Honestly looking at the above image there is a lot of redundant information.

mikeller, 2019-03-18:
@ChilledoutUK:

> Considering most screens crop more than the dvr i would just use a smaller font and push it right up the edge of what your dvr can record.

The character resolution is fixed for the commonly used MAX7456 chip, so using a smaller font won't make a difference.

krzysztofmatula, 2019-03-18:
> Honestly looking at the above image there is a lot of redundant information.

I think content of my osd screen is not the thing that is discussed here :)

leocb, 2019-03-20:
Pretty sure the iNav PR is this one: iNavFlight/inav#4158

ChilledoutUK, 2019-03-22:
@mikeller 
The resolution of the font is not the point the point i am making is that using a smaller font allows you to have the same information whilst obscuring less of the most important thing the camera video feed.

@krzysztofmatula 
You bought up the content of your osd screen first "It's not for remembering but for taking ~2x less space on the OSD screen compared to standard lat+lon."

My point about cleaning up your osd screen so you don't have so much stuff on it and optimizing the layout so that the gps coordinates are in the screens dead space but still recorded on the dvr is a valid one.

It is complete folly to request a feature of reducing the string length of gps coordinates by 50% when your screen is completely overtaken with redundant information in a huge font.

bizmar, 2019-03-26:
Plus codes is neat and all and I would welcome it. 

I propose chopping some decimal places from the GPS coords as a simpler solution, if I understand [this](https://en.wikipedia.org/wiki/Decimal_degrees) correctly our 8 decimal places locate with quad within ~1mm.. that's a bit excessive and optimistic. So If we slash that in half, we still have ~10m of resolution, but with uncertainty the search area may get big, so 5 decimal places with ~1m resolution sound like a good compromise to me. This would decrease the display area by ~30%

Raising this here to "test the waters" before I open an Issue. OSD profiles obviate the need for a lot if this. :)

McGiverGim, 2019-03-26:
We have 7 decimal places, not 8.  We can reduce it to 6, maybe 5, but to me is too short. This was discussed before here: https://github.com/betaflight/betaflight/issues/4804

bizmar, 2019-03-26:
my bad, mistyped. Ah, I see the reason for why it's like it is... with it being one big number, and not being able to specify just decimal places.. oh well.

chadmbrown, 2019-09-15:
Would a quick workaround be to add GPS Lat and Lon to the post flight stats page option? You don't need it when flying around just when you are down. Possible?

bizmar, 2019-09-15:
You can set the coordinates on a separate OSD profile, then set a aux to switch it, and then set the failsafe condition of that aux too the value that will switch to that OSD profile. 

if you have telemetry, coords will be in there too, OpenTX also displays the last good sensor value, on the sensors page (before you turn it off). Taking a photo of that is usually the most practical. :)

laurentopia, 2019-09-17:
except when you lose a quad and the only thing you have is inflight coordinates via DVR

mikeller, 2019-09-17:
@laurentopia: Not if you use logging on your TX to log the GPS coordinates that are sent over telemetry.

laurentopia, 2019-09-21:
> @laurentopia: Not if you use logging on your TX to log the GPS coordinates that are sent over telemetry.

Can logs be read from the Taranis now?

mikeller, 2019-09-21:
@laurentopia: Writing telemetry data into a file was an option in OpenTX for at least 4 years.

laurentopia, 2019-09-22:
> @laurentopia: Writing telemetry data into a file was an option in OpenTX for at least 4 years.

I asked you if you can read it from the Taranis, not if it writes a file.

bizmar, 2019-09-22:
Telemetry logging can be enabled in the special functions page. Logs are saved to the SD card  in a .csv which can be read by anything, the radio itself does not have a way to display the contents of logs. OpenTX companion has a viewer built in. 
I log every half second, the size of the LOG folder after 6 months is 22MB. 

This is how I would deal with a lost craft, which had GPS and telemetry. OpenTX keeps the last sensor values (before RC link was lost) until the radio is powered off. 
After the craft is lost:
1. don't turn off your radio
2. go to the sensors page
3. write down the coordinates displayed there, or alternatively take a photo with your phone.

mikeller, 2019-09-22:
@laurentopia: Sorry, I misunderstood your question. And no, OpenTX does not yet have a text editor built in. But you can either eject the SD card and read the log file on a device of your choice, or you can connect the Taranis to something like a mobile phone or a laptop, select 'USB Storage', and then open the log file on this device. Or do as @bizmar suggested, and note down / photograph the GPS coordinates from the sensors page.

laurentopia, 2019-09-22:
> 3\. write down the coordinates displayed there, or alternatively take a photo with your phone.

That's a decent solution. Thanks.

etracer65, 2021-02-18:
See: https://github.com/betaflight/betaflight/pull/10566

closed 2021-04-26 by mikeller