betaflight/betaflight #7792: BF 4.0 RC2 - DALRC F404 motors 1 and 2 not spinning (Dshot Bidir)

digex, 2019-03-15:
Describe the bug
After installing BF4.0  RC2 and blheli FW 32.6.7 , with bidirectional dshot and rpm filters enabled, motor 1 and motor 2 doesn`t spin.

When I disable dshot_bidir all motors spin correctly.

To Reproduce
install BF4.0 RC2 and blheli FW 32.6.7 , enable bidirectional dshot and rpm filters.

Expected behavior
After arming motor 1 and 2 does`t spin.

This is installed 
[https://github.com/betaflight/bidircfg/blob/master/DALRCF4.cf](url)

```
timer C06 0
timer C07 1

dma pin A10 1
dma pin A08 2
dma pin C08 1

set scheduler_optimize_rate=on
set dshot_burst=off
set dshot_bidir=on

set dyn_notch_width_percent = 0
set dyn_notch_range = medium
set dyn_notch_q = 250

set gyro_lowpass_hz = 150
set dyn_lpf_gyro_min_hz = 150
set dyn_lpf_gyro_max_hz = 575

set dterm_lowpass2_hz = 0

set d_min_pitch=29
set d_min_roll=27
set d_min_boost_gain=35

set debug_mode=rpm_filter

set motor_pwm_protocol=DSHOT600
set gyro_sync_denom = 2
set pid_process_denom = 1

save
```

SebiWolze, 2019-03-15:
seems to be the same problem like here:
https://github.com/betaflight/betaflight/issues/7759

digex, 2019-03-22:
Help.
set beeper_frequency = 0 not working for dalrc f405 aio
Motors 1 and 2 not spinning

JSpielz, 2019-03-23:
I have this EXACT same problem.

While I was trying to test out the RPM filter/Bidirectional Dshot, motor number 1 wouldn't spin or boot up and motor number 2 would make the startup tone a couple of seconds after 3 & 4 and not spin as well. After a bit of troubleshooting, I realized that by turning " set Dshot_burst = on" my motors would spin normally and make the startup tone up at the same time. Once I turned dshot_burst back off and changed nothing, the same thing happened. 

I started from 4.0 defaults and pasted the snippet for the DalRC F405 from the RPM filter page and that is when I noticed the issue. I didn't change anything else other than the PIDs. I even tried updating the firmware to the current latest build of 4.0 (3/22) and still the same thing. 

I just wanted to let you guys know. Hopefully you can solve the issue.

Thanks so much for all your work on BetaFlight!

(I have a DalRC F405, JB30a 4-in-1, TBS Unify 5G8 V3, FlySky X6B, Foxeer Predator V3, Johnny FPV 2700kv 2207)

promekar, 2019-03-29:
Same here i can confirm the issue.

bobbfwed, 2019-04-02:
Also confirm this is happening. I'm using BF 4.0 RC5 on DALRC F405, and the 32.6.1 test code firmware for my DALRC Engine 40A ESCs. Motors 1 and 2 don't spin, and they don't seem to get startup tone.

EnricoBara, 2019-04-03:
Confirm the same as @bobbfwed said