betaflight/betaflight #7793: RPM Filter: dterm starts heavily oscillating

pango, 2019-03-15:
Setup:
Quad was running on RC1 with default snippet, installed on 10th of March.
Diff_all is attached.
FC is an AIO FC including 32 Bit ESCs - Asgard32

Problem:
The the RPM feature and default snippet the quad instantly showed heavy dterm oscilations.
The quad did not have these oscilations with a recent BF 4.0 Version and without RPM Filtering. 

Diff, Log and Plasmatree will be attached soon.

3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and

pango, 2019-03-15:
[image]

pango, 2019-03-15:
settings:
diff all  and dump:

[0.1_initial_tune.txt](https://github.com/betaflight/betaflight/files/2970522/0.1_initial_tune.txt)

pango, 2019-03-15:
BBLog File:
[1.0_btfl_2.bbl.zip](https://github.com/betaflight/betaflight/files/2970535/1.0_btfl_2.bbl.zip)

pango, 2019-03-15:
Filter:
[image]

netzwergracer, 2019-03-15:
You have turned off all your D-term filters in your BB-Log....

[image]

dadoribeiro, 2019-03-15:
Can you post a “get dyn” from the CLI? Also, what are the specs from your quad (motors) and BLHELI 32 firmware version

pango, 2019-03-15:
@netzwergracer I guess the Betaflight Blackbox Analyser does nit show it correctly. See the above configurator Screenshot. Dterm Filter is acticive on Biquad beetween 150 and 250 hz.

McGiverGim, 2019-03-15:
The Blackbox can't show the dynamic filters (the information is not being written into the Blackbox log file).

pango, 2019-03-15:
@dadoribeiro have no acces to the quad at the moment, but above is a txt file attached with the complete dump. You should find the dyn sertings within for the moment or give me an hour or two and I will post the get dyn.

McGiverGim, 2019-03-15:
I will forget about the snippet of the wiki, at least in your case. You need to enable the RPM filter, and let all the others filters and d_min values by default. Test it like this. And then start removing the filters or adjusting them if you want.

pango, 2019-03-15:
@dadoribeiro

Motors are RCX 2207 with 2450kv
I flashed with the recommend and latest blheli hex on last sunday. And yes I picked the file again for each individual ESC. I did not spin up the motors and control status, but the bblog is writen with debug mode for  rpm_filter.
So you can find the rpm values from the ESCs in the log. What I've seen they look ok.

dadoribeiro, 2019-03-15:
Ok. I asked because several people are having problem with the latest BLHeli 32 and so far there is no single explanation. And although it is probably right I would confirm that motor poles are set at 14 (they do not show on the diff, so probably right at 14). I set it wrong the first time and my noise graph was very weird, even with filters enabled. I would also reset all filters to complete default and confirm each one in the CLI as @McGiverGim suggested. I don´t trust diff and configurators when troubleshooting. I have to see for myself.

pango, 2019-03-16:
I'm doing a test without RPM Filtering on the latest RC2.
Settings are more or less default: 
[0.1_initial_tune_no_rpm_filter.txt](https://github.com/betaflight/betaflight/files/2974082/0.1_initial_tune_no_rpm_filter.txt)

bblog is coming soon.

pango, 2019-03-16:
here is the bblog:
[btfl_001.zip](https://github.com/betaflight/betaflight/files/2974514/btfl_001.zip)

and here the Plasmatree:
[image]

pango, 2019-03-16:
I'm activating RPM Filter now.
I'd do it with these settings:

´´´ set scheduler_optimize_rate=on
set dshot_burst=off
set dshot_bidir=on

set dyn_notch_width_percent = 0

klutvott123, 2019-03-17:
@pango I would just start with defaults, RPM filter off, and get the tune right first.

I'm looking at your latest log with more or less default settings right now, and my first thoughts are that D gains are too high. Here's a roll from your log:
[image]
Notice how the pids start to oscillate as you approach full stick deflection and when the RC rate flattens out they oscillate even more. Then it settles back down as the D gain are adjusted back down to the D_min value.  I would lower D gains to get that under control. If you have powerful motors, 35/38 is just too much IMO. I'm using T-motor f40 pro II 2400KV, and the first thing I do is lower the D gains down to 30/32. 

Also I see oscillations when you raise the throttle quickly. Try lowering D_min a little maybe? Could also try increasing TPA rate a little and lowering the breakpoint. Looks like it starts when the throttle is around 1300-1400. At high throttle there's also 150Hz oscillations. Maybe try a static notch on gyro to get rid of that. 

I also had a look at the first log you posted with RPM filters on. There's still oscillations around 150Hz. There's also some noise around 400Hz which I would leave d-term lpf2 at defaults to deal with. 

That's just how i would do it, but it should give you an idea about what has to be dealt with 🙂

pango, 2019-03-17:
I just came back from two testfilghts with RPM Filtering, when I read your comment, @klutvott123  
luckily I did lower D gains and min D in the second log. 
The first one with comparable settings was done as written above. 
Here are the two log files:

[0.2_rpm_filter_on_same_settings_as_0.1_.zip](https://github.com/betaflight/betaflight/files/2975282/0.2_rpm_filter_on_same_settings_as_0.1_.zip)
[0.3_rpm_filter_on_Dgains_down.zip](https://github.com/betaflight/betaflight/files/2975283/0.3_rpm_filter_on_Dgains_down.zip)

Overall I noticed a little more propwash compared to the first flight without rpm filtering, and somhow did not like how the quad felt, can't tell why at the moment. 
Despite that the Plasma Tree shows the exact opposite, a way more cleaner gyro log for both flights.

Plasma Tree for Log 0.2:
[image]

Plasma Tree for Log 0.3:
[image]

klutvott123, 2019-03-17:
@pango Those look much better than the first log. The increased propwash is probably because with both rpm and the default filtering enabled there will be more latency. Also when lowering D_min you will get more propwash. Personally I disable both D_mins by setting them to 0 and tune D as usual. Try it and see what happens. 

Then you could go through the tuning guide https://github.com/betaflight/betaflight/wiki/Bidirectional-DSHOT-and-RPM-Filter step by step, do some logging between each step. Notice that a very clean setup was used for the guide. If you're quad is a little rough you might not get all the way there. I thought my quad was reasonably clean until I damaged the bearings on two of the motors. Not until I replaced the damaged bearings did I notice just how rough the "good" motors were. 🙂

pango, 2019-03-17:
Yes, it looks way better and most important: the issue is gone. Also the Noise peak at 150 hz is adressed with the rpm filter. 
I'm going to tune it slowly towards less filtering. let's see how it finally feels.

closed 2019-03-17 by pango